patjak/facetimehd #9: Basic functionality

kitt3n, 2015-01-13:
Basic functionality in the driver so it can work

danmilon, 2015-01-19:
added $15 to [the bounty](https://www.bountysource.com/issues/7700750-basic-functionality).

rodrigoaguilera, 2015-01-19:
I added another $15

beyerle, 2015-01-19:
and another $15

seumac, 2015-01-19:
I had the same idea a few days ago but didn't know about bountysource! Added $50 to the bounty.

kitt3n, 2015-01-19:
Yah, I thought this project needed a little omph to get going properly, since its the last thing missing for Linux on Macbook Air :+1: :D

edsiper, 2015-04-08:
I just added $50!

MasseGuillaume, 2015-04-08:
cool I'm waiting for this to get to 1M$ then I solve the issue.

kitt3n, 2015-04-08:
^ I suggest you start pushing code to the project already then, eh.

patjak, 2015-08-08:
Hi all,

Quite a few people have been donating money to this cause (currently 28 backers at $730!) therefore I've created a team on Bountysource. I personally don't need money to develop this driver (it's a fun and educational hobby project). Unfortunately finding new developers seems hard so if I'm the one getting the bounty I've decided to donate all of it minus any fees from Bountysource to charity. I'm not sure which organization to support but right now I'm thinking about Doctors Without Borders. I'm also not sure how this is usually handled on Bountysource but if other developers join in it might be appropriate to split the bounty in some reasonable way. My part of it will still go to charity. This will keep me motivated and hopefully still attract new developers.

Please note that it's still possible that this project fails so no promises.

SicVolo, 2015-08-08:
Please don't go re-donating to some charity. Give it to another developer who could help. We all need this driver, and there are other talented embedded developers who can use seven hundred bucks in a summer as a little encouragement to move this project along.

hobarrera, 2015-08-08:
I completely agree with @SicVolo. If not a developer, the Linux Foundation seems like a natural second choice, since all users of the [linux] driver will be using linux.

patjak, 2015-08-08:
@SicVolo You do have a point but as I said it would at least keep me motivated. And people with the expertise needed to do this kind of work usually have no problems finding a job where the $ per hour is much higher. Students would be the exception ofc. If the majority of people is against this I'll ofc reconsider keeping the current status quo.

patjak, 2015-08-08:
@hobarrera I appreciate what the Linux foundation is doing but my idea was to do some good for people who do not even come close to care about the main focus of what the Linux foundation is doing. But I do value everyone's opinion on this matter and I think discussing it is worthwhile.

SicVolo, 2015-08-08:
I don't think the intent of the bountysource is to pay for the time someone puts into this. Like you said, it's rather to motivate folks, who have the skills and interested in contributing to the community. It's a "thank you, here is a cup of coffee/beer on us". Plus, a student or a coder in not-so-fortunate circumstances may value this more than others. It can also go toward paying for access to the closed specs, buying stuff to test with, or shipping hardware to others. I dunno, just something better.

dakrone, 2015-08-08:
I would love to work on this project, however, I don't want to brick my work laptop, what about putting it towards hardware for other developers interested in contributing to the project?

thijskh, 2015-08-08:
@patjak: You do the work on the driver. When it's done, you receive the bounty. That is the deal. Of course you're free to spend it on booze, or donate it to a charity. That's entirely up to you and you don't need anyone's approval for that.

t1msammut, 2015-08-08:
@patjak, I 100% agree with @thijskh. What you choose to do with the money is completely up to you. 

If you're able to complete the driver, wonderful. I just pitched into the bounty, and I'd still happily buy you a beer or tea for helping me do something I could not do myself.

patjak, 2015-08-11:
Well, lets just wait and see what happens. Regardless I'll have a beer and post a picture (taken with the webcam) if we get this thing working.

kyselejsyrecek, 2015-08-22:
Hello @patjak,

I'm only a random IT student who was looking through Bountysource requests and I am certainly not the right person to help you with writing the driver. But I was fairly amazed while reading through your (and the others') comments and feel the need to express you my high esteem for your attentive intentions.

It's only by coincidence that a very close friend of mine is looking to volunteer for Doctors without borders. And so, knowing the extent of their devoted and hazardous activities, I can hardly imagine seven hundred bucks being better dispent by a talented programmer during the summer for the principal need of one Linux driver, despite being a programmer myself.

It is a pleasure and a great encouragement to encounter a person like you just by chance.

Have a nice time and good luck hacking.

hobarrera, 2015-12-23:
I'm wondering what the official status of this issue is, since the driver (even though experimental) is pretty much working (at least with mpv and a few other apps).

kitt3n, 2015-12-25:
^ Power management is still missing, so hibernation breaks the driver

JPtheJP, 2015-12-25:
Hi kitt3n, on Fedora I just added a script to unload facetimehd at sleep and reload it at resume. Maybe not the nicest solution, but reliable.

wvengen, 2016-01-08:
@kitt3n @JPtheJP the driver unloads/loads itself at suspend/resume now since b5e3e74daafe67f371973e8484ac7492075701b6

maximilianduell, 2017-04-14:
Dear @sschnelle and @patjak, please close this issue finally and decide on how to share the bounty! I am using the driver now for longer time and it is really working reliably enough (Probably I even got an older version). 

And yes, there are (were?) crashes sporadically. But I am used to those - after all I am running an "unsupported" OS on the Mac (just saying "smart battery,...", sometimes freezes coming out of suspend, apparently higher power usage than under mac os, etc.)! And yes, maybe support for power management of facetimehd is not perfect yet...

But I did not put my bounty into this so it can rot on the bank accounts of bountysource, but because I appreciate the work that you guys were and are doing. And because it is a real help if I can fire up e.g. skype from time to time without booting to mac - even if power management may still be lacking a bit, it is good enough for me.

To summarize, in my opinion please cash the money already! Proper power management is in my opinion not a basic feature - people who care about this deeply should just open another issue and put a bounty on that, please!

hobarrera, 2017-04-14:
Wow, I couldn't agree more with @maximilianduell. I use this several hours a day for work (videocalls, etc), so it's way past the point of "basic functionality", and I'd be amazed if anyone argued otherwise.

If you guys simply don't want that bounty, just re-donate it to another pending feature/project *you* want as motivation for others.

seumac, 2017-04-17:
I agree. I use this every day, so I'm happy for it to be considered functional and the bounty claimed.

sschnelle, 2017-07-06:
I had to give back my Macbook Pro to my  last employer, so i can't develop any further. I'm thinking about taking the bounty to buy a used macbook which i can use for continuing development. What do you think?

rodrigoaguilera, 2017-07-06:
@sschnelle as one of the backers of the bounty I thing is a great use of the bounty. Can it be claimed partially?

wvengen, 2017-07-06:
@sschnelle as another backer, I also think that would be a great idea. And seeing your [past contributions](https://github.com/patjak/bcwc_pcie/commits?author=sschnelle), it looks like you're quite capable of doing so :)

patjak, 2017-07-06:
@sschnelle I'm fine with you having the entire bounty if you'd like. You should be able to get a MBA for around what the bounty is at right now.

wvengen, 2017-07-06:
Cool! @sschnelle Curious to hear what you have in mind for future development.

patjak, 2017-07-06:
Closing the issue so the bounty can be awarded.

closed 2017-07-06 by patjak