patjak/facetimehd #10: What is the current status of the code supporting these webcams?

caio1982, 2015-01-17:
Hi there! I just wanted to thank you for your effort and thanks to a recent Hacker News post I decided to stop by not to bug you but to actually ask (quite generally speaking) what is the current status of the work?

Just to be clear, I am not demanding anything, I am simply curious because there is some activity going on but it's hard to tell what's up from the outside :-)

patjak, 2015-01-18:
Hi Caio
I'm currently busy with other things but I'm still working on the driver. I'll be pushing code and reporting anything of interest to this github project.

Cheers

kitt3n, 2015-01-18:
Awesome! There's a 50 dollar bounty on bountysource by me claimable when the driver's functional enough for use, hence the "issue" by me on the project.

kie27, 2015-01-19:
For anyone else wishing to add to the bounty, the URL is https://www.bountysource.com/teams/bcwc_pcie
I can also add this to the wikipedia entry.

kitt3n, 2015-01-19:
That would be amazing! :D

0xbb, 2015-01-19:
I also added 25 dollar to the bounty.

danmilon, 2015-01-19:
Also added $15.

kie27, 2015-01-19:
I have also added $10.

pquentin, 2015-01-19:
Added 25$.

kie27, 2015-01-19:
I have added this to the wiki.

patjak, 2015-01-19:
Wow, that was quick. Thanks for the help. Hopefully this will bring in more people interested in development as well.

rodrigoaguilera, 2015-01-19:
http://i1290.photobucket.com/albums/b530/Rhinosus13/meme-well-that-escalated-quickly_zps70fb859c.jpg

Se6, 2015-01-19:
Hi,

Can't contribute to the Bounty any more it seems. It says "Note: Donations are disabled because this team has no members."

kitt3n, 2015-01-19:
https://www.bountysource.com/issues/7700750-basic-functionality On this page, you can "post a bounty" to increase the bounty. :)

zimbatm, 2015-01-24:
[image]

mrjesseolsson, 2015-02-20:
@patjak I can't add to the bounty but this would be super awesome and I hope you get it working :)

JacobMillward, 2015-02-28:
I'd love to help with this, where do I get started?

patjak, 2015-02-28:
Hi @JacobMillward
It depends on how much you know about reverse engineering. I've set up an IRC channel on freenode for discussion: #bcwc_pcie

JacobMillward, 2015-02-28:
@patjak I'll be hanging out on the IRC for a while, and whenever I have some free time.

christianbundy, 2015-03-17:
@patjak I believe this issue is ready to be closed now that we have #9

closed 2015-04-10 by patjak