patjak/facetimehd #14: Extract firmware on Linux

wvengen, 2015-08-13:
I assume that we'll need a program to extract the firmware from the kext bundle at some point, so that users can obtain the firmware legally.

The current firmware extract script requires `segedit`, which is available on MacOSX. I cannot seem to find a version of `objcopy` that understands the mach-o kext bundle.

Possible solutions:
* get permission to redistribute the binary firmware image
* find a version of objcopy that understands kext bundles
* find a way to use llvm 3.7+ kext bundle parser output the section
* get segedit to run on linux (like [toolwhip](https://code.google.com/p/toolwhip/) or [this](https://github.com/rotten-apples/cctools))

radare, 2015-08-13:
Use me

wvengen, 2015-08-14:
Thanks. I'm also thinking of when the driver is ready, and users need to obtain firmware. They'll need a tool as well, in the spirit of [isight-firmware-tools](https://launchpad.net/isight-firmware-tools) (they use a string search to locate the blob).

wvengen, 2015-08-14:
I got [segedit compile on linux](https://github.com/wvengen/segedit). But it turns out that my version of AppleCameraInterface does not have the `S2ISPFIRMWARE` section. Now trying to use @radare's amazing tool to find it.

```
../AppleCameraInterface:	file format Mach-O 64-bit x86-64

Sections:
Idx Name          Size      Address          Type
  0 __text        0000d88c 0000000000000650 TEXT 
  1 __cstring     00003d1e 000000000000dedc DATA 
  2 __const       000003f8 0000000000011c00 DATA 
  3 __got         00000060 0000000000012000 DATA 
  4 __mod_init_func 00000010 0000000000012060 DATA 
  5 __mod_term_func 00000010 0000000000012070 DATA 
  6 __const       000017c8 0000000000012080 DATA 
  7 __data        00000184 0000000000013850 DATA 
  8 __common      00000060 00000000000139d8 BSS
  9 __bss         00000038 0000000000013a38 BSS
```

patjak, 2015-08-14:
It might be possible to distribute the firmware since it should be included with the bootcamp drivers. Need to check what the license is for bootcamp.

wvengen, 2015-08-23:
Interesting! [Here](https://support.apple.com/en-us/HT205016) are links to different versions of Bootcamp updates.

wvengen, 2015-08-27:
Found something in [BootCamp5.zip](https://support.apple.com/downloads/DL1720/en_US/BootCamp5.1.5621.zip), file `BootCamp/Drivers/Apple/AppleCamera64.exe`, which is a rar archive. The file `AppleCamera.sys` is the driver for the PCI device (the corresponding inf file contains `PCI\VEN_14E4&DEV_1570`). Not sure about the license (included in the rar archive).

patjak, 2015-08-27:
Yeah I also found it and the license is included there. My take on the license is that we're not allowed to distribute it but I'm no lawyer so it's not easy to say. We could ofc have a script that downloads the bootcamp drivers and extracts the firmware but the download is a bit on the heavy side :(

gltripp, 2015-08-27:
Or, probably, give the user a manual how to place the file somewhere and let the script pick it up.

however, does the file contain the needed driver anyway?

wvengen, 2015-08-27:
One thing that might be possible, is to locate the relevant portion of the zip-file, and download just the relevant portions of the zipfile to obtain AppleCamera64.exe (perhaps even something like creating a sparse zipfile). A bit tricky, but I'd be happy to see if that might work.
But let me first see if I can find the blob in there at all, good point :)

johnf, 2015-12-24:
If you can manage to extract the firmware here is some code that could be used to download directly out of Apple's Zipfile.

``` bash
# Grab Partial Zip
git clone git@github.com:planetbeing/partial-zip.git
cd partial-zip

# Compile it
cmake CMakeLists.txt
make

# Grab the exe
./partialzip https://support.apple.com/downloads/DL1720/en_US/BootCamp5.1.5621.zip BootCamp/Drivers/Apple/AppleCamera64.exe AppleCamera64.exe
```

wvengen, 2015-12-24:
Awesome! Now we only need to find the firmware in there, and make the driver [work with it](https://github.com/patjak/bcwc_pcie/issues/17#issuecomment-166680101).

mauritiusdadd, 2015-12-24:
I already did something similar in one of my [PKGBUILD] (https://github.com/mauritiusdadd/AUR/blob/master/bcwc_pcie-firmware-bc/PKGBUILD) for Arch Linux, I think I can convert it to a stand-alone script.

wvengen, 2015-12-24:
Great, I noticed that I needed to retrieve the sys file using `unrar`, and could then run the firmware extraction script from #23 on it.

wvengen, 2015-12-25:
I'm curious if we can extract the El Capitan driver from http://support.apple.com/downloads/DL1850/en_US/osxupdcombo10.11.2.dmg or http://support.apple.com/downloads/DL1849/en_US/osxupd10.11.2.dmg or http://support.apple.com/downloads/DL1845/en_US/osxupd10.11.1.dmg

The first does contain the following line:

    <bundle CFBundleVersion="5.41.0" CFBundleShortVersionString="5.41.0" SourceVersion="5041000000000" id="com.apple.driver.AppleCameraInterface" path="./System/Library/Extensions/AppleCameraInterface.kext"/>

Haven't found the kext or firmware yet, though.

mauritiusdadd, 2015-12-25:
@wvengen : That's interesting, I will look at those files as soon as I can! I'm not a lawyer, but if we are allowed to download them and they contain the driver then we made it!

patjak, 2015-12-25:
Looks promising! A bit annoying that the images are huge but maybe we can do a partial download or something (but that's a later problem).

cunei, 2015-12-26:
I can confirm that the firmware works, and can be extracted entirely on Linux (Ubuntu 14.04, in my case), by doing the following:
1) Download this update file:
```
wget http://support.apple.com/downloads/DL1849/en_US/osxupd10.11.2.dmg
```
2) to extract its content, we need xar. So, move to a new empty directory and:
```
wget https://xar.googlecode.com/files/xar-1.5.2.tar.gz
tar xfz xar-1.5.2.tar.gz 
cd xar-1.5.2
```
To compile, we may need some additional packages:
```
sudo apt-get install libxml2-dev libssl-dev
./configure
make
```
Now we have ``xar`` in ``bin/src/xar``.
3) extract the package contents. Create an empty dir, then:
```
<path-to-xar>/bin/src/xar -x -f <path-to-package>/OSXUpd10.11.2.pkg
```
You need to issue this command in an empty dir, as it will try to create a directory named exactly ``OSXUpd10.11.2.pkg``, which will fail if the source package file (which has the same name) is already in the current dir.
4) We now have a bunch of files and directories. We are interested in the file named ``OSXUpd10.11.2.pkg/Payload``, which is an archive contaning the actual packaged files. You will find on the net that the payload file is gzipped; unfortunately on recent versions the encoding changed, and is now something else entirely. There is a python script that can unpack it though, it is available at:
```
https://gist.github.com/pudquick/ff412bcb29c9c1fa4b8d
```
Copy the script to a file, add a first line containing ``#!/usr/bin/env python``, then ``chmod 700 <file>``, and run ``<file> <path>/Payload``. You will obtain a further bunch of files named ``Payload.part0?.cpio.xz``. Use unxz to uncompress the .xz files, then:
```
cat Payload.part0* | cpio -id
```
5) You can now find the needed file in:
```
./System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface
```
You can extract the firmware from there by following the instructions on the website, then proceed to the rest of the installation.

mauritiusdadd, 2015-12-26:
@cunei : thank you, that information was useful! I update the `bcwc_pcie-firmware-osx` [PKGBUILD] (https://github.com/mauritiusdadd/AUR/blob/master/bcwc_pcie-firmware-osx/PKGBUILD) so it will download the OS X update, extract the firmware and create a package for Arch Linux. Archer are encouraged to test the package while I'm updating the extraction script.

wvengen, 2015-12-26:
First step in unpacking the russian dolls (or Chinese boxes):
```sh
echo "./System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface" >files
dd if=OSXUpdCombo10.11.2.pkg/Payload.part04.cpio.xz skip=140774388 count=4416596 bs=1 | xzcat | cpio --format odc -i -E files -d
mv System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface .
```
So we only need bytes 140774388 to 145190984 from /Payload.part04.cpio.xz.

mauritiusdadd, 2015-12-26:
The script I made seems working, can someone else test it (see #27)?

wvengen, 2015-12-27:
Unpacking of Chinese boxes succeeded!
```sh
dd if=osxupd10.11.2.dmg skip=420107885 count=1783k bs=1 | xzcat | cpio --format odc -t | tail -n 10
cpio: warning: skipped 2601658 bytes of junk
1825792+0 records in
1825792+0 records out
1825792 bytes (1.8 MB) copied, 2.1127 s, 864 kB/s
xzcat: (stdin): Unexpected end of input
cpio: premature end of file
./System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS
./System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface
./System/Library/Extensions/AppleCameraInterface.kext/Contents/_CodeSignature
./System/Library/Extensions/AppleCameraInterface.kext/Contents/_CodeSignature/CodeResources
./System/Library/Extensions/AppleCameraInterface.kext/Contents/version.plist
./System/Library/Extensions/AppleCredentialManager.kext
./System/Library/Extensions/AppleCredentialManager.kext/Contents
./System/Library/Extensions/AppleCredentialManager.kext/Contents/Info.plist
./System/Library/Extensions/AppleCredentialManager.kext/Contents/MacOS
./System/Library/Extensions/AppleCredentialManager.kext/Contents/MacOS/AppleCredentialManager
```

Script for downloading `AppleCameraInterface` with only about 2MB:
```sh
#!/bin/sh
#
# Download AppleCameraInterface from Mac OS X 10.11.2 update
#

URL=https://support.apple.com/downloads/DL1849/en_US/osxupd10.11.2.dmg 
RANGE=420107885-421933677
FILE=./System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface

# get xz stream #33 from OSXUpdCombo10.11.2.pkg/Payload (part 4) inside OSXUpd10.11.2.pkg inside osxupd10.11.2.dmg
curl -s -L -r "${RANGE}" "${URL}" | xzcat -q | cpio --format odc -i -d "${FILE}"
mv "${FILE}" .
rm -Rf $(dirname "${FILE}")
```

Jancis, 2015-12-27:
```
jancis@sunshine:~/temp/bcwc_pcie/firmware$ cat ./get-fw.sh 
#!/bin/sh
#
# Download AppleCameraInterface from Mac OS X 10.11.2 update
#

URL=https://support.apple.com/downloads/DL1849/en_US/osxupd10.11.2.dmg 
RANGE=420107885-421933677
FILE=./System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface

# get xz stream #33 from OSXUpdCombo10.11.2.pkg/Payload (part 4) inside OSXUpd10.11.2.pkg inside osxupd10.11.2.dmg
curl -s -L -r "${RANGE}" "${URL}" | xzcat -q | cpio --format odc -i -d "${FILE}"
mv "${FILE}" .
rm -Rf $(dirname "${FILE}")
jancis@sunshine:~/temp/bcwc_pcie/firmware$ ./get-fw.sh 
cpio: warning: skipped 2601658 bytes of junk
xzcat: (stdin): Unexpected end of input
cpio: premature end of file
```

wvengen, 2015-12-27:
Don't worry about the errors, they are caused by not working with complete files. You should still find `AppleCameraInterface`.

Jancis, 2015-12-27:
I got file AppleCameraInterface now
```
$ md5sum ./AppleCameraInterface 
ccea5db116954513252db1ccb639ce95  ./AppleCameraInterface
```
extract_from_osx.sh now yields checksum is wrong
```
$ ./extract_from_osx.sh 
Mismatching driver hash for AppleCameraInterface (ccea5db116954513252db1ccb639ce95)
No firmware extracted!
```
Sorry for offtopic, but which is the correct way to extract bin file now? Should i edit out the md5 checking part?

wvengen, 2015-12-27:
I didn't get to extracting the driver yet ... perhaps something is still going wrong. I'm not even sure we get the correct driver version. A binary compare shows that `AppleCameraInterface` has some small differences.

Jancis, 2015-12-27:
My friend has retina + el capitan, i asked him to send the file. It's exatly the same as extracted using your script, same size, same md5sum.

```
$ ls -Al  ~/Desktop/AppleCameraInterface  ./AppleCameraInterface 
-rwxr-xr-x 1 jancis jancis 754832 dec 28 01:20 ./AppleCameraInterface
-rw-rw-r-- 1 jancis jancis 754832 dec 28 01:32 /home/jancis/Desktop/AppleCameraInterface

$ md5sum ~/Desktop/AppleCameraInterface  ./AppleCameraInterface 
ccea5db116954513252db1ccb639ce95  /home/jancis/Desktop/AppleCameraInterface
ccea5db116954513252db1ccb639ce95  ./AppleCameraInterface
```

Extraction must be correct now, your script works (it looks elegant). I just wish we could avoid compiling xar, are there any alternatives to that?

wvengen, 2015-12-27:
The firmware is _not_ affected (I get exactly the same version as the known good one). You should be able to get it using:
```
$ dd bs=1 skip=81920 count=603715 if=AppleCameraInterface | gunzip >firmware.bin
$ md5sum firmware.bin
4e1d11e205e5c55d128efa0029b268fe  firmware.bin
```

wvengen, 2015-12-27:
@Jancis thanks for checking, it looks like perhaps there are multiple El Capitan drivers floating around having the same firmware.

wvengen, 2015-12-27:
@Jancis xar is not needed, only `xzcat`, which is part of [xz-utils](http://tukaani.org/xz/) and this available at least on Debian and Ubuntu by default.

Jancis, 2015-12-27:
I removed the checking part and bin file indeed has the correct checksum. So that part should be either added or removed.
Your script works for me, I think you can make pull request with the code you made.

toaster, 2015-12-28:
Thanks for the good work!
I put the firmware extraction steps into a Gentoo ebuild, added it to my overlay and referenced it by the driver ebuild as dependency.

Are the `.dat` files also part of the `AppleCameraInterface`?

wvengen, 2015-12-29:
For reference (in case someone needs to unpack another archive in the future), some notes when preparing https://github.com/patjak/bcwc_pcie/issues/14#issuecomment-167446787
```
in uncompressed Payload.part04:
  the file:  offset 508880359 length 754832
  stream 31: offset 503316480 length 16777216
in compressed Payload.part04.xz:
  stream 31: offset 140774388 length 4416596
  > dd if=OSXUpdCombo10.11.2.pkg/Payload.part04.cpio.xz skip=140774388 count=4416596 bs=1 | xzcat | cpio --format odc -i -d
in Payload:
  offset 295318288 length 4416596 (binary search of stream 31)
  > dd if=OSXUpdCombo10.11.2.pkg/Payload skip=295318288 count=4416596 bs=1 | xzcat | cpio --format odc -i -d

Payload in xar archive OSXUpd10.11.2.pkg:
  offset 128164720 length 1266678760 (according to TOC)
  seek offset in file 128178718 length 1266678760 (patched xar to report offset when doing lseek) -> is a pbzx stream
  > dd if=OSXUpd10.11.2.pkg skip=128178718 count=1266678760 bs=1 of=xarchunk
  > python pbxz.py xarchunk

In uncompressed pkg xar archive xarchunk.part02:
  AppleCameraInterface has offset 541654125
In compressed pkg xar archive xarchunk.part02.xz:
  stream 33: offset 169514640 length 4466960
  xarchunk.part02.xz has offset 250362334 within OSXUpd10.11.2.pkg (binary search of xarchunk.part02.xz)
  > dd if=OSXUpd10.11.2.pkg skip=250362334 count=16777216 bs=1 of=xarchunk.part02.cpio.xz
  (and after a binary search of stream 33)
  > dd if=OSXUpd10.11.2.pkg skip=419877486 count=16777216 bs=1 | xzcat | cpio --format odc -t
  > dd if=OSXUpd10.11.2.pkg skip=419877486 count=1785k bs=1 | xzcat | cpio --format odc -t

5.hfs -> OSXUpd10.11.2.pkg
  offset 189001728
osxupd10.11.2.dmg -> OSXUpd10.11.2.pkg
  offset 230399

  stream 33 has offset 420107885 in dmg (binary search)
  > dd if=osxupd10.11.2.dmg skip=420107885 count=1783k bs=1 | xzcat | cpio --format odc -t | tail -n 10
```

wvengen, 2016-01-07:
I'm curious where to find the `dat` files too.

In any case, we're able to obtain firmware on Linux now. \o/

closed 2016-01-07 by wvengen

offlinehacker, 2019-11-04:
I used this command if anyone would try to figure out how to get correct offsets:

```
binwalk -R "\xFD\x37\x7A\x58\x5A" -c OSXUpd10.11.5.dmg | tail -n +4 | cut -d " " -f 1 | xargs -I{} sh -c "echo stream offset {} >&2 && dd if=OSXUpd10.11.5.dmg skip={}  bs=1 | xzcat | cpio --format odc -t" | grep -i AppleCameraInterface
```