patjak/facetimehd #15: Compatible firmware versions

frenetic1, 2015-12-11:
Apparently there are a lot of different firmware versions in the drivers and not all of them work with the driver here. The one I got from my old Mavericks System didn't work. Also the one in the Windows drivers linked in one of the defects also didn't work. 

However, after upgrading to El Capitan I got a firmware which worked. Maybe it's a good idea to document which ones work and which don't.

Working version:
OSX_HASH=a9ad5b3dc36025ff92cc41b23b7f2437
FW_HASH=4e1d11e205e5c55d128efa0029b268fe
extraction works with your script, no other changes needed.

Set files on the other hand seem to be more universal. Using the one from the Windows driver with the new Mac OS driver seems to work. Btw. will you add an extraction script for those too?

patjak, 2015-12-11:
I'm going to add some documentation for this. I've seen version 1.05, 1,38 and 1.43. Only 1.43 (El Capitan) works at the moment. The other firmwares load fine but when we start streaming we get no data. It would be nice to get one of the windows versions working since they can be downloaded (in case you've wiped the OSX partition) but we need to figure out the streaming problem first.

I need to look some more at the set files but the plan is to extract them as well. If you have working code for extracting them, please let me know and I can include it.

xgnata, 2015-12-19:
On mw up to date El Capitan, the md5sum of AppleCameraInterface is
OSX_HASH=ccea5db116954513252db1ccb639ce95
and it works : Firmware successfully extracted (4e1d11e205e5c55d128efa0029b268fe)

mauritiusdadd, 2015-12-23:
Hi,
starting from the extraction script for OS X, I made a script for extracting the firmware from both the OS X driver and the Windows Boot Camp driver, maybe it can be useful: https://gist.github.com/mauritiusdadd/ea82507d46f0d77fb922

toaster, 2015-12-23:
@patjak I got the current driver working with a firmware version 1.40.0 of Yosemite (I think it was 10.10.4). The quality was not very high (in a Google hangout) and `cheese` produced garbage and eventually made the driver fail.
With the 1.43.0 firmware it is the same on my system (Gentoo, kernel 4.3.3). I wonder if the `1871_01XX.dat` would improve the quality but I don't know how to obtain this.

patjak, 2015-12-23:
@mauritiusdadd nice, looks more robust than my version. We'll need to add offsets for the set files but that can be done later.

@toaster cheese is known not to work atm, driver is still very experimental. Running without the set files on my MBA 6,2 makes very little difference to the image quality so not sure it'll help you much either. However, there are a couple of features of the webcam that we haven't implemented yet. Will experiment with that when the driver is more stable.

closed 2015-12-26 by patjak