patjak/facetimehd #16: Corrupted image

xgnata, 2015-12-20:
The driver does stream (after a clean reload of all the related modules /dev/video0 is created) but with cheese the stream is corrupted:
https://www.dropbox.com/s/63zbswro7vyf4jo/2015-12-20-120607.jpg?dl=0
Only the very top of the image is ok.
Could be that cheese does not support high resolution correctly but so far I don't manage to get the video with another software (e.g. vlc).
Tell me if you need more testing/logs/whatever.

JPtheJP, 2015-12-22:
Hi,

dnf install xawtv

Then use the streamer command line to try to grab a picture then a video. Check http://www.tldp.org/HOWTO/Webcam-HOWTO/framegrabbers.html

On my Fedora 23 I have to unload the facetimehd (mod probe -r facetimehd) then reload it (mod probe facetimehd) between EACH try. the camera initiates only ONCE, then I have to cycle the module.

I could get a picture, and I tried a 5 seconds video, but only got like 3 frames. Hey, considering we're early to the party, it is great!!!

JPtheJP, 2015-12-22:
Also as per patjak's comment in #17 try mplayer and run player tv:// It works for me.

toaster, 2015-12-23:
I found out, that `cheese` (and `gnome-contacts` too) don't display a corrupt image if the driver is limited to the maximum hardware resolution:

```c
#define FTHD_MAX_WIDTH 1280
#define FTHD_MAX_HEIGHT 720
```

But `cheese` is still unusable (as the picture grab feature in `gnome-contacts` too).

xgnata, 2015-12-24:
Haha it looks some Linux apps have issues with very high resolution webcam.
Apple faced the same issue a few years ago. Lowering the resolution indeed
"solves" the issue with cheese. The v4l2 interface of the driver is still a
big shaky but it 's already great job.

hpe009, 2015-12-27:
@xgnata How do I lower the resolution on ubuntu 15.10?

hpe009, 2015-12-28:
@xgnata How do you force driver to load lower resolution ubuntu 15.10?

patjak, 2016-01-06:
I've lowered the max resolution in the driver to 1280x720. The ISP might be able to handle higher resolution but other components probably can't. Does this fix the issue?

xgnata, 2016-01-06:
Yes it does solve the issue at least with cheese. Thanks.
According to the Apple website (http://www.apple.com/macbook-air/features.html) it is a 720p FaceTime HD camera". Therefore limiting the resolution to 1280x720 seems to be the right thing to do.

patjak, 2016-01-06:
Great, I'll close this for now. Reopen if needed.

closed 2016-01-06 by patjak