patjak/facetimehd #17: Documentation help proposed

JPtheJP, 2015-12-21:
Hi,

I am not a developper but I'd be willing to help. I would like to write some documentation to help less technical users to install the driver on their Mac. I have a MacBook Air mid-2013 that I can use.

So far I managed to dump the firmware of my FaceTime under OSX then go back to linux and compile the module (on Fedora 23). I am now battling mod probe errors, module is not loading.

Let me know.

patjak, 2015-12-21:
Hi, the wiki is open for anyone to edit so please go ahead and add documentation there.

You probably need to load a few modules before loading facetimehd:
```
modprobe v4l2-common
modprobe videobuf2-dma-sg
modprobe videobuf2-core
```
It's possible you also need to unload a module that is for some odd reason is sharing the PCI ID with the webcam:
```
modprobe -r bdc_pci
```

Hope that helps

JPtheJP, 2015-12-22:
Thank you. It helps a lot. I'll add some generic instructions to get Linux installed, then the driver complied. I'll reinstall Fedora on my Mac to document the extra packages I pulled to get a working compile, then the modprobe order once I get it right.

saulm, 2015-12-22:
Hi guys, I'm a Web developer with a MacBook Pro 2013 and Linux installed,
I'm not a driver developer but I can help translate the docs to spanish. I
could also test on Arch and update the Wiki

SM

JPtheJP, 2015-12-22:
Hi saulm,

Help greatly welcome. I am writing the English version of a Get Started for less technical users. I'll cover Fedora as I use it. I can translate to French but, well, first I have to make it work and will update the English version only up to this point. Test on Arch would be great. If we get someone working on Ubuntu we would have a decent coverage to start with.

JP

kie27, 2015-12-22:
I am running debian so would be happy to try to do the documentation for that.

slashrsm, 2015-12-22:
I am running Arch and will try it too. Can help with docs. 

I have a problem though.... I entirely deleted OS X partition from my MBP and can't get to firmware as a result of that. Could anyone share it with me?

mandx, 2015-12-22:
I'm running Ubuntu 15.10 and I can test the driver, and like @slashrsm I also wiped all OSX partitions, so I can't extract the firmware using my own laptop.

Also, is there a section in the docs on how to cleanly "uninstall" this driver? Asking because this is also my work computer :)

hobarrera, 2015-12-22:
I'd love to help with documentation, but I too don't have OS X installed.

I personally don't know any dual-booting users either, and I don't think it's a common scenario at all, so maybe the main target should be extraction from the bootcamp zip (which would work universally) (see #14).

gianlucapettinello, 2015-12-22:
I'm arch user,
I have only archlinux installed as many others here
But on apple site there is a firmware download page. Is it ok to dowload from that?
I have also created the aur package for arch linux users
Once installed the module and made the modprobe commands, how can I test the webcam?

JPtheJP, 2015-12-22:
Hi,

We'll figure out a way to get you, guys without OSX partition, the firmware dump.

mandx, it is great to have some help on Ubuntu. I would advise to start testing out of an USB based installation if you can (check the Getting Started wiki), because work computer & in-progress driver testing might be problematic, mostly if testing lead to computer freeze.

That being said, the compiled driver is a kernel module we manually load for testing. It should not load at boot by default, and can be deleted easily.

I have to document, in the Getting Started Wiki page, what goes where when you compile the driver & how to remove it.

hobarrera, yes there is a thread about what is the best way to extract the firmware out of the Windows boot-camp drivers. If someone has the knowledge to work on this, help is appreciated (I don't have such knowledge). The easy way so far is a dump out of a working OS X, plus see my first comment.

gianlucapettinello, can you provide the link?

Guys, add an entry in the Getting Started Wiki for the distro you want to document, so we know what is covered. Even if it is just the main title (Getting Started with XXX) and a "coming soon" line below.

edsiper, 2015-12-22:
about translations: I'd suggest not to document in other languages than English, despites the good intentions and benefits, it's common to find outdated information after a few months and on this case when dealing with firmwares and HW, may increase the risk to damage something if the instructions are not 100% aligned.

JPtheJP, 2015-12-22:
Hi edsiper. I agree, the focus now is to have documentation about how to compile and use the driver on major Linux distributions, all in English. Once the driver matures and we have a stable core documentation in English we can talk about translation. But good will is good will, we need it all!

gianlucapettinello, 2015-12-22:
JPtheJP
the firmware I've found is the one in the link
https://support.apple.com/kb/DL1848?viewlocale=it_IT
it is the EFI firmware. I now realize that it is probably the wrong one.
I have also OSX installed in an external HD (El Captain) maybe when I'm back from holidays I can get the firmware from there

JPtheJP, 2015-12-22:
Hi gianlucapettinello, no we don't need the EFI firmware but the FaceTime Camera firmware dump, that is either in the Windows driver (thus the #14 issue, we don't know yet how to do it), or from the OS X driver (in 10.11.2, it is located in /System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface and you dump the firmware with the extract_from_osx.sh script).

gianlucapettinello, 2015-12-22:
Hi JptheJP, ok I will get it from the external hd installation when I have it on my hands (Jan 3rd 2016). Once I have dumped the firmware.bin, what I have to do next?
Is it pre requisite for the compilation of the module?
I have compiled the module facetimehd.ko without any firmware dump and installed in /usr/lib/modules/extramodules-4.2-ARCH
I have ran depmod and now before loading facetimehd the system loads the other modules mentioned by patjak
But no sign of /dev/video0
I suppose I'm doing something wrong
Could you help me?
Thanks

JPtheJP, 2015-12-22:
Hi gianlucapettinello, you are ahead of myself, and we need patjak's input here.

I decided to wipe my Linux to clean start and document the required packages needed on top of vanilla setup + current updates (like kernel-devel).

patjak, the question is: where/when do we need the firmware.bin file? I assumed it needed to be in the firmware subfolder at the time of compile, but is it so? Or has it to be placed somewhere & invoked another way?

gianlucapettinello, I am pretty 100% sure that without the camera firmware, you cannot get the camera to work, so obviously you miss the firmware file, and we miss the proper way to use it :-)

patjak, 2015-12-22:
Hi all, the firmware goes in ```/lib/firmware/facetimehd/firmware.bin```
Only the OSX driver works at the moment. I've been able to extract the windows versions but they don't work with our driver.

JPtheJP, 2015-12-22:
Hi patjak,

Yes it helps and I'll add it to the Getting Started document :)
I had dmesg errors about missing firmware.bin file before. Now dmesg complains about a missing `"1871_01XX.dat"` file

I did run dmesg while the firmware was missing and it reported (so gianlucapettinello, yes you can load the module without the firmware, but it fails to initiate):

`[  412.552356] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[  412.553656] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[  412.554016] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[  412.560266] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[  412.560300] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[  412.570333] facetimehd 0000:02:00.0: PLL reset finished
[  412.570339] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[  412.827785] facetimehd 0000:02:00.0: Failed to lock S2 PLL: 0xc902c902
[  412.847878] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[  412.847893] facetimehd 0000:02:00.0: STRAP valid
[  412.847895] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[  412.847902] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[  412.847912] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[  412.847923] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[  412.847924] facetimehd 0000:02:00.0: Using step size 163
[  412.847929] facetimehd 0000:02:00.0: VDL set to: coarse=0x1000a, fine=0x1011d
[  412.847934] facetimehd 0000:02:00.0: Virtual VTT enabled
[  412.868497] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[  412.868511] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[  412.868709] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[  412.871891] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/firmware.bin failed with error -2
[  412.872061] facetimehd: probe of 0000:02:00.0 failed with error -2`

Now that I copied the firmware at proper location: /lib/firmware/facetimehd/firmware.bin I am making progresses and see load errors with dmesg:

`[ 1075.824384] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[ 1075.824639] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[ 1075.831295] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[ 1075.831333] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[ 1075.841362] facetimehd 0000:02:00.0: PLL reset finished
[ 1075.841365] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[ 1075.841378] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[ 1075.851420] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[ 1075.871501] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[ 1075.871517] facetimehd 0000:02:00.0: STRAP valid
[ 1075.871519] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[ 1075.871527] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[ 1075.871537] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[ 1075.871549] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[ 1075.871550] facetimehd 0000:02:00.0: Using step size 164
[ 1075.871556] facetimehd 0000:02:00.0: VDL set to: coarse=0x1000a, fine=0x1011d
[ 1075.871562] facetimehd 0000:02:00.0: Virtual VTT enabled
[ 1075.892125] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[ 1075.892140] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[ 1075.892335] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[ 1076.349858] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[ 1076.391639] facetimehd 0000:02:00.0: ISP woke up after 0ms
[ 1076.391646] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[ 1076.391650] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[ 1076.401682] facetimehd 0000:02:00.0: ISP second int after 0ms
[ 1076.401686] facetimehd 0000:02:00.0: Channel description table at 00800000
[ 1076.411870] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[ 1076.411872] facetimehd 0000:02:00.0: Enabling interrupts
[ 1076.412048] FWMSG: 
[ 1076.412069] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[ 1076.412078] FWMSG: 
[ 1076.412090] FWMSG:   Restart count : 0
[ 1076.412111] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[ 1076.412124] FWMSG:   System Clock : 200000000 Hz
[ 1076.412243] FWMSG:   Processor mode : SUPERVISOR
[ 1076.412258] FWMSG:   Cache architecture type : SEPARATE
[ 1076.412271] FWMSG:   Cache type : WRITEBACK
[ 1076.412284] FWMSG:   Data Cache Line : 32 [0x20]
[... 6 lines cut ...]
[ 1076.412502] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[ 1076.412536] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[ 1076.412551] FWMSG:   Shared memory : base : 0x00800000
[ 1076.412570] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[ 1076.412726] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[ 1076.412748] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[ 1076.412790] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[ 1076.412807] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[ 1076.412851] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[ 1076.412865] FWMSG:   Application : adc [release]
[... cut ...]
[ 1076.413021] FWMSG: 
[ 1076.413067] FWMSG: -------------------------------------------------------------------------------
[ 1076.413102] FWMSG: 
[ 1076.486405] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[ 1076.486586] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[ 1076.486894] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[ 1076.487213] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2`

Looks like I miss a "1871_01XX.dat" firmware file.

patjak, 2015-12-22:
firmware.bin is required but the .dat files are optional. They are calibration files for the sensor and help to improve image quality. Did you try the webcam? Try ```mplayer tv://```

gianlucapettinello, 2015-12-22:
Can you share firmware.bin? I'm impatient... of course if it doesn't violate any rule. Good work in any case!

patjak, 2015-12-22:
Sorry the license prevents us from doing that.

JPtheJP, 2015-12-22:
I used streamer, which is part of the xawtv package (installed with all dependancies). I'll try player next.

I have to unload the facetimehd (modprobe -r facetimehd) then reload it (modprobe facetimehd) between EACH try. The camera initiates only ONCE, then I have to cycle the module.

I could get a picture, and I tried a 5 seconds video, but only got like 3 frames. Hey, considering we're early to the party, it is great!!!

[image]

JPtheJP, 2015-12-22:
Player installed.

mplayer tv:// just works well. Stream is clear, the video is perfectly usable. With player, I can start player, quit player and restart it: the camera initiate and turns off properly, I don't have to unload/reload the facetimehd module.

Congratulation to the developers, it is working!
I'll update the Get Started guide.

mauritiusdadd, 2015-12-22:
Hi,

I can confirm that the driver works quite well on my Macbook Pro (Retina 13", Early 2015) which is running Arch Linux.

I compiled the driver, extracted the firmware from OS X (El Capitan), installed both the module and the firmware in Arch Linux, unloaded the bdc_pci module and modprobed the facetimehd module.

The colours of the image are not very natural: the blue channel is really dark compared to the green and red ones. I also can't control the frame resolution or the FPS but hey, it actually works! And using qv4l2 I got a stable and smooth video stream of 1280x720 pixels at 30 FPS!

So, congratulation to the devs!

PS: I've created also a simple PKGBUILD (you can find it [here] (https://github.com/mauritiusdadd/AUR/tree/master/bcwc-pcie) ) for who want to try the driver under Arch, but you still need the firmware, though.

JPtheJP, 2015-12-22:
gianlucapettinello,I would need a way to contact you off site.

gianlucapettinello, 2015-12-22:
JPtheJP, I can give you my mail g_pet@hotmail.com

gianlucapettinello, 2015-12-22:
mauritiusdadd, similar to my pkgbuild. Only remark make it independent of the kernel version. Thanks

JPtheJP, 2015-12-22:
gianlucapettinello, delete your post, I have it.

JPtheJP, 2015-12-22:
mauritiusdadd, when I tested I had natural light and it looked good. Now it's dark and with light bulbs I see the image is less great, and color a bit off, indeed. Still, it is WORKING!!!!

jonlorusso, 2015-12-22:
Is there any way to access the firmware without OSX running on my machine?

JPtheJP, 2015-12-22:
Moved my Linux USB stick to the Retina MacBook Pro 15" (mid-2012, first gen) and booted it. The camera is working in there too :-) That sounds like a very good start.

slashrsm, 2015-12-23:
I was able to compile and load module on Arch with 4.2.5-1-ARCH kernel. modprobe failed with

````
➜  ~  sudo modprobe facetimehd
modprobe: FATAL: Module facetimehd not found in directory /lib/modules/4.2.5-1-ARCH
````

even if located in extramodules subfolder. However, I was able to load the module using insmod. It works like a charm. I tried it with mplayer and with Google Hangouts in Chromium.

I am willing to help with documentation for Arch.

Thanks to all the devs, you are great.

mauritiusdadd, 2015-12-23:
You have to run <b>depmod</b> after you installed the module (see the [comment above] (https://github.com/patjak/bcwc_pcie/issues/17#issuecomment-166631410) of gianlucapettinello)

slashrsm, 2015-12-23:
Yeah... Figured out shortly after I posted a comment. Thanks!

franciscoj, 2015-12-23:
I'd be happy to help with the Arch Linux part if you think it can be useful @slashrsm , or at least I can provide another machine to test it :)

slashrsm, 2015-12-23:
@franciscoj I added Arch section to the wiki. Feel free to try those steps and let me know if you notice any problems:  https://github.com/patjak/bcwc_pcie/wiki/Get-Started#get-started-on-arch

beyerle, 2015-12-23:
I have the driver running on MacBook Air 11.6" on openSUSE Leap 42.1. It works well with mplayer. The installation of the driver on openSUSE Leap does not really differ from other distros, but I'd be happy to add a section for openSUSE to the wiki.

JPtheJP, 2015-12-23:
beyerle, please do! :-)
The more people we have running / testing the driver, the better. We have years of "Camera does not work, likely never will, too proprietary" kind of comments to overcome. Spread the word! Hope we will attract further developers to help in core driver development.

hobarrera, 2015-12-23:
It's not working me on ArchLinux. Not sure if I did something wrong, or maybe there's some hardware deviation (most likely the former).

I used [this PKGBUILD](https://aur.archlinux.org/cgit/aur.git/plain/PKGBUILD?h=bcwc-pcie-git) here to build/install it, then `modprobe facetimehd`, and tried running `cheese` and `mpv tv://`:

```
$ mpv tv://
Playing: tv://
[tv] Selected driver: v4l2
[tv]  name: Video 4 Linux 2 input
[tv] unable to open '/dev/video0': No such file or directory
[tv] ioctl set mute failed: Bad file descriptor
[tv] 0 frames successfully processed, 0 frames dropped.
Failed to recognize file format.

Exiting... (Errors when loading file)
```

All I see in dmesg is:

```
[177817.455472] media: Linux media interface: v0.10
[177817.460770] Linux video capture interface: v2.00
```

Can another Arch user confirm if this works for them?
How can I get further debugging info for this?

JPtheJP, 2015-12-23:
Hi hobarrera. It looks like you load the v4l2-common and the two videobuf2-xxx modules, thus the entries you see in dmesg. It looks like you don't even load the facetimehd module. Can you check with lsmod what ends up being loaded and not? Then retry modprobe facetimehd or even insmod facetimehd (all dependancies should be there) and see if something shows up in dmesg.

hobarrera, 2015-12-23:
Actually, it did get inserted fine:

```
$ sudo lsmod | grep facetimehd
facetimehd             90112  0
videobuf2_dma_sg       20480  1 facetimehd
videobuf2_core         40960  1 facetimehd
videodev              143360  3 v4l2_common,facetimehd,videobuf2_core
```

gianlucapettinello, 2015-12-23:
Hobarrera, did you modprobe -r bdc_pci?

JPtheJP, 2015-12-23:
hobarrera, do you have a /dev/video0 entry? If not, the module can be loaded but failed to initialize (as when you don't have the firmware.bin).

hobarrera, 2015-12-23:
> Hobarrera, did you modprobe -r bdc_pci?

Oops, nope. That did it.

psanford, 2015-12-23:
I've gotten it working on ubuntu 14.04 with a little bit of effort. The driver doesn't compile against the default 14.04 kernel (3.13) but I upgraded to the linux-image-generic-lts-vivid kernel (3.19) which is compatible. 

I didn't have an OS X partition, so I followed these instructions to install OS X onto an external drive to extract the firmware binary: https://support.apple.com/en-us/HT202796.

gianlucapettinello, 2015-12-23:
Hi mauritiusdadd, I proposed in your git the mods to the PKGBUILD and install file that allow to compile also for kernel 4.2.5-1 and other kernel sources

mauritiusdadd, 2015-12-23:
@gianlucapettinello : I'm sorry, but I don't see any comment/pull request in any of my github repository, can you elaborate on how/what/where you proposed? Also, having a single PKGBUILD for multiple kernels is Very Bad™: the ideal wold be having a separate package providing the firmware and then as many packages for the kernel module an may kernels versions exist. In fact, the module must be compiled using the kernel headers and the configuration of the kernel it should be installed in.

hobarrera, 2015-12-23:
> having a single PKGBUILD for multiple kernels

This is pretty much how any other kernel-module-package works (see: broadcom-wl). It builds for the currently installed kernel by default, regardless of version.  
The package for this driver currently in the AUR does the same thing.

> the ideal would be having a separate package providing the firmware

I agree completely. I'd love to see bcwc-pcie-firmware on the AUR, and have this driver (all variants) depend on that. Especially since obtaining the firmware for every rebuild may not be feasible (and keeping expecting users to keep it around indefinitely is kind of a pain).

psanford, 2015-12-23:
I've published an ubuntu dkms package to a ppa in case it is useful for anyone else: https://launchpad.net/~psanford/+archive/ubuntu/facetimehd/

hobarrera, 2015-12-23:
@psanford Please add it to the [wiki](https://github.com/patjak/bcwc_pcie/wiki/Get-Started), if you're going to maintain it. :)

kie27, 2015-12-23:
woohoo, it works on debian testing with a macbookpro retina 15"

graingert, 2017-02-13:
@psanford can you update for yakkety?

ivanperez-keera, 2017-05-21:
@psanford Thanks for the PPA.

It would be great if the PPA was updated automatically for new releases (I'm doing this with travis for my Haskell packages and it works pretty well).

This might get more people to use the driver and, hopefully, motivate some to work on smoothing out the rough edges :)

ivanperez-keera, 2019-06-29:
@psanford Is there a chance to update this, so that we have the latest version and the driver is re-installed every time there's a new kernel update?

psanford, 2019-07-02:
I don't have this hardware anymore so I'm not able to maintain the ppa. However if you install the driver via dkms it should automatically build new versions for any new kernels you install.

BenjaminWLarson, 2024-04-02:
I realize this is an older issue, but I can not get this to work using Cheese or mplayer. I had an ssl issue during compile, found the workaround for that, then I still can't get it to work.

[root@fedora bcwc_pcie]# lsmod | grep facetimehd
facetimehd            143360  0
videobuf2_v4l2         40960  1 facetimehd
videobuf2_dma_sg       24576  3 facetimehd
videodev              393216  2 videobuf2_v4l2,facetimehd
videobuf2_common       94208  4 videobuf2_v4l2,facetimehd,videobuf2_dma_sg,videobuf2_memops

[root@fedora bcwc_pcie]# dmesg | grep facetimehd
[    2.331756] facetimehd: loading out-of-tree module taints kernel.
[    2.331761] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[    2.332946] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[    2.333084] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[    2.337940] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[    2.337966] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[    2.347979] facetimehd 0000:02:00.0: PLL reset finished
[    2.347982] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[    2.347995] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[    2.358001] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[    2.378037] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[    2.378052] facetimehd 0000:02:00.0: STRAP valid
[    2.378053] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[    2.378060] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[    2.378069] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[    2.378079] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[    2.378080] facetimehd 0000:02:00.0: Using step size 153
[    2.378085] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[    2.378090] facetimehd 0000:02:00.0: Virtual VTT enabled
[    2.398610] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[    2.398624] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[    2.398818] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[    2.521462] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[    2.564639] facetimehd 0000:02:00.0: ISP woke up after 0ms
[    2.564647] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[    2.564651] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[    2.574668] facetimehd 0000:02:00.0: ISP second int after 0ms
[    2.574672] facetimehd 0000:02:00.0: Channel description table at 00800000
[    2.584922] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[    2.584925] facetimehd 0000:02:00.0: Enabling interrupts
[    2.659625] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
[  381.245209] facetimehd 0000:02:00.0: fthd_hw_deinit
[  387.945421] facetimehd: version magic '6.7.7-100.fc38.x86_64 SMP preempt mod_unload ' should be '6.7.10-100.fc38.x86_64 SMP preempt mod_unload '
[  407.578933] facetimehd: version magic '6.7.7-100.fc38.x86_64 SMP preempt mod_unload ' should be '6.7.10-100.fc38.x86_64 SMP preempt mod_unload '
[  411.551760] facetimehd: version magic '6.7.7-100.fc38.x86_64 SMP preempt mod_unload ' should be '6.7.10-100.fc38.x86_64 SMP preempt mod_unload '
[  435.492078] facetimehd: version magic '6.7.7-100.fc38.x86_64 SMP preempt mod_unload ' should be '6.7.10-100.fc38.x86_64 SMP preempt mod_unload '
[  444.724663] facetimehd: version magic '6.7.7-100.fc38.x86_64 SMP preempt mod_unload ' should be '6.7.10-100.fc38.x86_64 SMP preempt mod_unload '
[  464.603935] facetimehd: version magic '6.7.7-100.fc38.x86_64 SMP preempt mod_unload ' should be '6.7.10-100.fc38.x86_64 SMP preempt mod_unload '
[  600.245062] facetimehd: version magic '6.7.7-100.fc38.x86_64 SMP preempt mod_unload ' should be '6.7.10-100.fc38.x86_64 SMP preempt mod_unload '
[  668.192486] facetimehd: version magic '6.7.7-100.fc38.x86_64 SMP preempt mod_unload ' should be '6.7.10-100.fc38.x86_64 SMP preempt mod_unload '
[  749.430072] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[  749.430393] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[  749.436071] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[  749.436103] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[  749.446117] facetimehd 0000:02:00.0: PLL reset finished
[  749.446120] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[  749.446133] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[  749.456139] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[  749.476177] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[  749.476193] facetimehd 0000:02:00.0: STRAP valid
[  749.476194] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[  749.476201] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[  749.476211] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[  749.476221] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[  749.476222] facetimehd 0000:02:00.0: Using step size 152
[  749.476227] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[  749.476231] facetimehd 0000:02:00.0: Virtual VTT enabled
[  749.496751] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[  749.496766] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[... cut ...]
[  749.672125] facetimehd 0000:02:00.0: ISP second int after 0ms
[  749.672127] facetimehd 0000:02:00.0: Channel description table at 00800000
[  749.682376] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[  749.682378] facetimehd 0000:02:00.0: Enabling interrupts
[  749.757470] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2

[root@fedora facetimehd]# ls /lib/firmware/facetimehd/
firmware.bin

What am I missing? if someone can help me fix, I can update the docs! ty

WhyNotHugo, 2024-04-02:
@BenjaminWLarson I'm no expert here, so I might be completely off, but this line looks relevant:

     [ 387.945421] facetimehd: version magic '6.7.7-100.fc38.x86_64 SMP preempt mod_unload ' should be '6.7.10-100.fc38.x86_64 SMP preempt mod_unload '

Sounds like you drivers are built for kernel 6.7.7, but you're running 6.7.10. You probably need to rebuild (you need to rebuild them on each kernel upgrade, ideally with something like [DKMS](https://wiki.archlinux.org/title/Dynamic_Kernel_Module_Support).

BenjaminWLarson, 2024-04-02:
@WhyNotHugo  - That sounds right...

tried to recompile and it doesn't look right:

[root@fedora bcwc_pcie]# make clean
make -C /lib/modules/6.7.10-100.fc38.x86_64/build M=/root/bcwc_pcie clean
make[1]: Entering directory '/usr/src/kernels/6.7.10-100.fc38.x86_64'
  CLEAN   /root/bcwc_pcie/Module.symvers
make[1]: Leaving directory '/usr/src/kernels/6.7.10-100.fc38.x86_64'
[root@fedora bcwc_pcie]# ls
dkms.conf            fthd_buffer.c  fthd_ddr.h      fthd_drv.c  fthd_hw.h   fthd_reg.h      fthd_v4l2.c  Makefile
facetimehd-firmware  fthd_buffer.h  fthd_debugfs.c  fthd_drv.h  fthd_isp.c  fthd_ringbuf.c  fthd_v4l2.h  README.md
firmware             fthd_ddr.c     fthd_debugfs.h  fthd_hw.c   fthd_isp.h  fthd_ringbuf.h  LICENSE
[root@fedora bcwc_pcie]# make
make -C /lib/modules/6.7.10-100.fc38.x86_64/build M=/root/bcwc_pcie modules
make[1]: Entering directory '/usr/src/kernels/6.7.10-100.fc38.x86_64'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: gcc (GCC) 13.2.1 20231011 (Red Hat 13.2.1-4)
  You are using:           gcc (GCC) 13.2.1 20240316 (Red Hat 13.2.1-7)
  CC [M]  /root/bcwc_pcie/fthd_ddr.o
  CC [M]  /root/bcwc_pcie/fthd_hw.o
  CC [M]  /root/bcwc_pcie/fthd_drv.o
  CC [M]  /root/bcwc_pcie/fthd_ringbuf.o
  CC [M]  /root/bcwc_pcie/fthd_isp.o
  CC [M]  /root/bcwc_pcie/fthd_v4l2.o
  CC [M]  /root/bcwc_pcie/fthd_buffer.o
  CC [M]  /root/bcwc_pcie/fthd_debugfs.o
  LD [M]  /root/bcwc_pcie/facetimehd.o
  MODPOST /root/bcwc_pcie/Module.symvers
  CC [M]  /root/bcwc_pcie/facetimehd.mod.o
  LD [M]  /root/bcwc_pcie/facetimehd.ko
  BTF [M] /root/bcwc_pcie/facetimehd.ko
Skipping BTF generation for /root/bcwc_pcie/facetimehd.ko due to unavailability of vmlinux
make[1]: Leaving directory '/usr/src/kernels/6.7.10-100.fc38.x86_64'

tried to follow the directions on that link and got an error missing a symbolic link

[root@fedora bcwc_pcie]# dkms status
Deprecated feature: MODULES_CONF (/var/lib/dkms/facetimehd/0.6.8a/source/dkms.conf)
Deprecated feature: MODULES_CONF (/var/lib/dkms/facetimehd/0.6.8a/source/dkms.conf)
facetimehd/0.5.18: broken
Error! facetimehd/0.5.18: Missing the module source directory or the symbolic link pointing to it.
Manual intervention is required!
facetimehd/0.6.8a, 6.7.10-100.fc38.x86_64, x86_64: installed
facetimehd/0.6.8a, 6.7.5-100.fc38.x86_64, x86_64: installed
[root@fedora bcwc_pcie]# 

I feel like I'm likely close, but no dice

BenjaminWLarson, 2024-04-02:
Okay, I'm in Google meet and it's working.... so that's interesting - guess I should try Zoom next. Will do and report my findings....

BenjaminWLarson, 2024-04-02:
Yep works for Zoom too - so I guess it's good enough for video conferencing, which was my main goal.

[1 more reply]