patjak/facetimehd #18: Proposal: Rename repository to facetimehd

psanford, 2015-12-23:
Since the kernel module is called facetimehd I think it would be good to have the repo match that name. Github knows to redirect from the old url to the new one if you use the github rename facility.

hobarrera, 2015-12-23:
I'd doble-check legal issues first, since some might arise if the project has the same name as an Apple product.

JPtheJP, 2015-12-23:
I would avoid Apple (TM) names. If we were to rename. I think the current name stands, and people will very well know that bcwb_pcie is what is needed for the camera, as they know which name to lookup for the Wireless Card. In fact the facetimehd module name should likely be changed to a non-Apple one (like bcwc_pcie :-) )

edsiper, 2015-12-23:
tuxtimehd ?

JacobMillward, 2015-12-23:
http://manpages.ubuntu.com/manpages/hardy/man1/tuxtime-conf.1.html

psanford, 2015-12-23:
Meh. The in kernel iSight drivers are called things like alsa_isight and uvc_isight and no-one seems to care.  

I just think it is confusing that the module name doesn't match the repo name. Renaming the module to bcwc_pcie would also work for me.

closed 2015-12-23 by psanford

asmagill, 2015-12-24:
Since we're talking names, might I add a suggestion that the firmware name be changed to something more specific than `firmware.bin`? Not having written a kernel module before, I don't know if this is as simple as changing a string somewhere or if it's more involved and tied to something in the firmware/code itself, but having managed linux servers for a while with older hardware that required special firmware packages or rippers like this one, I can say that a generic name will quickly get forgotten or lost when rebuilding.  Just a thought.

I hope to take a closer look at the module this weekend, but thanks for the great work!

hobarrera, 2015-12-24:
@asmagillThe firmware is located in `/usr/lib/firmware/facetimehd/`, so I don't really think there's any chance for collision.