patjak/facetimehd #20: compile problems

tr33, 2015-12-23:
Hello,

great work so far! I was very pleased to read your all success to extract the firmware.
However, I was trying to compile the driver but i got a couple of compiler errors.

I'm running Ubuntu with a self compiled  kernel (3.16.7-ckt15):
===============================
$ make
make -C /lib/modules/3.16.7-ckt15/build M=/tmp/1rba/bcwc_pcie modules
make[1]: Entering directory `/usr/src/linux-headers-3.16.7-ckt15'
  CC [M]  /tmp/1rba/bcwc_pcie/fthd_v4l2.o
/tmp/1rba/bcwc_pcie/fthd_v4l2.c: In function ‘fthd_v4l2_register’:
/tmp/1rba/bcwc_pcie/fthd_v4l2.c:697:2: error: implicit declaration of function ‘vb2_dma_sg_init_ctx’ [-Werror=implicit-function-declaration]
  dev_priv->alloc_ctx = vb2_dma_sg_init_ctx(&dev_priv->pdev->dev);
  ^
/tmp/1rba/bcwc_pcie/fthd_v4l2.c:697:22: warning: assignment makes pointer from integer without a cast [enabled by default]
  dev_priv->alloc_ctx = vb2_dma_sg_init_ctx(&dev_priv->pdev->dev);
                      ^
/tmp/1rba/bcwc_pcie/fthd_v4l2.c: In function ‘fthd_v4l2_unregister’:
/tmp/1rba/bcwc_pcie/fthd_v4l2.c:732:2: error: implicit declaration of function ‘vb2_dma_sg_cleanup_ctx’ [-Werror=implicit-function-declaration]
  vb2_dma_sg_cleanup_ctx(dev_priv->alloc_ctx);
  ^
cc1: some warnings being treated as errors
make[2]: *** [/tmp/1rba/bcwc_pcie/fthd_v4l2.o] Error 1
make[1]: *** [_module_/tmp/1rba/bcwc_pcie] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.16.7-ckt15'
make: *** [all] Error 2
===============================

Any ideas whats missing?

psanford, 2015-12-23:
This driver currently depends on a function `vb2_dma_sg_init_ctx` that is newer than your kernel. The first kernel release that should work with no modifications to the driver is 3.19. You should be able to upgrade just the kernel by installing the `linux-generic-lts-vivid` package which is a 3.19 kernel.

hobarrera, 2015-12-23:
> Hello,

> great work so far! I was very pleased to read your all success to
> extract the firmware
However, I was trying to compile the driver but i got a couple of
compiler errors

> I'm running Ubuntu with a self compiled  kernel (3167-ckt15):

> $ make
make -C /lib/modules/3167-ckt15/build M=/tmp/1rba/bcwc_pcie modules
make[1]: Entering directory /usr/src/linux-headers-3167-ckt15'
  CC [M]  /tmp/1rba/bcwc_pcie/fthd_v4l2o
/tmp/1rba/bcwc_pcie/fthd_v4l2c: In function ‘fthd_v4l2_register’:
/tmp/1rba/bcwc_pcie/fthd_v4l2c:697:2: error: implicit declaration of
function ‘vb2_dma_sg_init_ctx’ [-Werror=implicit-function-declaration]
  dev_priv->alloc_ctx = vb2_dma_sg_init_ctx(&dev_priv->pdev->dev);
  ^
/tmp/1rba/bcwc_pcie/fthd_v4l2c:697:22: warning: assignment makes pointer
from integer without a cast [enabled by default]
  dev_priv->alloc_ctx = vb2_dma_sg_init_ctx(&dev_priv->pdev->dev);
                      ^
/tmp/1rba/bcwc_pcie/fthd_v4l2c: In function ‘fthd_v4l2_unregister’:
/tmp/1rba/bcwc_pcie/fthd_v4l2c:732:2: error: implicit declaration of
function ‘vb2_dma_sg_cleanup_ctx’ [-Werror=implicit-function-
declaration]
  vb2_dma_sg_cleanup_ctx(dev_priv->alloc_ctx);
  ^
cc1: some warnings being treated as errors
make[2]: *** [/tmp/1rba/bcwc_pcie/fthd_v4l2o] Error 1
make[1]: *** [_module_/tmp/1rba/bcwc_pcie] Error 2
make[1]: Leaving directory/usr/src/linux-headers-3167-ckt15'

> make: *** [all] Error 2

> Any ideas whats missing?

There's a comment.on #17 stating that kernel 3.13 did not work, but 3.19
did. Maybe you need to upgrade yours?
--
Hugo Osvaldo Barrera
 

Links:

  1. https://github.com/patjak/bcwc_pcie/issues/20

closed 2015-12-26 by patjak