patjak/facetimehd #24: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 781: FlowIC00: Frame done timeout! frameCount = -1.

danmilon, 2015-12-24:
I installed `bcwc_pcie-firmware-bc` and `bcwc_pcie` through @mauritiusadd [AUR](https://github.com/mauritiusdadd/AUR/) packages.
I did `sudo modprobe -r bdc_pci`
I loaded `facetimehd` and then `/dev/video0` was created.
I tried `qv4l2` and `mplayer tv://`, and both times at the kernel log I get this error:
(BTW the camera LED turns green).

## Kernel log during qv4l2

```
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0106 [CISP_CMD_CH_CAMERA_CONFIG_GET]
kernel: FWMSG: [DSC] CH = 0   Camera current config Get [0] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0106 [CISP_CMD_CH_CAMERA_CONFIG_GET]
kernel: FWMSG: [DSC] CH = 0   Camera current config Get [1] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0107 [CISP_CMD_CH_CAMERA_CONFIG_SELECT]
kernel: FWMSG: [DSC] CH = 0   Camera set to config -> 0 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0801 [CISP_CMD_CH_CROP_SET]
kernel: FWMSG: [DSC] CH = 0   CROP -> [0, 0][1280, 720] within [0, 0][1280, 720]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0b01 [CISP_CMD_CH_OUTPUT_CONFIG_SET]
kernel: FWMSG: [DSC] CH = 0   New Output config -> format = 1, range 0, size = 1280x720
kernel: FWMSG: [ISP] CH = 0 CMD = 0x010e [CISP_CMD_CH_BUFFER_RECYCLE_MODE_SET]
kernel: FWMSG: [DSC] CH = 0   Dynamic Buffers Recycling Mode Set [EMPTY ONLY] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x010f [CISP_CMD_CH_BUFFER_RECYCLE_START]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x8206 [CISP_CMD_APPLE_CH_AE_METERING_MODE_SET]
kernel: FWMSG: [DSC] CH = 0   AE Metering Mode Set [3] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0c00 [CISP_CMD_CH_DRC_START]
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc300 [CISP_CMD_APPLE_CH_TONE_CURVE_ADAPTATION_START]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0218 [CISP_CMD_CH_AE_SPEED_SET]
kernel: FWMSG: [DSC] CH = 0   AE Speed Set [60] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x021a [CISP_CMD_CH_AE_STABILITY_SET]
kernel: FWMSG: [DSC] CH = 0   AE Stability Set [75] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0229 [CISP_CMD_CH_AE_STABILITY_TO_STABLE_SET]
kernel: FWMSG: [DSC] CH = 0   AE Stability Set [8] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0115 [CISP_CMD_CH_SIF_PIXEL_FORMAT_SET]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0124 [CISP_CMD_CH_CAMERA_ERR_HANDLE_CONFIG]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0d05 [CISP_CMD_CH_FACE_DETECTION_ENABLE]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0d00 [CISP_CMD_CH_FACE_DETECTION_START]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0208 [CISP_CMD_CH_AE_FRAME_RATE_MAX_SET]
kernel: FWMSG: [DSC] CH = 0   AE Frame Rate Max Set [10240] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x020a [CISP_CMD_CH_AE_FRAME_RATE_MIN_SET]
kernel: FWMSG: [DSC] CH = 0   AE Frame Rate Min Set [10240] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc100 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_START]
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc102 [CISP_CMD_APPLE_CH_MOTION_HISTORY_START]
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc113 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_ENABLE]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x8106 [CISP_CMD_APPLE_CH_STREAMING_MODE_SET]
kernel: FWMSG: [DSC] CH = 0   Streaming Mode Set [0]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0b02 [CISP_CMD_CH_SCALER_BRIGHTNESS_SET]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0b03 [CISP_CMD_CH_SCALER_CONTRAST_SET]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0100 [CISP_CMD_CH_START]
kernel: FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 781: FlowIC00: Frame done timeout! frameCount = -1.
```

## Kernel log during mplayer tv://
```
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0101 [CISP_CMD_CH_STOP]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0104 [CISP_CMD_CH_BUFFER_RETURN]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0d01 [CISP_CMD_CH_FACE_DETECTION_STOP]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0d04 [CISP_CMD_CH_FACE_DETECTION_DISABLE]
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc114 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_DISABLE]
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc103 [CISP_CMD_APPLE_CH_MOTION_HISTORY_STOP]
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc101 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_STOP]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0106 [CISP_CMD_CH_CAMERA_CONFIG_GET]
kernel: FWMSG: [DSC] CH = 0   Camera current config Get [0] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0106 [CISP_CMD_CH_CAMERA_CONFIG_GET]
kernel: FWMSG: [DSC] CH = 0   Camera current config Get [1] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0107 [CISP_CMD_CH_CAMERA_CONFIG_SELECT]
kernel: FWMSG: [DSC] CH = 0   Camera set to config -> 0 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0801 [CISP_CMD_CH_CROP_SET]
kernel: FWMSG: [DSC] CH = 0   CROP -> [160, 0][960, 720] within [0, 0][1280, 720]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0b01 [CISP_CMD_CH_OUTPUT_CONFIG_SET]
kernel: FWMSG: [DSC] CH = 0   New Output config -> format = 1, range 0, size = 640x480
kernel: FWMSG: [ISP] CH = 0 CMD = 0x010e [CISP_CMD_CH_BUFFER_RECYCLE_MODE_SET]
kernel: FWMSG: [DSC] CH = 0   Dynamic Buffers Recycling Mode Set [EMPTY ONLY] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x010f [CISP_CMD_CH_BUFFER_RECYCLE_START]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x8206 [CISP_CMD_APPLE_CH_AE_METERING_MODE_SET]
kernel: FWMSG: [DSC] CH = 0   AE Metering Mode Set [3] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0c00 [CISP_CMD_CH_DRC_START]
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc300 [CISP_CMD_APPLE_CH_TONE_CURVE_ADAPTATION_START]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0218 [CISP_CMD_CH_AE_SPEED_SET]
kernel: FWMSG: [DSC] CH = 0   AE Speed Set [60] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x021a [CISP_CMD_CH_AE_STABILITY_SET]
kernel: FWMSG: [DSC] CH = 0   AE Stability Set [75] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0229 [CISP_CMD_CH_AE_STABILITY_TO_STABLE_SET]
kernel: FWMSG: [DSC] CH = 0   AE Stability Set [8] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0115 [CISP_CMD_CH_SIF_PIXEL_FORMAT_SET]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0124 [CISP_CMD_CH_CAMERA_ERR_HANDLE_CONFIG]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0d05 [CISP_CMD_CH_FACE_DETECTION_ENABLE]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0d00 [CISP_CMD_CH_FACE_DETECTION_START]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0208 [CISP_CMD_CH_AE_FRAME_RATE_MAX_SET]
kernel: FWMSG: [DSC] CH = 0   AE Frame Rate Max Set [10240] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x020a [CISP_CMD_CH_AE_FRAME_RATE_MIN_SET]
kernel: FWMSG: [DSC] CH = 0   AE Frame Rate Min Set [10240] 
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc100 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_START]
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc102 [CISP_CMD_APPLE_CH_MOTION_HISTORY_START]
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc113 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_ENABLE]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x8106 [CISP_CMD_APPLE_CH_STREAMING_MODE_SET]
kernel: FWMSG: [DSC] CH = 0   Streaming Mode Set [0]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0b02 [CISP_CMD_CH_SCALER_BRIGHTNESS_SET]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0b03 [CISP_CMD_CH_SCALER_CONTRAST_SET]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0100 [CISP_CMD_CH_START]
kernel: FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 781: FlowIC00: Frame done timeout! frameCount = -1. 
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0101 [CISP_CMD_CH_STOP]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0104 [CISP_CMD_CH_BUFFER_RETURN]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0d01 [CISP_CMD_CH_FACE_DETECTION_STOP]
kernel: FWMSG: [ISP] CH = 0 CMD = 0x0d04 [CISP_CMD_CH_FACE_DETECTION_DISABLE]
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc114 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_DISABLE]
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc103 [CISP_CMD_APPLE_CH_MOTION_HISTORY_STOP]
kernel: FWMSG: [ISP] CH = 0 CMD = 0xc101 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_STOP]
```

And this is the error printed by mplayer tv:// 
```
Starting playback...
[VS] error (vdpVideoSurfacePutBitsYCbCr_glsl): not implemented source YCbCr format 'VDP_YCBCR_FORMAT_YUYV'
[vdpau] Error when calling vdp_video_surface_put_bits_y_cb_cr: VDP_STATUS_INVALID_Y_CB_CR_FORMAT
```

## lshw
```
danmilon-laptop           
    description: Notebook
    product: MacBookPro11,1 (System SKU#)
    vendor: Apple Inc.
    version: 1.0
    serial: C02M40BKFH04
    width: 64 bits
    capabilities: smbios-2.4 dmi-2.4 vsyscall32
    configuration: boot=normal chassis=notebook family=Mac sku=System SKU# uuid=8047E3A1-AA5E-EF5F-9219-8072761D237B
  *-core
       description: Motherboard
       product: Mac-189A3D4F975D5FFC
       vendor: Apple Inc.
       physical id: 0
       version: MacBookPro11,1
       serial: C024034051PFJYY18

...
           *-multimedia
                description: Multimedia controller
                product: 720p FaceTime HD Camera
                vendor: Broadcom Corporation
                physical id: 0
                bus info: pci@0000:02:00.0
                version: 00
                width: 64 bits
                clock: 33MHz
                capabilities: pm msi pciexpress bus_master cap_list
                configuration: driver=facetimehd latency=0
                resources: irq:51 memory:b0900000-b090ffff memory:90000000-9fffffff memory:b0800000-b08fffff
```

Any idea what's wrong? :)

patjak, 2015-12-24:
You probably have the wrong firmware version. 1.43 is required atm. Do you get any output at all?

danmilon, 2015-12-24:
I'm not sure what firmware version I've got. How can I check? It was pulled right now from the bootctamp windows drivers.

Output from what? The camera? On `qv4l2` I get all-green. On `mplayer tv://` It shuts down immediately after the error.

patjak, 2015-12-24:
You can see the version from the FWMSG dump in dmesg. Either way, only the the latest OSX firmware version (1.43) works. None of the windows versions are working atm.

danmilon, 2015-12-24:
Oh okay. I need to find a way to get hands on that then. I wonder why @mauritiusdadd went in the process to write a script to get the firmware from the windows version, given that it doesn't work.

closed 2015-12-24 by danmilon

mauritiusdadd, 2015-12-24:
Hi @danmilon,
`bcwc_pcie-firmware-bc` is for the Boot Camp Windows driver, which currently is version 1.38. I Wrote that PKGBUILD because when the version 1.43  will be (hopfully) released also for Windows, I will just need to update the download url.

danmilon, 2015-12-24:
Right, makes sense.

hobarrera, 2015-12-26:
A word of warning (mostly for posterity's sake), the above linked PKGBUILD **is not** the AUR package, but a different version.  
Noticeably, I don't see it either installing, or requiring a firmware package, which it should.

mauritiusdadd, 2015-12-26:
Yes, you're right, that's my personal PKGBUILD in my own github repo.

>Noticeably, I don't see it either installing, or requiring a firmware package, which it should.

 Howgever, it *does* download and package the Windows version of the firmware.

hobarrera, 2015-12-26:
@mauritiusdadd I don't see that on the main package: https://github.com/mauritiusdadd/AUR/blob/master/bcwc_pcie/PKGBUILD

If it's a separate one, you'd need to depend on it.

mauritiusdadd, 2015-12-26:
I'm sorry, I thought you were talking about `bcwc_pcie-firmware-bc`. Thank you for reporting the missing dependency, now it should be fixed.