patjak/facetimehd #28: Cannot get mplayer to work ubuntu

hpe009, 2015-12-26:
Getting this message when trying to load mplayer with: mplayer tv://.
MPlayer2 2.0-728-g2c378c7-4build1 (C) 2000-2012 MPlayer Team
Cannot open file '/home/havardpettersen/.mplayer/input.conf': No such file or directory
Failed to open /home/havardpettersen/.mplayer/input.conf.
Cannot open file '/etc/mplayer/input.conf': No such file or directory
Failed to open /etc/mplayer/input.conf.

Playing tv://.
Detected file format: TV
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: unable to open '/dev/video0': No such file or directory
v4l2: ioctl set mute failed: Bad file descriptor
v4l2: 0 frames successfully processed, 0 frames dropped.
Opening as detected format "TV" failed.
Failed to recognize file format.

Exiting... (End of file)

Camera does not work in Cheese either

patjak, 2015-12-26:
See https://github.com/patjak/bcwc_pcie/issues/29

hobarrera, 2015-12-26:
For posterity's sake, this is documented [here](https://github.com/patjak/bcwc_pcie/wiki/Get-Started#additional-notes).

closed 2015-12-27 by hpe009