patjak/facetimehd #29: /dev/video not created (Ubuntu 15.04)

edsiper, 2015-12-26:
Ubuntu 15.04 + Kernel 4.3

My firmware:

```
4e1d11e205e5c55d128efa0029b268fe  /lib/firmware/facetimehd/firmware.bin
```

after loading the module there is no signs the /dev/videoX interface is created:

```
$ lsmod |grep facetime
facetimehd             94208  0
videobuf2_core         49152  1 facetimehd
videobuf2_dma_sg       16384  1 facetimehd
videodev              172032  3 v4l2_common,facetimehd,videobuf2_core

$ ls -l /dev/video*
ls: cannot access /dev/video*: No such file or directory
```

How can I debug what's happening with the module ?

hobarrera, 2015-12-26:
Can you try running `modprobe -r bdc_pci` and then reinserting the module.

edsiper, 2015-12-26:
:)

[image]

edsiper, 2015-12-26:
note: this is a MacBookPro11,1

hobarrera, 2015-12-26:
For posterity's sake, this is documented [here](https://github.com/patjak/bcwc_pcie/wiki/Get-Started#additional-notes).

closed 2015-12-27 by edsiper

Jancis, 2015-12-27:
Thank you @hobarrera for suggestion to add "modprobe -r bdc_pci"
Driver works on MacBookPro11,2

begin29, 2020-09-15:
tried to do following actions for macbook 15, late 2016: 
```
# modprobe -r bdc_pci
# modprobe -r facetimehd
# modprobe facetimehd
```
/dev/video won't be created, and apps says "there no any video devices"

@WhyNotHugo any ideas?

patjak, 2020-09-17:
@begin29 what does dmesg say?