patjak/facetimehd #30: Camera just black

hpe009, 2015-12-27:
Camera just black in mplayer

patjak, 2015-12-27:
Can you load the module with debugging and post the entire dmesg after the error has occured?
```insmod ./facetimehd.ko dyndbg==pmf```

hpe009, 2015-12-27:
insmod: ERROR: could not load module ./facetimehd.ko: No such file or directory
PS: I am running ubuntu 15.10

patjak, 2015-12-27:
You need to supply the path to the module. If you've installed the module you can try
```modprobe dyndbg==pmf```

hpe009, 2015-12-27:
@patjak 
There is no error, the camera is on, but the mplayer windows is black with no video. Cheese webcamera player gives me a corrupted image but that is it.

hpe009, 2015-12-27:
If you have teamviewer on your computer you can do those commands from your pc, do you have that?

hpe009, 2015-12-27:
Getting this error from terminal while using mplayer:
[   vdpau] Error when calling vdp_video_surface_put_bits_y_cb_cr: VDP_STATUS_INVALID_Y_CB_CR_FORMAT
[VS] error (vdpVideoSurfacePutBitsYCbCr_glsl): not implemented source YCbCr format 'VDP_YCBCR_FORMAT_YUYV'

hpe009, 2015-12-27:
@patjak It is working in skype though, but not in mplayer,cheese or in websites

abustany, 2015-12-27:
@hpe009 use another "video output" for mplayer, for example "gl2": `mplayer -vo gl2 tv://`. Apparently some outputs cause mplayer to request a different pixel format for the input.

hpe009, 2015-12-27:
@abustany Thanks this command worked: mplayer -vo gl3 tv://

@patjak please add this to the install instructions

closed 2015-12-27 by hpe009

miekg, 2016-01-03:
Bumped into the same thing (sorry for adding noise to a closed issue). This was working in 14.04 (with upgraded 4.2 kernel), but is giving me this issue in ubuntu 16.04 (with kernel 4.3). 
(Yes I upgraded to 16.04 slightly before its release :-) )