patjak/facetimehd #31: Get the firmware without Mac OS Installed

jrock2004, 2015-12-27:
Ok, so I have not ran the mac os for the longest time on my Macbook Pro. Do I really need to reinstall El Capitan just to get this or is there another way to get the firmware? Thanks

miekg, 2015-12-27:
[ Quoting <notifications@github.com> in "[bcwc_pcie] Get the firmware withou..." ]
>Ok, so I have not ran the mac os for the longest time on my Macbook Pro Do I really need to reinstall El Capitan just to get this or is there another way to get the firmware? Thanks

There seem to be other ways (in the works). But yes, you'll need the driver
from El Captain. Note that upgrading made my system *not* boot Linux anymore.
Had to install refind to make that work again.

You can also try to email me privately ...

/Miek

--
Miek Gieben

jrock2004, 2015-12-27:
Yep, the problem I have is Mac OS is no longer installed. HD way too small to support multiple OS

hobarrera, 2015-12-27:
Give the code over at #27 a try, it should download the firmware for you (plus, more people testing it is a good thing).

jrock2004, 2015-12-28:
I figured it out. Thanks

closed 2015-12-28 by jrock2004