patjak/facetimehd #34: Macbook Pro 11,3 Camera not working

jrock2004, 2015-12-28:
Ok, so I followed the directions and I cannot get the camera to work. Running Ubuntu 15.10

```
❯ lsmod| grep face                                                    
facetimehd             94208  0
videobuf2_dma_sg       16384  1 facetimehd
videobuf2_core         49152  1 facetimehd
videodev              172032  3 v4l2_common,facetimehd,videobuf2_core
```

I have the firmware in the right place
```
 ls /lib/firmware/facetimehd                                        
firmware.bin

```

Any ideas?

jrock2004, 2015-12-28:
Ok, so #29 seemed to get me closer. Skype now loads my camera. Cheese loads a snowy window. Mplayer, it just seems like mplayer itself is broken.

hpe009, 2015-12-28:
For mplayer try: mplayer -v gl tv://

For cheese: go to cheese settings and lower the resolution, TURN OFF and ON again not reboot and try cheese again

patjak, 2015-12-30:
We're still working on the v4l2 and videbuf2 integration. When that is done many of these problems will go away. Not sure it's useful to find workarounds for these kind of issues at the moment. When we're out of the "experimental" stage we can revisit any remaining issues.

patjak, 2019-12-09:
This issue is very old so closing. Please reopen if needed.

closed 2019-12-09 by patjak