patjak/facetimehd #35: Module init/probe failure on MBP12,1 & Linux 4.4

mjog, 2015-12-28:
Following #14 I extracted the firmware locally, then used the README directions for Debian to build, install and load the driver under Linux 4.4 (without forgetting a `modprobe -r bdc_pci` first), but I'm not getting a working camera. This is what dmesg reports on module load:

```
Dec 28 22:31:51 payens kernel: [59190.523392] media: Linux media interface: v0.10
Dec 28 22:31:51 payens kernel: [59190.530521] Linux video capture interface: v2.00
Dec 28 22:31:51 payens kernel: [59190.539434] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
Dec 28 22:31:51 payens kernel: [59190.539535] facetimehd 0000:02:00.0: Setting 64bit DMA mask
Dec 28 22:31:51 payens kernel: [59190.544939] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
Dec 28 22:31:51 payens kernel: [59190.544969] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
Dec 28 22:31:51 payens kernel: [59190.554915] facetimehd 0000:02:00.0: PLL reset finished
Dec 28 22:31:51 payens kernel: [59190.554917] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
Dec 28 22:31:51 payens kernel: [59190.554929] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
Dec 28 22:31:51 payens kernel: [59190.564885] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
Dec 28 22:31:51 payens kernel: [59190.584804] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
Dec 28 22:31:51 payens kernel: [59190.584819] facetimehd 0000:02:00.0: STRAP valid
Dec 28 22:31:51 payens kernel: [59190.584820] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
Dec 28 22:31:51 payens kernel: [59190.584827] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
Dec 28 22:31:51 payens kernel: [59190.584836] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
Dec 28 22:31:51 payens kernel: [59190.584847] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
Dec 28 22:31:51 payens kernel: [59190.584848] facetimehd 0000:02:00.0: Using step size 152
Dec 28 22:31:51 payens kernel: [59190.584853] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
Dec 28 22:31:51 payens kernel: [59190.584857] facetimehd 0000:02:00.0: Virtual VTT enabled
Dec 28 22:31:51 payens kernel: [59190.605255] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
Dec 28 22:31:51 payens kernel: [59190.605269] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
Dec 28 22:31:51 payens kernel: [59190.605461] facetimehd 0000:02:00.0: Full memory verification failed! (65535)
Dec 28 22:31:52 payens kernel: [59191.062205] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
Dec 28 22:32:02 payens kernel: [59201.125599] facetimehd 0000:02:00.0: Init failed! No wake signal
Dec 28 22:32:02 payens kernel: [59201.125853] facetimehd: probe of 0000:02:00.0 failed with error -5
```

patjak, 2015-12-28:
The memory verification fails which means we might need the ddr phy calibration on MBP 12,1. I've been working on this for some time but not been able to get it working. It requires quite a bit of code. Is there anyone else around that tried this on a MBP 12,1?

slashrsm, 2015-12-28:
I did. Arch linux, kernel 4.2.5, camera works.

jbarnettwomply, 2015-12-28:
I'm seeing the same thing as @mjog on a MacBookPro11,4 

$ uname -a
Linux spiny 4.2.0-22-generic #27-Ubuntu SMP Thu Dec 17 22:57:08 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

mjog, 2015-12-29:
Just tried with 4.4-rc7 (was on rc5 before), same deal.

> $ uname -a
> Linux payens 4.4.0-040400rc7-generic #201512272230 SMP Mon Dec 28 03:32:16 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

@jbarnettwomply & @slashrsm: Where did you get your firmwares from? Using the method described in #14 or from an El Cap installation?

jbarnettwomply, 2015-12-29:
Mine was from #14 comments, and matched the 4e1d11e205e5c55d128efa0029b268fe md5

patjak, 2015-12-29:
@mjog I've seen the SMC behave differently depending on whether you warm boot from OSX or cold boot directly into Linux. Could you try cold booting directly into linux and see if that makes a difference?

mjog, 2015-12-30:
The firmware blob I have also matches that MD5 hash. Are there possibly different revisions of the camera chipset?

@patjak I don't have OSX installed at all on this machine, so it's always booted into Linux directly. Is there a difference in behaviour between cold boot into Linux and warm boot into Linux?

jbarnettwomply, 2015-12-30:
I also don't have an OSX installation, fwiw.

patjak, 2015-12-30:
Yes, possibly a different revision of the hardware or different BIOS/SMC version. I don't think you can see the SMC version from Linux but you can find the BIOS version with ```dmidecode```. But I'm leaning towards different hardware revision so most likely I need to finish the DDR PHY calibration code for this to work.

ddalex, 2016-01-04:
Also report the same problem, with the firmware referenced above, same dmesg stack trace, on MBP12,1; 

``` 
md5sum /lib/firmware/facetimehd/firmware.bin 
4e1d11e205e5c55d128efa0029b268fe  /lib/firmware/facetimehd/firmware.bin
```

Camera is reported as:

```
02:00.0 Multimedia controller: Broadcom Corporation Device 1570
	Subsystem: Broadcom Corporation Device 1570
	Flags: bus master, fast devsel, latency 0
	Memory at c1700000 (64-bit, non-prefetchable) [size=64K]
	Memory at a0000000 (64-bit, prefetchable) [size=256M]
	Memory at c1600000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [48] Power Management version 3
	Capabilities: [58] MSI: Enable- Count=1/1 Maskable- 64bit+
	Capabilities: [68] Vendor Specific Information: Len=44 <?>
	Capabilities: [ac] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [13c] Device Serial Number 00-00-00-ff-ff-00-00-00
	Capabilities: [150] Power Budgeting <?>
	Capabilities: [160] Virtual Channel
	Capabilities: [1b0] Latency Tolerance Reporting
	Capabilities: [220] #15
```

jbarnettwomply, 2016-01-04:
```
Handle 0x0006, DMI type 0, 24 bytes
BIOS Information
	Vendor: Apple Inc.
	Version: MBP114.88Z.0172.B04.1506051511
	Release Date: 06/05/2015
	ROM Size: 8192 kB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		ACPI is supported
		Smart battery is supported
		Function key-initiated network boot is supported
		UEFI is supported
	BIOS Revision: 0.1
```

ndavis6, 2016-01-06:
This also affects me on MB Air 7,2.  I dumped some info I thought might be helpful, happy to give more info if helpful.
```
# lspci -d 14e4:1570 -vv:

02:00.0 Multimedia controller: Broadcom Corporation 720p FaceTime HD Camera
	Subsystem: Broadcom Corporation 720p FaceTime HD Camera
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 256 bytes
	Interrupt: pin A routed to IRQ 17
	Region 0: Memory at c1500000 (64-bit, non-prefetchable) [size=64K]
	Region 2: Memory at a0000000 (64-bit, prefetchable) [size=256M]
	Region 4: Memory at c1400000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [48] Power Management version 3
		Flags: PMEClk- DSI- D1+ D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable+ DSel=0 DScale=2 PME-
	Capabilities: [58] MSI: Enable- Count=1/1 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [68] Vendor Specific Information: Len=44 <?>
	Capabilities: [ac] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 512 bytes, PhantFunc 0, Latency L0s <4us, L1 unlimited
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
		DevCtl:	Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- UncorrErr- FatalErr- UnsuppReq- AuxPwr- TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <32us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM L0s L1 Enabled; RCB 64 bytes Disabled- CommClk+
			ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Via WAKE#
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
		LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance De-emphasis: -6dB
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
			 EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
	Capabilities: [13c v1] Device Serial Number 00-00-00-ff-ff-00-00-00
	Capabilities: [150 v1] Power Budgeting <?>
	Capabilities: [160 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [1b0 v1] Latency Tolerance Reporting
		Max snoop latency: 3145728ns
		Max no snoop latency: 3145728ns
	Capabilities: [220 v1] #15
```

```
# demidecode | grep -A 10 Apple

	Vendor: Apple Inc.
	Version: MBA71.88Z.0166.B09.1510261436
	Release Date: 10/26/2015
	ROM Size: 8192 kB
	Characteristics:
		PCI is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		ACPI is supported
		Smart battery is supported
		Function key-initiated network boot is supported
		UEFI is supported
	BIOS Revision: 0.1
```

```
# dmesg

[150770.296590] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[150770.296881] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[150770.304746] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[150770.304803] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[150770.314781] facetimehd 0000:02:00.0: PLL reset finished
[150770.314787] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[150770.314849] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[150770.324859] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[150770.344953] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[150770.345708] facetimehd 0000:02:00.0: STRAP valid
[150770.345710] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[150770.346088] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[150770.346315] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 0 us
[150770.346618] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 0 us
[150770.346620] facetimehd 0000:02:00.0: Using step size 154
[150770.346847] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[150770.347074] facetimehd 0000:02:00.0: Virtual VTT enabled
[150770.368136] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[150770.368157] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[150770.368472] facetimehd 0000:02:00.0: Full memory verification failed! (65535)
[150770.371108] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/firmware.bin failed with error -2
[150770.371213] facetimehd: probe of 0000:02:00.0 failed with error -2
```

patjak, 2016-01-06:
Thanks. I'm going to take a look at a never version of the driver to see if we need some additional DDR stuff. Otherwise my bet is on the DDR PHY calibration and that will take a while to get working (if possible).

svartalf, 2016-01-14:
Can confirm it for a macbook air 2015 on the arch linux with the latest AUR packages.

```
[svartalf@tardis ~]$ dmesg | grep facetimehd
[    1.524970] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[    1.525113] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[    1.531433] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[    1.531470] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[    1.541499] facetimehd 0000:02:00.0: PLL reset finished
[    1.541502] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[    1.541515] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[    1.551553] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[    1.571662] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[    1.571677] facetimehd 0000:02:00.0: STRAP valid
[    1.571679] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[    1.571686] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[    1.571722] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[    1.571733] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[    1.571735] facetimehd 0000:02:00.0: Using step size 158
[    1.571740] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011c
[    1.571772] facetimehd 0000:02:00.0: Virtual VTT enabled
[    1.592392] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[    1.592407] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[    1.592716] facetimehd 0000:02:00.0: Full memory verification failed! (65535)
[    2.076926] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[   12.233215] facetimehd 0000:02:00.0: Init failed! No wake signal
[   12.233348] facetimehd: probe of 0000:02:00.0 failed with error -5
```

Camera detected as following:
```
[svartalf@tardis ~]$ sudo lspci -d 14e4:1570 -vv
02:00.0 Multimedia controller: Broadcom Corporation 720p FaceTime HD Camera
        Subsystem: Broadcom Corporation 720p FaceTime HD Camera
        Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Interrupt: pin A routed to IRQ 0
        Region 0: Memory at c1500000 (64-bit, non-prefetchable) [size=64K]
        Region 2: Memory at a0000000 (64-bit, prefetchable) [size=256M]
        Region 4: Memory at c1400000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [48] Power Management version 3
                Flags: PMEClk- DSI- D1+ D2- AuxCurrent=0mA PME(D0+,D1-,D2-,D3hot+,D3cold-)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=2 PME-
        Capabilities: [58] MSI: Enable- Count=1/1 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Capabilities: [68] Vendor Specific Information: Len=44 <?>
        Capabilities: [ac] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 512 bytes, PhantFunc 0, Latency L0s <4us, L1 unlimited
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset-
                DevCtl: Report errors: Correctable- Non-Fatal- Fatal- Unsupported-
                        RlxdOrd+ ExtTag- PhantFunc- AuxPwr+ NoSnoop+
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr+ UncorrErr- FatalErr- UnsuppReq+ AuxPwr- TransPend-
                LnkCap: Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <32us
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM L1 Enabled; RCB 64 bytes Disabled- CommClk+
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 5GT/s, Width x1, TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, LTR+, OBFF Via WAKE#
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR+, OBFF Disabled
                LnkCtl2: Target Link Speed: 2.5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete-, EqualizationPhase1-
                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
        Capabilities: [100 v1] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
                AERCap: First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
        Capabilities: [13c v1] Device Serial Number 00-00-00-ff-ff-00-00-00
        Capabilities: [150 v1] Power Budgeting <?>
        Capabilities: [160 v1] Virtual Channel
                Caps:   LPEVC=0 RefClk=100ns PATEntryBits=1
                Arb:    Fixed- WRR32- WRR64- WRR128-
                Ctrl:   ArbSelect=Fixed
                Status: InProgress-
[... cut ...]
                        Arb:    Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
                        Ctrl:   Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
                        Status: NegoPending- InProgress-
        Capabilities: [1b0 v1] Latency Tolerance Reporting
                Max snoop latency: 3145728ns
                Max no snoop latency: 3145728ns
        Capabilities: [220 v1] #15
        Kernel modules: bdc_pci
```

bdc_pci module is not loaded:
```
[svartalf@tardis ~]$ lsmod | grep bdc
[svartalf@tardis ~]$
```

yaguphire, 2016-01-16:
i am also getting error below with macbook pro 13" early 2015.  If you would like me to test anything I don't mind to give it a try.

[root@arch facetimehd]# md5sum firmware.bin 
4e1d11e205e5c55d128efa0029b268fe  firmware.bin

facetimehd: probe of 0000:02:00.0 failed with error -5

patjak, 2016-01-16:
ok, so it's all MacBooks from 2015. Sounds like they've put a different camera memory chip in these devices.

slashrsm, 2016-01-16:
I got my MBP 13 in June 2015. As I mentioned already it works on Arch with 4.2 kernel. 

I got my FW from ElCap. md5: 4e1d11e205e5c55d128efa0029b268fe

I apologize for late reply.

yaguphire, 2016-01-17:
Slashrsm: I have arch kernel 4.3 and 4.4 with no luck.  You don't mind sending me your copy of firmware?

yaguphire, 2016-01-17:
I've already removed Mac OS X el Capitan for arch Linux.

slashrsm, 2016-01-17:
Ping me on Twitter or use contact form on my blog.

yaguphire, 2016-01-17:
Sent message on the form on the blog.

slashrsm, 2016-01-17:
Apparently it doesn't work. Can you try Twitter? I apologize for the inconvenience.

yaguphire, 2016-01-17:
Possible to upload here?

https://www32.speedyshare.com/upload_page.php

slashrsm, 2016-01-17:
I'd rather not upload it to a public server. AFAIK it is against license...

yaguphire, 2016-01-17:
Understood.  Let me make twitter account.

yaguphire, 2016-01-17:
Just created account name yagufire

yaguphire, 2016-01-17:
if i load bdc_pci loading facetimehd loads without the probe -5 error.

Jan 17 00:08:33 arch kernel: media: Linux media interface: v0.10
Jan 17 00:08:33 arch kernel: Linux video capture interface: v2.00

although i dont see a /dev/video0

[root@arch dev]# lsmod | grep -i face
facetimehd             86016  0
videobuf2_dma_sg       20480  1 facetimehd
videobuf2_v4l2         24576  1 facetimehd
videobuf2_core         32768  2 facetimehd,videobuf2_v4l2
videodev              147456  4 v4l2_common,facetimehd,videobuf2_core,videobuf2_v4l2

lshw output

           *-multimedia
                description: Multimedia controller
                product: 720p FaceTime HD Camera
                vendor: Broadcom Corporation
                physical id: 0
                bus info: pci@0000:02:00.0
                version: 00
                width: 64 bits
                clock: 33MHz
                capabilities: pm msi pciexpress bus_master cap_list
                configuration: driver=bdc-pci latency=0
                resources: irq:63 memory:c1700000-c170ffff memory:a0000000-afffffff memory:c1600000-c16fffff

slashrsm, 2016-01-17:
/dev/video0 appears for me:

```
➜  ~  uname -a
Linux pineapple 4.2.5-1-ARCH #1 SMP PREEMPT Tue Oct 27 08:13:28 CET 2015 x86_64 GNU/Linux
➜  ~  facetime_probe  // tiny bash script which unloads bdc_pci and loads fatetimehd + it's dependencies
[sudo] password for slashrsm: 
➜  ~  ls /dev/video0 
/dev/video0
➜  ~  dmesg
// ..... some output omitted
[79595.559964] FWMSG: [ISP] CMD = 0x0001 [CISP_CMD_STOP]
[79595.571173] facetimehd 0000:02:00.0: fthd_hw_deinit
[79603.207050] media: Linux media interface: v0.10
[79603.214925] Linux video capture interface: v2.00
[79603.258811] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[79603.258922] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[79603.264145] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[79603.264172] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[79603.274208] facetimehd 0000:02:00.0: PLL reset finished
[79603.274209] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[79603.274222] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[79603.284252] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[79603.304323] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[79603.304336] facetimehd 0000:02:00.0: STRAP valid
[79603.304337] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[79603.304344] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[79603.304354] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[79603.304364] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[79603.304365] facetimehd 0000:02:00.0: Using step size 150
[79603.304370] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[79603.304374] facetimehd 0000:02:00.0: Virtual VTT enabled
[79603.324934] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[79603.324948] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[79603.325146] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[79603.784114] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[79603.932084] facetimehd 0000:02:00.0: ISP woke up after 0ms
[79603.932090] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[79603.932093] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[79603.942121] facetimehd 0000:02:00.0: ISP second int after 0ms
[79603.942123] facetimehd 0000:02:00.0: Channel description table at 00800000
[79603.952302] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[79603.952304] facetimehd 0000:02:00.0: Enabling interrupts
[79603.952532] FWMSG: 
[79603.952555] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[79603.952564] FWMSG: 
[79603.952576] FWMSG:   Restart count : 0
[79603.952597] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[79603.952685] FWMSG:   System Clock : 200000000 Hz
[79603.952698] FWMSG:   Processor mode : SUPERVISOR
[79603.952713] FWMSG:   Cache architecture type : SEPARATE
[79603.952726] FWMSG:   Cache type : WRITEBACK
[79603.952740] FWMSG:   Data Cache Line : 32 [0x20]
[79603.952753] FWMSG:   Boot arguments entries : 0
[79603.952772] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[79603.952786] FWMSG:   Main memory :   base : 0x00000000
[79603.952915] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[79603.952933] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[79603.952950] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[79603.952967] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[79603.952986] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[79603.953000] FWMSG:   Shared memory : base : 0x00800000
[79603.953138] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[79603.953158] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[79603.953179] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[79603.953303] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[79603.953319] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[79603.953337] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[79603.953350] FWMSG:   Application : adc [release]
[79603.953364] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[79603.953377] FWMSG:   Release : S2ISP-01.43.00
[79603.953390] FWMSG:   H4ISPAPPLE : 11536
[79603.953402] FWMSG:   H4ISPCD : 4081
[79603.953412] FWMSG:   ffw : 4143
[79603.953429] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[79603.953439] FWMSG: 
[79603.953564] FWMSG: -------------------------------------------------------------------------------
[79603.953576] FWMSG: 
[79604.026786] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[79604.027025] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[79604.027357] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[79604.027664] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
```

yaguphire, 2016-01-20:
i think arch linux is on kernel 4.3.  still on kernel 4.2?

svartalf, 2016-01-20:
@yaguphire I have the latest available for arch kernel (4.3.3), but it doesn't help.

```
[svartalf@tardis ~]$ uname -a
Linux tardis 4.3.3-2-ARCH #1 SMP PREEMPT Wed Dec 23 20:09:18 CET 2015 x86_64 GNU/Linux
```

hobarrera, 2016-01-20:
FWIW:
```
$ uname -a
Linux athena.barrera.io 4.4.0-3-ARCH #1 SMP PREEMPT Fri Jan 15 08:19:19 CET 2016 x86_64 GNU/Linux
```

yaguphire, 2016-01-20:
The script to extract firmware didn't work for me.. I found this dd command to pull the firmware.bin ;

$ dd bs=1 skip=81920 count=603715 if=AppleCameraInterface | gunzip >firmware.bin
$ md5sum firmware.bin
4e1d11e205e5c55d128efa0029b268fe  firmware.bin

Which gave me the same md5sum shown here.

yaguphire, 2016-01-22:
Dd command above, is okay to use to produce the firmware.bin?

yaguphire, 2016-01-24:
let me know if you guys want me to test anything.

mauritiusdadd, 2016-01-24:
@yaguphire: Why the extraction script did not work? The command you used should work because basically it is what the extraction script uses, and the MD5 sum is the correct one for the firmware version 1.43.0

yaguphire, 2016-01-24:
nm i think it worked.  i tried to pull the firmware in mac os X rather i
mounted the hfsplus partition and copied the apple firmware to the firmware
folder in the build.

[root@arch firmware]# make

Found matching hash from OS X, El Capitan 10.11.2
==> Extracting firmware...
 --> Decompressing the firmware using gzip...
 --> Deleting temporary files...
 --> Extracted firmware version 1.43.0-a

[root@arch firmware]# ls
AppleCameraInterface  extract-firmware.sh  firmware.bin  Makefile
[root@arch firmware]# md5sum firmware.bin
4e1d11e205e5c55d128efa0029b268fe  firmware.bin

yaguphire, 2016-01-24:
Jan 24 09:20:36 arch kernel: media: Linux media interface: v0.10
Jan 24 09:20:36 arch kernel: Linux video capture interface: v2.00
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: Setting 64bit DMA mask
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: S2 PCIe link init succeeded
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: PLL reset finished
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: S2 PLL is in bypass mode
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: STRAP valid
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: Using step size 151
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: Virtual VTT enabled
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: Full memory verification failed! (65535)
Jan 24 09:20:37 arch kernel: facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
Jan 24 09:20:47 arch kernel: facetimehd 0000:02:00.0: Init failed! No wake signal
Jan 24 09:20:47 arch kernel: facetimehd: probe of 0000:02:00.0 failed with error -5

[root@arch tux]# lsmod | grep facetimehd
facetimehd             86016  0
videobuf2_dma_sg       20480  1 facetimehd
videobuf2_v4l2         24576  1 facetimehd
videobuf2_core         32768  2 facetimehd,videobuf2_v4l2
videodev              147456  4 v4l2_common,facetimehd,videobuf2_core,videobuf2_v4l2
[root@arch tux]# uname -a
Linux arch 4.4.0-ARCH #1 SMP PREEMPT Fri Jan 15 20:47:22 CST 2016 x86_64 GNU/Linux

[root@arch tux]# cd /lib
[root@arch lib]# cd firmware/
[root@arch firmware]# cd facetimehd/
[root@arch facetimehd]# md5sum firmware.bin 
4e1d11e205e5c55d128efa0029b268fe  firmware.bin

askulysh, 2016-01-26:
please try 0d9c17e, it resolves the issue with MBP 12,1 and 4e1d11e205e5c55d128efa0029b268fe firmware.

mjog, 2016-01-26:
Just tried it out on my 12,1 with that rev and firmware, and no dice:

> Jan 26 12:41:31 payens kernel: [81514.006447] media: Linux media interface: v0.10
> Jan 26 12:41:31 payens kernel: [81514.017798] Linux video capture interface: v2.00
> Jan 26 12:41:31 payens kernel: [81514.029002] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
> Jan 26 12:41:31 payens kernel: [81514.029162] facetimehd 0000:02:00.0: Setting 64bit DMA mask
> Jan 26 12:41:31 payens kernel: [81514.034537] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
> Jan 26 12:41:31 payens kernel: [81514.034569] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
> Jan 26 12:41:31 payens kernel: [81514.044518] facetimehd 0000:02:00.0: PLL reset finished
> Jan 26 12:41:31 payens kernel: [81514.044520] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
> Jan 26 12:41:31 payens kernel: [81514.044533] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
> Jan 26 12:41:31 payens kernel: [81514.054490] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
> Jan 26 12:41:31 payens kernel: [81514.074413] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
> Jan 26 12:41:31 payens kernel: [81514.074428] facetimehd 0000:02:00.0: STRAP valid
> Jan 26 12:41:31 payens kernel: [81514.074429] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
> Jan 26 12:41:31 payens kernel: [81514.074436] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
> Jan 26 12:41:31 payens kernel: [81514.074445] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
> Jan 26 12:41:31 payens kernel: [81514.074456] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
> Jan 26 12:41:31 payens kernel: [81514.074457] facetimehd 0000:02:00.0: Using step size 151
> Jan 26 12:41:31 payens kernel: [81514.074461] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
> Jan 26 12:41:31 payens kernel: [81514.074466] facetimehd 0000:02:00.0: Virtual VTT enabled
> Jan 26 12:41:31 payens kernel: [81514.094865] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
> Jan 26 12:41:31 payens kernel: [81514.094881] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
> Jan 26 12:41:31 payens kernel: [81514.095082] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
> Jan 26 12:41:32 payens kernel: [81514.551988] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
> Jan 26 12:41:42 payens kernel: [81524.628227] facetimehd 0000:02:00.0: Init failed! No wake signal
> Jan 26 12:41:42 payens kernel: [81524.628567] facetimehd: probe of 0000:02:00.0 failed with error -5

> $ uname -a
> Linux payens 4.4.0-040400-generic #201601101930 SMP Mon Jan 11 00:32:41 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

yaguphire, 2016-01-26:
@askulysh will give it a try tonight will let you what I find.  I'm going to assume to make clean before replacing this file and do a make; make install.

yaguphire, 2016-01-26:
@askulysh with your changes it looks like i was able to create a /dev/video0.  Logs shown below for analysis.

Jan 25 20:39:40 arch kernel: media: Linux media interface: v0.10
Jan 25 20:39:40 arch kernel: Linux video capture interface: v2.00
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: Setting 64bit DMA mask
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: S2 PCIe link init succeeded
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: PLL reset finished
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: Failed to lock S2 PLL: 0xc902c902
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: STRAP valid
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: Using step size 150
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: Virtual VTT enabled
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
Jan 25 20:39:40 arch kernel: facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
Jan 25 20:39:41 arch kernel: facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
Jan 25 20:39:41 arch kernel: facetimehd 0000:02:00.0: ISP woke up after 0ms
Jan 25 20:39:41 arch kernel: facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
Jan 25 20:39:41 arch kernel: facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
Jan 25 20:39:41 arch kernel: facetimehd 0000:02:00.0: ISP second int after 0ms
Jan 25 20:39:41 arch kernel: facetimehd 0000:02:00.0: Channel description table at 00800000
Jan 25 20:39:41 arch kernel: facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
Jan 25 20:39:41 arch kernel: facetimehd 0000:02:00.0: Enabling interrupts
Jan 25 20:39:41 arch kernel: FWMSG: 
Jan 25 20:39:41 arch kernel: FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
Jan 25 20:39:41 arch kernel: FWMSG: 
Jan 25 20:39:41 arch kernel: FWMSG:   Restart count : 0
Jan 25 20:39:41 arch kernel: FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
Jan 25 20:39:41 arch kernel: FWMSG:   System Clock : 200000000 Hz
Jan 25 20:39:41 arch kernel: FWMSG:   Processor mode : SUPERVISOR
Jan 25 20:39:41 arch kernel: FWMSG:   Cache architecture type : SEPARATE
Jan 25 20:39:41 arch kernel: FWMSG:   Cache type : WRITEBACK
Jan 25 20:39:41 arch kernel: FWMSG:   Data Cache Line : 32 [0x20]
Jan 25 20:39:41 arch kernel: FWMSG:   Boot arguments entries : 0
Jan 25 20:39:41 arch kernel: FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
Jan 25 20:39:41 arch kernel: FWMSG:   Main memory :   base : 0x00000000
Jan 25 20:39:41 arch kernel: FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
Jan 25 20:39:41 arch kernel: FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
Jan 25 20:39:41 arch kernel: FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
Jan 25 20:39:41 arch kernel: FWMSG:   Shared window : base : 0x00800000 [static wiring]
Jan 25 20:39:41 arch kernel: FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
Jan 25 20:39:41 arch kernel: FWMSG:   Shared memory : base : 0x00800000
Jan 25 20:39:41 arch kernel: FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
Jan 25 20:39:41 arch kernel: FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
Jan 25 20:39:41 arch kernel: FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
Jan 25 20:39:41 arch kernel: FWMSG:   Heap free space : 10002626 [0x0098a0c2]
Jan 25 20:39:41 arch kernel: FWMSG:   Heap allocated space : 94656 [0x000171c0]
Jan 25 20:39:41 arch kernel: FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
Jan 25 20:39:41 arch kernel: FWMSG:   Application : adc [release]
Jan 25 20:39:41 arch kernel: FWMSG:   Linked on : Jul 25 2015 - 08:48:55
Jan 25 20:39:41 arch kernel: FWMSG:   Release : S2ISP-01.43.00
Jan 25 20:39:41 arch kernel: FWMSG:   H4ISPAPPLE : 11536
Jan 25 20:39:41 arch kernel: FWMSG:   H4ISPCD : 4081
Jan 25 20:39:41 arch kernel: FWMSG:   ffw : 4143
Jan 25 20:39:41 arch kernel: FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
Jan 25 20:39:41 arch kernel: FWMSG: 
Jan 25 20:39:41 arch kernel: FWMSG: -------------------------------------------------------------------------------
Jan 25 20:39:41 arch kernel: FWMSG: 
Jan 25 20:39:41 arch kernel: FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
Jan 25 20:39:41 arch kernel: FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
Jan 25 20:39:41 arch kernel: FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
Jan 25 20:39:41 arch kernel: facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
Jan 25 20:39:41 arch chromium.desktop[1268]: [1268:1294:0125/203941:ERROR:video_capture_device_factory_linux.cc(112)] Not implemented reached in void media::GetSupportedFormatsForV4L2BufferType(int, v4l2_buf_type, media::VideoCaptureFormats*)

yaguphire, 2016-01-26:
i can confirm webcam works.

Linux arch 4.4.0-ARCH #1 SMP PREEMPT Fri Jan 15 20:47:22 CST 2016 x86_64 GNU/Linux

[root@arch dev]# ls -l video*
crw-rw----+ 1 root video 81, 0 Jan 25 20:39 video0

[root@arch facetimehd]# md5sum firmware.bin 
4e1d11e205e5c55d128efa0029b268fe  firmware.bin

gchiesa78, 2016-01-26:
I tried the https://github.com/patjak/bcwc_pcie/issues/55 but no luck: 

```
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.881873] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.881982] facetimehd 0000:02:00.0: Setting 64bit DMA mask
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.887080] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.887108] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.897043] facetimehd 0000:02:00.0: PLL reset finished
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.897045] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.897057] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.907003] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.926920] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.926936] facetimehd 0000:02:00.0: STRAP valid
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.926937] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.926944] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.926953] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.926962] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.926963] facetimehd 0000:02:00.0: Using step size 152
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.926968] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011b
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.926972] facetimehd 0000:02:00.0: Virtual VTT enabled
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.947348] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.947362] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.947558] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
Jan 26 13:15:07 gchiesa-macbook kernel: [14506.403888] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
Jan 26 13:15:17 gchiesa-macbook kernel: [14516.468957] facetimehd 0000:02:00.0: Init failed! No wake signal
Jan 26 13:15:17 gchiesa-macbook kernel: [14516.469173] facetimehd: probe of 0000:02:00.0 failed with error -5
```

here my kernel and firmare: 
```
13:17 $ uname -a 
Linux gchiesa-macbook 4.2.0-22-generic #27~14.04.1-Ubuntu SMP Fri Dec 18 10:57:53 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
✔ ~/tmp/camera/bcwc_pcie [master|…9] 
13:17 $ md5sum /lib/firmware/facetimehd/firmware.bin 
4e1d11e205e5c55d128efa0029b268fe  /lib/firmware/facetimehd/firmware.bin
✔ ~/tmp/camera/bcwc_pcie [master|…9] 
```

askulysh, 2016-01-26:
#55 fixes only
> Jan 24 09:20:36 arch kernel: facetimehd 0000:02:00.0: Full memory verification failed! (65535)
@mjog , @gchiesa78 you get an error later.

svartalf, 2016-01-26:
Got a few minutes and tried to check 0d9c17e4b9f5907d07e8f74295303de26954f432 on the air 2015, memory verification process is still failing. Yeah, I read about the MBP 12 in [this](https://github.com/patjak/bcwc_pcie/issues/35#issuecomment-174752413) comment, but thought that I should try at least.

jbarnettwomply, 2016-01-26:
confirmed to work on my MacBookPro11,4 after doing make clean, make, checkinstall

gchiesa78, 2016-01-27:
sorry @askulysh , my bad. 
Then, I can confirm the memory verification seems to be fixed with your https://github.com/patjak/bcwc_pcie/pull/55 , but there is still the problem about the wake up signal
```
Jan 26 13:15:06 gchiesa-macbook kernel: [14505.947558] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
Jan 26 13:15:07 gchiesa-macbook kernel: [14506.403888] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
Jan 26 13:15:17 gchiesa-macbook kernel: [14516.468957] facetimehd 0000:02:00.0: Init failed! No wake signal
Jan 26 13:15:17 gchiesa-macbook kernel: [14516.469173] facetimehd: probe of 0000:02:00.0 failed with error -5
```

ndavis6, 2016-01-28:
0d9c17e fixed it for me!  Yay!

spencer-leopold, 2016-01-31:
[0d9c17e](https://github.com/patjak/bcwc_pcie/commit/0d9c17e4b9f5907d07e8f74295303de26954f432) fixed it for me as well.  MBP 11,5.

patjak, 2016-02-11:
Are people still seeing the "Full memory verification failed" error? If so, please open a new issue describing that bug. Thanks.

closed 2016-02-11 by patjak