patjak/facetimehd #37: Using camera after suspend makes the computer crash

hpe009, 2015-12-28:
On Ubuntu using camera after suspend freezes the computer

edsiper, 2015-12-28:
I'd assume this it's an expected behavior, the power management routines are not yet implemented:

https://github.com/patjak/bcwc_pcie/blob/master/fthd_drv.c#L520

hpe009, 2015-12-28:
@edsiper Anyway to go around this?

edsiper, 2015-12-28:
do not suspend ? :)

hobarrera, 2015-12-28:
Probably unloading the module before suspending should prevent this too.

hpe009, 2015-12-28:
@hobarrera Can you tell me how to do that? I hav etried but I do not think I have done it right

slashrsm, 2015-12-28:
````
modprobe -r facetimehd
````

hpe009, 2015-12-28:
Any way I can put this in a script so that it launches the commands before sleep : modprobe -r facetimehd.
And on resume: modprobe -r bdc_pci and modprobe facetimehd?

wvengen, 2015-12-28:
If you're on Ubuntu (and perhaps Debian), edit `/etc/default/acpi-support` and add `facetimehd` to its `MODULES` variable.

hpe009, 2015-12-28:
Solution. DEV please add to install description:
Create this file: gksu gedit /lib/systemd/system-sleep/99facetimehd
And paste this and save:
#!/bin/sh
case $1/$2 in
  pre/*)
    echo "Going to $2..."
    modprobe -r facetimehd 
    ;;
  post/*)
    echo "Waking up from $2..."
    modprobe -r bdc_pci
    modprobe facetimehd
    modprobe bdc_pci
    ;;
esac

patjak, 2015-12-29:
@hpe009 Thanks. Please add that to the wiki. The wiki is open for anyone to edit. Note that ```modprobe bdc_pci``` is not needed at post.

hpe009, 2015-12-29:
@patjak Will do, added that line jsut in case.

patjak, 2015-12-29:
@hpe009 Thanks

closed 2015-12-29 by patjak

derjohn, 2015-12-29:
@patjak 

For Ubuntu trusty we need:

/etc/pm/sleep.d/99_facetimehd

```
#!/bin/sh

case $1 in
    resume|thaw)
        logger -t pm 'Loading facetimehd kernel module'
        /sbin/modprobe facetimehd
        ;;
    suspend|hibernate)
        logger -t pm 'Removing facetimehd kernel module'
        /sbin/rmmod facetimehd
        exit 0
        ;;
esac
```

wvengen, 2015-12-30:
For Ubuntu we can use existing acpi-support functionality https://github.com/patjak/bcwc_pcie/issues/37#issuecomment-167600296

derjohn, 2015-12-30:
@wvengen I did try to do it via /etc/default/acpi-support - setting MODULES. Didn't work for me. I use Linux Mint 17, which is based on trusty.

hpe009, 2015-12-30:
@wvengen That does not work I have tried.

wvengen, 2016-01-08:
@hpe009 thanks for reporting back, strange.
btw, the driver unloads/loads itself now at suspend/resume since b5e3e74daafe67f371973e8484ac7492075701b6