patjak/facetimehd #39: Module init/probe failure on MBP11,2 & Linux 4.2

mmlb, 2015-12-30:
I initially thought this was related to #35 but there failure mode seems to be different and the MBPs are different revs.

```
uname -a
Linux maxwell 4.2.5-1-ARCH #1 SMP PREEMPT Tue Oct 27 08:13:28 CET 2015 x86_64 GNU/Linux

md5sum /lib/firmware/facetimehd/firmware.bin 
4e1d11e205e5c55d128efa0029b268fe  /lib/firmware/facetimehd/firmware.bin

lsmod | grep -e face -e bdc
facetimehd             90112  0
videobuf2_dma_sg       20480  1 facetimehd
videobuf2_core         40960  1 facetimehd
videodev              143360  5 gspca_main,v4l2_common,facetimehd,videobuf2_core,gspca_zc3xx
```
only Linux no OSX

```
Dec 30 11:13:47 maxwell kernel: facetimehd 0000:03:00.0: Found FaceTime HD camera with device id: 1570
Dec 30 11:13:47 maxwell kernel: facetimehd 0000:03:00.0: Setting 64bit DMA mask
Dec 30 11:13:47 maxwell kernel: facetimehd 0000:03:00.0: S2 PCIe link init succeeded
Dec 30 11:13:47 maxwell kernel: facetimehd 0000:03:00.0: Refclk: 25MHz (0xa)
Dec 30 11:13:47 maxwell kernel: facetimehd 0000:03:00.0: PLL reset finished
Dec 30 11:13:47 maxwell kernel: facetimehd 0000:03:00.0: Waiting for S2 PLL to lock at 450 MHz
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Failed to lock S2 PLL: 0xc902c902
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: DDR40 PHY PLL locked on safe settings
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: STRAP valid
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Configuring DDR PLLs for 450 MHz
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: DDR40 PLL is locked after 0 us
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: First DDR40 VDL calibration completed after 2 us
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Second DDR40 VDL calibration completed after 1 us
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Using step size 151
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: VDL set to: coarse=0x10008, fine=0x1011a
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Virtual VTT enabled
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: S2 DRAM memory address: 0x22159559
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Rewrite DDR mode registers succeeded
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Full memory verification succeeded! (0)
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Loaded firmware, size: 1392kb
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: ISP woke up after 0ms
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Number of IPC channels: 7, queue size: 44865
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Firmware requested heap size: 3072kb
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: ISP second int after 0ms
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Channel description table at 00800000
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: magic value: 00000000 after 0 ms
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Enabling interrupts
Dec 30 11:13:48 maxwell kernel: FWMSG: 
Dec 30 11:13:48 maxwell kernel: FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
Dec 30 11:13:48 maxwell kernel: FWMSG: 
Dec 30 11:13:48 maxwell kernel: FWMSG:   Restart count : 0
Dec 30 11:13:48 maxwell kernel: FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
Dec 30 11:13:48 maxwell kernel: FWMSG:   System Clock : 200000000 Hz
Dec 30 11:13:48 maxwell kernel: FWMSG:   Processor mode : SUPERVISOR
Dec 30 11:13:48 maxwell kernel: FWMSG:   Cache architecture type : SEPARATE
Dec 30 11:13:48 maxwell kernel: FWMSG:   Cache type : WRITEBACK
Dec 30 11:13:48 maxwell kernel: FWMSG:   Data Cache Line : 32 [0x20]
Dec 30 11:13:48 maxwell kernel: FWMSG:   Boot arguments entries : 0
Dec 30 11:13:48 maxwell kernel: FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
Dec 30 11:13:48 maxwell kernel: FWMSG:   Main memory :   base : 0x00000000
Dec 30 11:13:48 maxwell kernel: FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
Dec 30 11:13:48 maxwell kernel: FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
Dec 30 11:13:48 maxwell kernel: FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
Dec 30 11:13:48 maxwell kernel: FWMSG:   Shared window : base : 0x00800000 [static wiring]
Dec 30 11:13:48 maxwell kernel: FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
Dec 30 11:13:48 maxwell kernel: FWMSG:   Shared memory : base : 0x00800000
Dec 30 11:13:48 maxwell kernel: FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
Dec 30 11:13:48 maxwell kernel: FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
Dec 30 11:13:48 maxwell kernel: FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
Dec 30 11:13:48 maxwell kernel: FWMSG:   Heap free space : 10002626 [0x0098a0c2]
Dec 30 11:13:48 maxwell kernel: FWMSG:   Heap allocated space : 94656 [0x000171c0]
Dec 30 11:13:48 maxwell kernel: FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
Dec 30 11:13:48 maxwell kernel: FWMSG:   Application : adc [release]
Dec 30 11:13:48 maxwell kernel: FWMSG:   Linked on : Jul 25 2015 - 08:48:55
Dec 30 11:13:48 maxwell kernel: FWMSG:   Release : S2ISP-01.43.00
Dec 30 11:13:48 maxwell kernel: FWMSG:   H4ISPAPPLE : 11536
Dec 30 11:13:48 maxwell kernel: FWMSG:   H4ISPCD : 4081
Dec 30 11:13:48 maxwell kernel: FWMSG:   ffw : 4143
Dec 30 11:13:48 maxwell kernel: FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
Dec 30 11:13:48 maxwell kernel: FWMSG: 
Dec 30 11:13:48 maxwell kernel: FWMSG: -------------------------------------------------------------------------------
Dec 30 11:13:48 maxwell kernel: FWMSG: 
Dec 30 11:13:48 maxwell kernel: FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
Dec 30 11:13:48 maxwell kernel: FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
Dec 30 11:13:48 maxwell kernel: FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
Dec 30 11:13:48 maxwell kernel: facetimehd 0000:03:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
```

patjak, 2015-12-30:
Hi, are you thinking about that last line "Direct firmware load for facetimehd/1871_01XX.dat failed with error -2"? That error is not fatal so you don't need to worry about that now.

mmlb, 2015-12-30:
Indeed I though that `-2` was an **actual** error, but only because for some reason `/dev/video` was not being created. I just went and checked and its there and mplayer is happy, so I'll chock it up doing trying many things at once. Sorry for the noise.

closed 2015-12-30 by mmlb