patjak/facetimehd #40: Unloading bdc_pci should not be needed.

xgnata, 2016-01-07:
On my box, a simple modprobe facetimehd does not work.
One has to unload bdc_pci first and then load facetimehd.
I know that with behaviour is documented in the Additional Notes of the wiki.
However, I think it is a issue which should be on the "FIXME" list (as the driver is now streaming more or less like a charm).

hobarrera, 2016-01-07:
What the Arch package does, is blacklists `bdc_pci`. This is done by including a `/etc/modprobe.d/bcwc-pcie.conf` file. Its contents of the file are just one line: `blacklist bdc_pci`.

I'd suggest adding that upstream/here, but currently there's no installation script which would do this (current Makefile merely builds the package, as far as I understand).

Malo87, 2016-01-08:
I just drop two lines in a loadWebcamModule.sh (chmod a+x):

#!/bin/sh
sudo modprobe -r bdc_pci
sudo modprobe facetimehd

then 
$> ./loadWebcamModule.sh

wvengen, 2016-01-08:
Do we have a pci device id collision here? Looking at the type of chip, it seems to me that the facetimehd device is more abundant, and it may be useful to blacklist bdc_pci by default if this driver gets mainstream.

https://github.com/torvalds/linux/blob/master/drivers/usb/gadget/udc/bdc/Kconfig
http://cateee.net/lkddb/web-lkddb/USB_BDC_PCI.html

hobarrera, 2016-01-08:
> I just drop two lines in a loadWebcamModule.sh (chmod a+x):

There's no such script in the repository. Using `modprobe.d` sound like the standard approach too.

patjak, 2016-01-09:
bdc_pci needs to be more restrictive on grabbing the device so the change should be done there. Not much we can do since broadcom decided to reuse the pci id.

xgnata, 2016-01-09:
Any chance to contact Ashwini Pahuja (bdc_pci (maintainer) and ask him 
to do the change in his driver?
Maybe it is to early?

Xavier

wvengen, 2016-01-10:
I don't think it's too early - kernel changes may take some time to ripple through. I'll happily contact him.

xgnata, 2016-01-14:
You can do so telling him that
02:00.0 Multimedia controller [0480]: Broadcom Corporation 720p FaceTime HD Camera [14e4:1570]
has

xgnata, 2016-01-14:
I did contact him asking if he could use PCI_CLASS to resolve this conflict.

proninyaroslav, 2016-01-18:
@hobarrera 
I can't add it to the blacklist on Ubuntu 15.10 (he still loaded).

wvengen, 2016-01-18:
@proninyaroslav I suppose you edited `/etc/modprobe.d/blacklist.conf` and rebooted? You may need to run `update-initramfs` in between (depends).
http://askubuntu.com/questions/110341/how-to-blacklist-kernel-modules
https://help.ubuntu.com/community/Loadable_Modules#Blacklisting_Modules
https://wiki.debian.org/KernelModuleBlacklisting
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/227158

xgnata, 2016-01-18:
Another (better) way of solving the problem is to fix bdc_pci. I have
already sent a bug report on the linux-pci list. However, coming with a
patch would probably speedup the process quite a bit. Both bdc-pci and
facetimehd should not only look at the pci-id to decide to grab the device
or not. If facetimehd code is already looking at e.g. the pci-class then it
should be easy to copy this part of the code into a patch for bdc-pci
(assuming bdc-pci class is documented somewhere...). I'll try to find a bit
of time to look at that.

wvengen, 2016-01-18:
that's even better in the long run, great!

proninyaroslav, 2016-01-19:
Maybe off topic, but after update-initramfs touchpad stopped working (MacBook Air 2014). Repeated update-initramfs and reboot didn't help.

dcunited001, 2016-01-28:
@proninyaroslav for me, after i installed the `mtrack` touchpad drivers, then `facetimehd` does not autoload and instead bdc_pci does

xgnata, 2016-02-11:
I wrote a patch to fix bdc_pci  but so far no feedback from the author or from kernel maintainers.
I would help a lot if someone could find a doc proving that this USB controller driven by bdc_pci has a device class ID equal to 0x0c (like any other usb controller).
It seems that "nobody" it using bdc_pci for real.
Could someone find even only a doc showing what this controller (it is a Broadcom IP block) really is??? It looks like it exist on some ARM plateforms....but it is annoying to have a driver of a device "nobody" seems to care about grabbing incorrectly a pretty widely use webcam.
Anyone with better google skills that me??

patjak, 2016-02-11:
I'll get in contact with him and if he is unresponsive I'll just send out the patch to fix it. He's working for Broadcom so if they are sloppy with PCI ID assignments it's their fault and they will have to adjust. Thanks for looking at this.