patjak/facetimehd #41: Firmware extraction

budrik, 2016-01-09:
Hi

I'm not able to extract the firmware:
on 4.2.0-23-generic #28-Ubuntu SMP Sun Dec 27 17:47:31 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux, 

I tried the following
* when running: 
- "bcwc_pcie/firmware/./extract-firmware.sh" nothing happens.

--> Why nothing is downloaded?

- "make"  Dowloading the driver, please wait...
mv: cannot stat ‘System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface’: No such file or directory
Makefile:24: recipe for target 'AppleCameraInterface' failed
make: *** [AppleCameraInterface] Error 1

- ./extract-firmware.sh -x AppleCameraInterface
Error: Mismatching driver hash for ../../AppleCameraInterface (copied from mac OSX)
Error: The uknown hash is 387097b5133e980196ac51504a60ae1ad8bab736eb0070a55774925ca0194892
Error: No firmware extracted!

- ./extract-firmware.sh --dmg osxupd10.11.2.dmg (downloaded manually:https://support.apple.com/downloads/DL1849/en_US/osxupd10.11.2.dmg )
Error: 'pbzx' needed but not found!

How can I "download the driver (zip version) and extract it" as mentioned in the wiki???

Thanks for some help

hobarrera, 2016-01-09:
Maybe one of the dependencies is missing?  
I had to install cpio for it to work, and had results similar to yours before I did.

Maybe the script can be improved to check that everything is present.

mauritiusdadd, 2016-01-10:
@budrik :
As @hobarrera already said, it seems you miss one of the program needed for downloading the driver. Also,
> Error: 'pbzx' needed but not found!
if you want to extract the driver from the full `.dmg` file you need then you need more program and, as the error message says, one of them is `pbzx`. It seems, however, that `pbzx` does not work as expected (at least for me), so for now the only working method to extract the firmware is using "make"

@hobarrera : 
I agree with you, some dependency checking  must be done before performing the actual download.

budrik, 2016-01-10:
Thanks for the replies:
I was not able to find the needed packages  pbzx and xar to extract the driver from .dmg. But I managed to extract the firmware as follows:

1) Boot to OS X, El Capitan 10.11.2, get driver: */System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/*
2) Boot to e.g. ubuntu, cp *AppleCameraInterface* to *bcwc_pcie-master/firmware*
3) Modify *extract-firmware.sh* and add sha256sum of *AppleCameraInterface* because it was unknown in may case.
4) *make* will generate firmware.bin
5) add sha256sum of *firmware.bin* to *extract-firmware.sh* 
6) make install

I added the following lines to *extract-firmware.sh*:

hash_drv_osx_143_2='387097b5133e980196ac51504a60ae1ad8bab736eb0070a55774925ca0194892'

hash_fw_osx_143_2='504fcf1565bf10d61b31a12511226ae51991fb55d480f82de202a2f7ee9c966e'

After this, the installation worked right away ;-) 

Thanks a lot for the working camera!!!

closed 2016-01-10 by budrik

mauritiusdadd, 2016-01-10:
Hi @budrik, can you confirm that the firmware works fine? If it so I could add the unknown hashes to the extraction script, but I should still need to know if the firmware file is good... can you please save the following script as `check.sh` and post the output of `./check.sh firmware.bin`?
```bash
#!/bin/bash
#check.sh
fw_bytes_header="feffffeafeffffeafeffffeafeffffeafeffffeafeffffeafeffffea"
fw_bytes_footer="00000000ffffffff"

if ! which hexdump &> /dev/null; then
  err "You need to install 'hexdump' in order to check a firmware with unknown hash!"
  return 1
fi

header=$(hexdump -v -e '"" /1 "%02x"' "$1" -s 4 -n 28)
footer=$(hexdump -v -e '"" /1 "%02x"' "$1" | tail -c 16)

if [[ "${header}" != "${fw_bytes_header}" ]]; then
  echo "The extracted firmware does not seem good (wrong header)"
elif [[ "${footer}" != "${fw_bytes_footer}" ]]; then
  echo "The extracted firmware does not seem good (wrong footer)"
else
  echo "You're lucky, the firmware looks good, " \
       "but it could also not work... use it at your own risk!"
fi
```

budrik, 2016-01-16:
@mauritiusdadd, sorry the for the late response - the camera is working fine!

./check.sh firmware.bin:

You're lucky, the firmware looks good,  but it could also not work... use it at your own risk!

mauritiusdadd, 2016-01-17:
Ok, I'll update the extraction script, thank you!

patjak, 2016-01-17:
@mauritiusdadd @budrik Hmm, I wonder what the difference is between the firmwares. Do they really report as the same version? I don't like the idea of having multiple firmware versions floating around since that makes debugging much harder. On the other hand, if it becomes hard for people to get a hold of the correct version we might have no other choice.

Either way, I'll merge your PR. Thanks.

budrik, 2016-01-18:
@mauritiusdadd: Can I somehow help to clarify the issue?

mauritiusdadd, 2016-01-19:
@patjak : Ok, thank you.
@budrik: I don't know if the version is always in the same position in the firmware file, but can you try to run the following command and paste here its output?
```bash
echo "[$(hexdump -v -e '"" /1 "%01c"' firmware.bin -s 4224 -n 14)]"
```

budrik, 2016-01-23:
@mauritiusdadd. Here we go:
 
[S2ISP-01.40.00]

yaguphire, 2016-01-23:
macbook pro 13" early 2015

[S2ISP-01.43.00]

mauritiusdadd, 2016-01-24:
@yaguphire: thanks, that is the current working version of the firmware.
@budrik : So @patjak was right, that is a different version of the firmware.

chris-milsted, 2016-01-25:
I was just getting around to trying this on my MBP (Late 2014 15" model) and after updating OSX I have found I am running 10.11.3 and the hashes are wrong again.

I copied the System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface file onto a memory key and when I sha256sum this on Fedora 23 I get:

dfac86799c6cf0aceb59bb4e732be8f030e7943eb1146830c7136f62621c9853  ./AppleCameraInterface

So another new Hash.

I ran the following to try and extract the firmware, just using the old offset:

dd bs=1 skip=81920 if=./AppleCameraInterface of=firmware.bin

But when I run the check.sh script it fails:

$ ./check.sh firmware.bin 
The extracted firmware does not seem good (wrong header)

So I need to figure out how to work out the right offset, then I can add the new hash and the new offset as well to a pull request. 

Could somebody advise the best way to work out the new offset from the file please?

mauritiusdadd, 2016-01-26:
@chris-milsted: In this case, the best thing to do is to use the extraction script to download the correct driver version. First of all go into the `firmware` directory and then run `make`:
```bash
$ cd firmware/
$ make

Checking dependencies for driver download...
/usr/bin/curl
/usr/bin/xzcat
/usr/bin/cpio

Downloading the driver, please wait...

Found matching hash from OS X, El Capitan 10.11.2
==> Extracting firmware...
 --> Decompressing the firmware using gzip...
 --> Deleting temporary files...
 --> Extracted firmware version 1.43.0
```

chris-milsted, 2016-01-26:
@mauritiusdadd I think I did try this.

I coped the firmware into the ./firmware/ folder:

$ ls
AppleCameraInterface  extract-firmware.sh  Makefile

I then ran make:

$ make 

Error: Mismatching driver hash for AppleCameraInterface
Error: The unknown hash is dfac86799c6cf0aceb59bb4e732be8f030e7943eb1146830c7136f62621c9853
Error: No firmware extracted!
Makefile:21: recipe for target 'all' failed
make: *** [all] Error 1

Hence I just tried to dd out the firmware using the old offsets and found that the firmware may have changed. 

Wondering what the next steps would be?

Thanks

Chris

lhupfeldt, 2016-03-11:
Hi,

I facing the same? issue with firmware extraction:

$ git show
commit 8cc44d67f3c924f30a895dd44b9065ed04b4ed0b
Merge: bb3c229 2894ef6
Author: Patrik Jakobsson <patrik.r.jakobsson@gmail.com>
Date:   Tue Mar 1 15:22:59 2016 +0100

make

Checking dependencies for driver download...
/usr/bin/curl
/usr/bin/xzcat
/usr/bin/cpio

Downloading the driver, please wait...
mv: cannot stat ‘System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface’: No such file or directory
Makefile:24: recipe for target 'AppleCameraInterface' failed
make: *** [AppleCameraInterface] Error 1

Regards
Lars Hupfeldt

jeffgus, 2016-05-12:
Here is what I have. My system has all the updates as of May 12th. I copied the AppleCameraInterface file from the MacOS partition.

Product Name: MacBookPro11,3

$ dd bs=1 skip=81920 if=./AppleCameraInterface |gunzip -c > firmware.bin
gzip: stdin: decompression OK, trailing garbage ignored

a23d1ad35cb33c4a0d96a78aaec20e5660b4b7c9071a6fd04eabba99e9a8b14f  AppleCameraInterface
e3e6034a67dfdaa27672dd547698bbc5b33f47f1fc7f5572a2fb68ea09d32d3d  firmware.bin

ls -l firmware.bin
1425412  firmware.bin

pfernandez, 2016-10-02:
I ran into this problem today with a fresh install of Ubuntu 16.04-- which doesn't include curl! After running `sudo apt-get install curl` I could run `make` successfully.

NickitaX, 2019-02-25:
@pfernandez you rock mate!