patjak/facetimehd #43: The driver doesn't work on Macbook Air mid 2015

matunixe, 2016-01-10:
Dear sir,

I followed the documentation on the wiki but the driver doesn't work.
I'm running Archlinux on the Macbook and I tryied both ethods (AUR packages and manual install from github).

```
[m_b@Macbook-Air ~]$ sudo modprobe -r bdc_pci
[m_b@Macbook-Air ~]$ sudo modprobe facetimehd
[m_b@Macbook-Air ~]$ mpv tv://
Playing: tv://
[tv] Selected driver: v4l2
[tv]  name: Video 4 Linux 2 input
[tv] unable to open '/dev/video0': Permission denied
[tv] ioctl set mute failed: Bad file descriptor
[tv] 0 frames successfully processed, 0 frames dropped.
Failed to recognize file format.

Exiting... (Errors when loading file)
```

Hope this help.

hobarrera, 2016-01-10:
I'm in the `video` group, dunno if that's legacy or required to access the webcam.

You don't need to unload `bdc_pci` on arch, since the AUR package will blacklist it for you anyway (though that applies after a reboot, AFAIK).

matunixe, 2016-01-10:
Thanks for the info mate, I add the user to the video group as you say:
``gpasswd -a m_b video``

I decided to purge and destroy unused files downloaded from Github and use only AUR package. I triyed to restart but nothing change.

daaelar, 2016-01-22:
Bumping this with some extra info: 
[  572.445344] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[  572.445487] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[  572.450571] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[  572.450598] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[  572.460620] facetimehd 0000:02:00.0: PLL reset finished
[  572.460625] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[  572.724074] facetimehd 0000:02:00.0: Failed to lock S2 PLL: 0xc902c902
[  572.744145] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[  572.744158] facetimehd 0000:02:00.0: STRAP valid
[  572.744159] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[  572.744166] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[  572.744175] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[  572.744185] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[  572.744186] facetimehd 0000:02:00.0: Using step size 158
[  572.744191] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011c
[  572.744196] facetimehd 0000:02:00.0: Virtual VTT enabled
[  572.764735] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[  572.764748] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[  572.764940] facetimehd 0000:02:00.0: Full memory verification failed! (65535)
[  573.238163] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[  583.461226] facetimehd 0000:02:00.0: Init failed! No wake signal
[  583.461345] facetimehd: probe of 0000:02:00.0 failed with error -5

The "Full memory verification failed!" line is highlighted in my dmesg output. Also running Arch on the 2015 MacBookAir 13", running 4.3.3-3-ARCH with the bcwc-pcie-git and firmware packages from AUR.

lspci shows the following:
02:00.0 Multimedia controller: Broadcom Corporation 720p FaceTime HD Camera
	Subsystem: Broadcom Corporation 720p FaceTime HD Camera
	Control: I/O- Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Interrupt: pin A routed to IRQ 0
	Region 0: Memory at c1500000 (64-bit, non-prefetchable) [size=64K]
	Region 2: Memory at a0000000 (64-bit, prefetchable) [size=256M]
	Region 4: Memory at c1400000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: <access denied>
	Kernel modules: bdc_pci

Which is curious, since the AUR package should be blacklisting bdc_pci (and indeed, it's not loaded on my system). I attempted unloading it and facteimehd and reloading facetimehd to the same errors above.  If there's more you want, I'd be happy to provide info.

patjak, 2016-01-22:
Don't know why bdc_pci got hold of the device. Did you try ```modprobe -r bdc_pci``` before loading the facetimehd module? In general, if you have the "Full memory verification failed! (xxxxx)" error you're unfortunately out of luck atm. It means the ISP memory cannot be reliably read / written by the host and we cannot upload the firmware. The ISP can probably not access the memory either. What needs to be done is a full calibration of the DDR PHY interface. I have already figured out much of the code for calibrating but for some reason it's not working. The only fix I can see right now is to finish the implementation of the DDR PHY calibration.

daaelar, 2016-01-23:
Yes, I ran ```modprobe -r bdc_pci``` before loading the module.  I have no idea how it was listed under lspci.  I confirmed it was absent from ```lsmod | grep bdc``` before loading bcwc, but I still had the memory error.  I understand this is a work in progress, and if there's anything I can do to help, just ask.  If you have anything you want to run by me, I'm willing to help test/provide debug info. It's a MacBookAir7,2 running the I7-5650U and I keep up to date with pacman and the AUR. Thanks for putting the effort into this!

svartalf, 2016-02-10:
I'd tried to install latest AUR package for my air2015 and webcam is working!

Installed packages:
```
aur/bcwc-pcie-firmware 1.43.0-1 [installed] (3)
    Firmware for the Broadcom 1570 PCIe webcam.
aur/bcwc-pcie-git latest-1 [installed: r213.bb3c229-1] (3)
    Reverse engineered Linux driver for the Broadcom 1570 PCIe webcam.
```

and dmesg log:
```
[    1.506911] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[    1.507071] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[    1.514513] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[    1.514570] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[    1.524598] facetimehd 0000:02:00.0: PLL reset finished
[    1.524601] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[    1.524614] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[    1.534660] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[    1.554766] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[    1.554780] facetimehd 0000:02:00.0: STRAP valid
[    1.554782] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[    1.554789] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[    1.554798] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[    1.554835] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[    1.554836] facetimehd 0000:02:00.0: Using step size 156
[    1.554841] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011c
[    1.554846] facetimehd 0000:02:00.0: Virtual VTT enabled
[    1.575448] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[    1.575462] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[    1.575764] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[    2.074002] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[    2.225562] facetimehd 0000:02:00.0: ISP woke up after 0ms
[    2.225587] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[    2.225592] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[    2.235629] facetimehd 0000:02:00.0: ISP second int after 0ms
[    2.235635] facetimehd 0000:02:00.0: Channel description table at 00800000
[    2.245823] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[    2.245826] facetimehd 0000:02:00.0: Enabling interrupts
[    2.320948] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
```

It is still talking something about a firmware loading, but `/dev/video0` was created and skype even can show video stream from the cam.

daaelar, 2016-02-15:
Confirmed working on my system as well, now Also geeting a firmware load error, even though it's functioning.
`[    2.046801] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
`

alexhornbake, 2016-08-12:
Confirmed Working! Thanks @patjak for the driver!! Just dropping some notes here in case they are helpful.

I'm running:
macbook air 7,2
Debian Jessie (kernel is 4.6.0-0.bpo.1-amd64)
built bcwc_pcie from master 48f9854e5d18ec360f36d9242a575ad072dedab9

I'm getting the same error when loading the module:

```
[  288.374802] facetimehd 0000:02:00.0: firmware: failed to load facetimehd/1871_01XX.dat (-2)
[  288.374811] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
```

I first attempted on 4.2.0-0.bpo.1-amd64 kernel with no success.
- Loading the module produced the same error message as above.
- Attempting to access the camera resulted in the system locking up and becoming unresponsive. 

logs from 4.2 kernel crash when initializing camera:

```
[ 1343.440545] CPU: 1 PID: 9000 Comm: mplayer Tainted: P        W  O    4.2.0-0.bpo.1-amd64 #1 Debian 4.2.5-1~bpo8+1
[ 1343.440634] Hardware name: Apple Inc. MacBookAir7,2/Mac-937CB26E2E02BB01, BIOS MBA71.88Z.0166.B08.1509021746 09/02/2015
[ 1343.440728] task: ffff880240868080 ti: ffff880217c38000 task.ti: ffff880217c38000
[ 1343.440793] RIP: 0010:[<ffffffffa0f5bc2c>]  [<ffffffffa0f5bc2c>] vb2_dma_sg_alloc+0x18c/0x300 [videobuf2_dma_sg]
[ 1343.440888] RSP: 0018:ffff880217c3bba8  EFLAGS: 00010246
[ 1343.440935] RAX: 0000000000096000 RBX: ffff88024e3cd040 RCX: 0000000000000000
[ 1343.440997] RDX: 0000160000000000 RSI: 0000000000000001 RDI: 0000000000000002
[ 1343.441059] RBP: 00000000000082d0 R08: 0000000000000001 R09: 0000000000001000
[ 1343.441121] R10: 000000000001ce00 R11: 0000000000000001 R12: 0000000000000000
[ 1343.441182] R13: 00000000ffffffff R14: 0000000000000001 R15: 0000000000000000
[ 1343.441246] FS:  00007fb449f5b940(0000) GS:ffff88026ec40000(0000) knlGS:0000000000000000
[ 1343.441316] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1343.441366] CR2: 0000000000096000 CR3: 000000022c153000 CR4: 00000000003406e0
[ 1343.441428] Stack:
[ 1343.441449]  ffff880240868080 ffffea0001e148c0 0000000000001000 0000000000096000
[ 1343.441525]  00000010000280da 0000000000000020 0000000000000001 ffff88006e0f25c0
[ 1343.441601]  ffff88006e0cfc5c ffff88006e0cfe88 ffff88006e0cfc00 ffff88006e0f2604
[ 1343.441676] Call Trace:
[ 1343.441709]  [<ffffffffa0f47889>] ? __vb2_queue_alloc+0x159/0x530 [videobuf2_core]
[ 1343.441781]  [<ffffffffa0f48165>] ? __reqbufs.isra.16+0x165/0x340 [videobuf2_core]
[ 1343.441852]  [<ffffffffa0f485f1>] ? vb2_ioctl_reqbufs+0x61/0x90 [videobuf2_core]
[ 1343.441925]  [<ffffffffa0f0f75e>] ? __video_do_ioctl+0x26e/0x300 [videodev]
[ 1343.441995]  [<ffffffffa0f0f4f0>] ? video_ioctl2+0x20/0x20 [videodev]
[ 1343.442059]  [<ffffffffa0f0f2e4>] ? video_usercopy+0x304/0x4f0 [videodev]
[ 1343.442124]  [<ffffffff8138daaa>] ? n_tty_write+0x27a/0x4b0
[ 1343.442176]  [<ffffffff8109e4de>] ? update_curr+0xbe/0xf0
[ 1343.442225]  [<ffffffff8109e857>] ? set_next_entity+0x67/0x460
[ 1343.442283]  [<ffffffffa0f0a641>] ? v4l2_ioctl+0xc1/0xd0 [videodev]
[ 1343.442344]  [<ffffffff811d26c3>] ? do_vfs_ioctl+0x2c3/0x4a0
[ 1343.442399]  [<ffffffff8155468f>] ? __schedule+0x28f/0x8b0
[ 1343.445911]  [<ffffffff811d2916>] ? SyS_ioctl+0x76/0x90
[ 1343.449113]  [<ffffffff815586f2>] ? system_call_fast_compare_end+0xc/0x6b
[ 1343.452286] Code: ff 48 8b 73 10 8b 53 40 48 8b 7b 30 31 c9 4c 8b 44 24 10 41 b9 d0 00 00 00 e8 71 d1 37 e0 85 c0 0f 85 33 01 00 00 48 8b 44 24 18 <48> 8b 38 e8 6c d5 47 e0 49 89 c4 48 89 03 48 8b 43 20 4d 85 e4 
[ 1343.459056] RIP  [<ffffffffa0f5bc2c>] vb2_dma_sg_alloc+0x18c/0x300 [videobuf2_dma_sg]
[ 1343.462421]  RSP <ffff880217c3bba8>
[ 1343.465702] CR2: 0000000000096000
[ 1343.483176] ---[ end trace 0c46b0db1140710a ]---
```

matunixe, 2017-01-11:
I don't face this issue anymore. Outdated post with maybe irrelevant infos. I'm closing it.

closed 2017-01-11 by matunixe