patjak/facetimehd #44: failed with error -5 on Ubuntu 15.10 (4.2.0-23-generic) on MBP11,1

ahachete, 2016-01-10:
Hi. I have tried to get the camera working but module seems not to load correctly. I have tried the unloading of bdc_pci before. The error reported by dmesg is (full output copied):

[ 3055.280538] Linux video capture interface: v2.00
[ 3055.287336] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[ 3055.287609] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[ 3055.292989] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[ 3055.293018] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[ 3055.302980] facetimehd 0000:02:00.0: PLL reset finished
[ 3055.302985] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[ 3055.557419] facetimehd 0000:02:00.0: Failed to lock S2 PLL: 0xc902c902
[ 3055.577366] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[ 3055.577380] facetimehd 0000:02:00.0: STRAP valid
[ 3055.577381] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[ 3055.577388] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[ 3055.577397] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[ 3055.577408] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[ 3055.577409] facetimehd 0000:02:00.0: Using step size 154
[ 3055.577414] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[ 3055.577419] facetimehd 0000:02:00.0: Virtual VTT enabled
[ 3055.597832] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[ 3055.597845] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[ 3055.598041] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[ 3056.055456] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[ 3066.052140] facetimehd 0000:02:00.0: Init failed! No wake signal
[ 3066.052234] facetimehd: probe of 0000:02:00.0 failed with error -5

Extra info:

$ uname -a             
Linux ushuaia 4.2.0-23-generic #28-Ubuntu SMP Sun Dec 27 17:47:31 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

$ sudo dmidecode -s system-product-name
MacBookPro11,1
Invalid entry length (0). DMI table is broken! Stop.

$ md5sum /lib/firmware/facetimehd/firmware.bin
4e1d11e205e5c55d128efa0029b268fe  /lib/firmware/facetimehd/firmware.bin

Current commit: dc55cf3 

Any hints would be appreciated.

ahachete, 2016-01-10:
I rebooted the system and tried again. Got a different output in dmesg:

[   92.745594] Linux video capture interface: v2.00
[   92.759754] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[   92.759891] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[   92.766601] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[   92.766638] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[   92.776595] facetimehd 0000:02:00.0: PLL reset finished
[   92.776599] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[   92.776612] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[   92.786575] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[   92.806511] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[   92.806528] facetimehd 0000:02:00.0: STRAP valid
[   92.806530] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[   92.806538] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[   92.806547] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 1 us
[   92.806558] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[   92.806560] facetimehd 0000:02:00.0: Using step size 154
[   92.806566] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[   92.806572] facetimehd 0000:02:00.0: Virtual VTT enabled
[   92.826987] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[   92.827003] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[   92.827202] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[   93.290473] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[   93.336230] facetimehd 0000:02:00.0: ISP woke up after 0ms
[   93.336237] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[   93.336241] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[   93.346199] facetimehd 0000:02:00.0: ISP second int after 0ms
[   93.346203] facetimehd 0000:02:00.0: Channel description table at 00800000
[   93.356316] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[   93.356321] facetimehd 0000:02:00.0: Enabling interrupts
[   93.356440] FWMSG: 
[   93.356463] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[   93.356473] FWMSG: 
[   93.356507] FWMSG:   Restart count : 0
[   93.356568] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[   93.356583] FWMSG:   System Clock : 200000000 Hz
[   93.356598] FWMSG:   Processor mode : SUPERVISOR
[   93.356614] FWMSG:   Cache architecture type : SEPARATE
[   93.356651] FWMSG:   Cache type : WRITEBACK
[   93.356664] FWMSG:   Data Cache Line : 32 [0x20]
[   93.356678] FWMSG:   Boot arguments entries : 0
[   93.356743] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[   93.356758] FWMSG:   Main memory :   base : 0x00000000
[   93.356792] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[   93.356810] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[   93.356881] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[   93.356898] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[   93.356962] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[   93.356977] FWMSG:   Shared memory : base : 0x00800000
[   93.357035] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[   93.357106] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[   93.357147] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[   93.357163] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[   93.357226] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[   93.357244] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[   93.357257] FWMSG:   Application : adc [release]
[   93.357271] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[   93.357285] FWMSG:   Release : S2ISP-01.43.00
[   93.357319] FWMSG:   H4ISPAPPLE : 11536
[   93.357331] FWMSG:   H4ISPCD : 4081
[   93.357342] FWMSG:   ffw : 4143
[   93.357390] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[   93.357400] FWMSG: 
[   93.357439] FWMSG: -------------------------------------------------------------------------------
[   93.357449] FWMSG: 
[   93.430702] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[   93.430843] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[   93.431073] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[   93.431400] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
[... 53 lines cut ...]
[  170.654628] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[  170.654648] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[  170.654756] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[  170.654772] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[  170.654790] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[  170.654803] FWMSG:   Application : adc [release]
[  170.654818] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[  170.654831] FWMSG:   Release : S2ISP-01.43.00
[  170.654842] FWMSG:   H4ISPAPPLE : 11536
[  170.654854] FWMSG:   H4ISPCD : 4081
[  170.654865] FWMSG:   ffw : 4143
[  170.654972] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[  170.654982] FWMSG: 
[  170.655002] FWMSG: -------------------------------------------------------------------------------
[... cut ...]
[  171.186658] FWMSG: [ISP] CH = 0 CMD = 0x0111 [CISP_CMD_CH_SET_FILE_LOAD]
[  171.186686] FWMSG: ERR: ./H4ISPCD/common/controller/target/CDSControllerBaseChannelCmd.cpp, 3794: [DSC] CMD = 0xfffe: UNKNOWN COMMAND

[  171.186823] FWMSG: ERR: ./H4ISPCD/common/controller/target/CDSControllerBase.cpp, 2032: Cmd 0xfffe failed
[  171.186848] FWMSG: ERR: ./ffw/ffw/CFSM.cpp, 1458: ISPFSMCh00: ACTION FAILED (4, <null>): [2, <null>]->[2, <null>]
[  171.186891] facetimehd 0000:02:00.0: fthd_hw_deinit
[  171.186936] facetimehd: probe of 0000:02:00.0 failed with error -5

patjak, 2016-01-10:
Hi,
You don't need the 1871_01XX.dat file. Don't link it to the ISP firmware. I've updated the wiki to be more clear on this. Something looks odd with your first try of loading the driver. Try rebooting your system and load it again. Does it still say "Init failed! No wake signal" when you do that?

ahachete, 2016-01-10:
Nope :(

Tried with 4.3.3 too:

[  170.136044] media: Linux media interface: v0.10
[  170.142712] Linux video capture interface: v2.00
[  170.149998] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[  170.150130] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[  170.155770] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[  170.155827] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[  170.165782] facetimehd 0000:02:00.0: PLL reset finished
[  170.165785] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[  170.165797] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[  170.175760] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[  170.195692] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[  170.195706] facetimehd 0000:02:00.0: STRAP valid
[  170.195707] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[  170.195714] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[  170.195723] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[  170.195734] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[  170.195735] facetimehd 0000:02:00.0: Using step size 154
[  170.195740] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[  170.195744] facetimehd 0000:02:00.0: Virtual VTT enabled
[  170.216158] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[  170.216172] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[  170.216367] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[  170.677313] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[  170.722771] facetimehd 0000:02:00.0: ISP woke up after 0ms
[  170.722778] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[  170.722781] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[  170.732741] facetimehd 0000:02:00.0: ISP second int after 0ms
[  170.732743] facetimehd 0000:02:00.0: Channel description table at 00800000
[  170.742854] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[  170.742856] facetimehd 0000:02:00.0: Enabling interrupts
[  170.743037] FWMSG: 
[  170.743059] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[  170.743069] FWMSG: 
[  170.743081] FWMSG:   Restart count : 0
[  170.743101] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[  170.743114] FWMSG:   System Clock : 200000000 Hz
[  170.743150] FWMSG:   Processor mode : SUPERVISOR
[  170.743173] FWMSG:   Cache architecture type : SEPARATE
[  170.743201] FWMSG:   Cache type : WRITEBACK
[  170.743214] FWMSG:   Data Cache Line : 32 [0x20]
[  170.743228] FWMSG:   Boot arguments entries : 0
[  170.743315] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[  170.743330] FWMSG:   Main memory :   base : 0x00000000
[  170.743347] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[  170.743363] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[  170.743458] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[  170.743475] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[  170.743493] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[  170.743507] FWMSG:   Shared memory : base : 0x00800000
[  170.743635] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[  170.743655] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[  170.743676] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[  170.743771] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[  170.743787] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[  170.743805] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[  170.743818] FWMSG:   Application : adc [release]
[  170.743832] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[  170.743845] FWMSG:   Release : S2ISP-01.43.00
[  170.743857] FWMSG:   H4ISPAPPLE : 11536
[  170.743869] FWMSG:   H4ISPCD : 4081
[  170.743880] FWMSG:   ffw : 4143
[  170.744006] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[  170.744017] FWMSG: 
[  170.744037] FWMSG: -------------------------------------------------------------------------------
[  170.744047] FWMSG: 
[  170.817380] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[  170.817630] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[  170.817962] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[  170.818269] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2

$ uname -a
Linux ushuaia 4.3.3-040303-generic #201512150130 SMP Tue Dec 15 06:32:30 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

patjak, 2016-01-10:
Everything looks fine in the dmesg above. Are you sure it's not working?

ahachete, 2016-01-10:
Dammit! You're right, the device is now created and working like a charm! I'm pretty sure it wasn't working for 4.2 kernel, but it definitely works now on 4.3.3. Thank you so much!

closed 2016-01-10 by ahachete

patjak, 2016-01-11:
There was in fact something odd going on with your first dmesg. "Init wake" shouldn't fail. Either way, it works now so no point spending too much time on it atm. Thanks.