patjak/facetimehd #48: early 2015 MBP 12,1 not working

fordodone, 2016-01-19:
ubuntu 15.10 is installed directly onto laptop (NOT a VM)

dmesg: 
[Tue Jan 19 15:56:47 2016] facetimehd 0000:02:00.0: Init failed! No wake signal
[Tue Jan 19 15:56:47 2016] facetimehd: probe of 0000:02:00.0 failed with error -5

$ lsmod |grep face
facetimehd             98304  0
videobuf2_dma_sg       20480  1 facetimehd
videobuf2_core         49152  1 facetimehd
videodev              172032  3 v4l2_common,facetimehd,videobuf2_core

/dev/video not created even after rmmod bdc_pci and modprobe facetimehd

gchiesa78, 2016-01-20:
this is a duplicate of [Module init/probe failure on MBP12,1 & Linux 4.4](https://github.com/patjak/bcwc_pcie/issues/35)

fordodone, 2016-01-21:
duplicates #35

closed 2016-01-21 by fordodone