patjak/facetimehd #49: Does not create /dev/video on MacBookPro11,4

rvangsgaard, 2016-01-21:
All commands from the Ubuntu/Debian guide runs without any errors, but /dev/video is not created.

I did unload `modprobe -r bdc_pci` before `modprobe facetimehd`.

```
Playing tv://.
Detected file format: TV
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: unable to open '/dev/video0': No such file or directory
v4l2: ioctl set mute failed: Bad file descriptor
v4l2: 0 frames successfully processed, 0 frames dropped.
Opening as detected format "TV" failed.
Failed to recognize file format.
```

proninyaroslav, 2016-01-21:
https://github.com/patjak/bcwc_pcie/wiki/Get-Started
>In some scenarios, you'll have to unload bdc_pci before inserting the kernel module, or /dev/video >won't be created. Do this with modprobe -r bdc_pci.

t1msammut, 2016-02-04:
I have a feeling the last line of my dmesg is an indicator of failure... This is on a 11,5 with 4.3.5 from gentoo.

[  148.314654] Linux video capture interface: v2.00
[  148.316538] facetimehd 0000:05:00.0: Found FaceTime HD camera with device id: 1570
[  148.316780] facetimehd 0000:05:00.0: Setting 64bit DMA mask
[  148.322527] facetimehd 0000:05:00.0: S2 PCIe link init succeeded
[  148.322557] facetimehd 0000:05:00.0: Refclk: 25MHz (0xa)
[  148.332601] facetimehd 0000:05:00.0: PLL reset finished
[  148.332603] facetimehd 0000:05:00.0: Waiting for S2 PLL to lock at 450 MHz
[  148.332616] facetimehd 0000:05:00.0: S2 PLL is locked after 10 us
[  148.342673] facetimehd 0000:05:00.0: S2 PLL is in bypass mode
[  148.362786] facetimehd 0000:05:00.0: DDR40 PHY PLL locked on safe settings
[  148.362801] facetimehd 0000:05:00.0: STRAP valid
[  148.362802] facetimehd 0000:05:00.0: Configuring DDR PLLs for 450 MHz
[  148.362810] facetimehd 0000:05:00.0: DDR40 PLL is locked after 0 us
[  148.362821] facetimehd 0000:05:00.0: First DDR40 VDL calibration completed after 2 us
[  148.362831] facetimehd 0000:05:00.0: Second DDR40 VDL calibration completed after 1 us
[  148.362832] facetimehd 0000:05:00.0: Using step size 140
[  148.362838] facetimehd 0000:05:00.0: VDL set to: coarse=0x10007, fine=0x10119
[  148.362843] facetimehd 0000:05:00.0: Virtual VTT enabled
[  148.383445] facetimehd 0000:05:00.0: S2 DRAM memory address: 0x22159559
[  148.383460] facetimehd 0000:05:00.0: Rewrite DDR mode registers succeeded
[  148.383664] facetimehd 0000:05:00.0: Full memory verification succeeded! (0)
[  148.847854] facetimehd 0000:05:00.0: Loaded firmware, size: 1392kb
[  148.989048] facetimehd 0000:05:00.0: ISP woke up after 0ms
[  148.989055] facetimehd 0000:05:00.0: Number of IPC channels: 7, queue size: 44865
[  148.989121] facetimehd 0000:05:00.0: Firmware requested heap size: 3072kb
[  148.999194] facetimehd 0000:05:00.0: ISP second int after 0ms
[  148.999198] facetimehd 0000:05:00.0: Channel description table at 00800000
[  149.010013] facetimehd 0000:05:00.0: magic value: 00000000 after 0 ms
[  149.010014] facetimehd 0000:05:00.0: Enabling interrupts
[  149.010211] FWMSG: 
[  149.010319] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[  149.010331] FWMSG: 
[  149.010366] FWMSG:   Restart count : 0
[  149.010459] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[  149.010506] FWMSG:   System Clock : 200000000 Hz
[  149.010553] FWMSG:   Processor mode : SUPERVISOR
[  149.010609] FWMSG:   Cache architecture type : SEPARATE
[  149.010649] FWMSG:   Cache type : WRITEBACK
[  149.010698] FWMSG:   Data Cache Line : 32 [0x20]
[  149.010749] FWMSG:   Boot arguments entries : 1
[  149.010785] FWMSG:     0000: 0xffff8804
[  149.010866] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[  149.010921] FWMSG:   Main memory :   base : 0x00000000
[  149.010999] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[  149.011076] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[  149.011155] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[  149.011229] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[  149.011312] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[  149.011366] FWMSG:   Shared memory : base : 0x00800000
[  149.011448] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[  149.011547] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[  149.011639] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[  149.011701] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[  149.011769] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[  149.011855] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[  149.011902] FWMSG:   Application : adc [release]
[  149.011957] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[  149.012000] FWMSG:   Release : S2ISP-01.43.00
[  149.012036] FWMSG:   H4ISPAPPLE : 11536
[  149.012067] FWMSG:   H4ISPCD : 4081
[  149.012094] FWMSG:   ffw : 4143
[  149.012165] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[  149.012177] FWMSG: 
[  149.012285] FWMSG: -------------------------------------------------------------------------------
[  149.012296] FWMSG: 
[  149.085704] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[  149.085928] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[  149.086034] CAMINFO 00000000: 80 f0 fa 02 02 00 00 00 02 00 00 00 00 00 00 00  ................
[  149.086035] CAMINFO 00000010: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  149.086249] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[  149.086509] CHINFO 00000000: 00 00 00 00 01 00 df 07 2b 00 05 00 07 00 03 00  ........+.......
[  149.086512] CHINFO 00000010: 01 00 00 00 01 00 00 00 70 97 00 00 03 00 00 00  ........p.......
[  149.086514] CHINFO 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  149.086515] CHINFO 00000030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[  149.086516] CHINFO 00000040: 01 00 00 00 ff ff 01 00 00 00 ff ff 00 00 00 00  ................
[  149.086517] CHINFO 00000050: 04 00 00 00 0a 00 00 00 02 00 00 00 01 00 00 00  ................
[  149.086518] CHINFO 00000060: 80 37 00 00 40 00 00 00 01 00 00 00 03 00 00 00  .7..@...........
[  149.086519] CHINFO 00000070: 00 10 00 00 40 00 00 00 01 00 00 00 00 00 00 00  ....@...........
[  149.086520] CHINFO 00000080: 00 00 00 00 34 08 01 00 40 0d 03 00 43 43 32 35  ....4...@...CC25
[  149.086521] CHINFO 00000090: 31 39 35 31 51 48 41 47 31 48 4e 41 32 00        1951QHAG1HNA2.
[  149.086544] facetimehd 0000:05:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
[  149.087016] FWMSG: [ISP] CMD = 0x0001 [CISP_CMD_STOP]
[  149.097126] facetimehd 0000:05:00.0: fthd_hw_deinit

t1msammut, 2016-02-25:
In case it helps... I am experiencing this with today's driver and 4.4.2 from gentoo. 

[ 1039.481866] facetimehd 0000:05:00.0: Found FaceTime HD camera with device id: 1570
[ 1039.482177] facetimehd 0000:05:00.0: Setting 64bit DMA mask
[ 1039.487815] facetimehd 0000:05:00.0: S2 PCIe link init succeeded
[ 1039.487846] facetimehd 0000:05:00.0: Refclk: 25MHz (0xa)
[ 1039.497880] facetimehd 0000:05:00.0: PLL reset finished
[ 1039.497884] facetimehd 0000:05:00.0: Waiting for S2 PLL to lock at 450 MHz
[ 1039.810733] facetimehd 0000:05:00.0: Failed to lock S2 PLL: 0xc902c902
[ 1039.830848] facetimehd 0000:05:00.0: DDR40 PHY PLL locked on safe settings
[ 1039.830863] facetimehd 0000:05:00.0: STRAP valid
[ 1039.830865] facetimehd 0000:05:00.0: Configuring DDR PLLs for 450 MHz
[ 1039.830873] facetimehd 0000:05:00.0: DDR40 PLL is locked after 0 us
[ 1039.830884] facetimehd 0000:05:00.0: First DDR40 VDL calibration completed after 2 us
[ 1039.830893] facetimehd 0000:05:00.0: Second DDR40 VDL calibration completed after 1 us
[ 1039.830894] facetimehd 0000:05:00.0: Using step size 140
[ 1039.830900] facetimehd 0000:05:00.0: VDL set to: coarse=0x10007, fine=0x10119
[ 1039.830905] facetimehd 0000:05:00.0: Virtual VTT enabled
[ 1039.851487] facetimehd 0000:05:00.0: S2 DRAM memory address: 0x22159559
[ 1039.851503] facetimehd 0000:05:00.0: Rewrite DDR mode registers succeeded
[ 1039.851707] facetimehd 0000:05:00.0: Full memory verification succeeded! (0)
[ 1040.308572] facetimehd 0000:05:00.0: Loaded firmware, size: 1392kb
[ 1040.449424] facetimehd 0000:05:00.0: ISP woke up after 0ms
[ 1040.449431] facetimehd 0000:05:00.0: Number of IPC channels: 7, queue size: 44865
[ 1040.449435] facetimehd 0000:05:00.0: Firmware requested heap size: 3072kb
[ 1040.459477] facetimehd 0000:05:00.0: ISP second int after 0ms
[ 1040.459481] facetimehd 0000:05:00.0: Channel description table at 00800000
[ 1040.470275] facetimehd 0000:05:00.0: magic value: 00000000 after 0 ms
[ 1040.470276] facetimehd 0000:05:00.0: Enabling interrupts
[ 1040.470465] FWMSG: 
[ 1040.470573] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[ 1040.470586] FWMSG: 
[ 1040.470620] FWMSG:   Restart count : 0
[ 1040.470713] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[ 1040.470760] FWMSG:   System Clock : 200000000 Hz
[ 1040.470806] FWMSG:   Processor mode : SUPERVISOR
[ 1040.470862] FWMSG:   Cache architecture type : SEPARATE
[ 1040.470903] FWMSG:   Cache type : WRITEBACK
[ 1040.470951] FWMSG:   Data Cache Line : 32 [0x20]
[ 1040.470996] FWMSG:   Boot arguments entries : 2
[ 1040.471047] FWMSG:     0000: 0x00000000 0x00000000
[ 1040.471159] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[ 1040.471215] FWMSG:   Main memory :   base : 0x00000000
[ 1040.471294] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[ 1040.471375] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[ 1040.471454] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[ 1040.471532] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[ 1040.471615] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[ 1040.471669] FWMSG:   Shared memory : base : 0x00800000
[ 1040.471752] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[ 1040.471851] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[ 1040.471943] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[ 1040.472005] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[ 1040.472069] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[ 1040.472161] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[ 1040.472208] FWMSG:   Application : adc [release]
[ 1040.472266] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[ 1040.472312] FWMSG:   Release : S2ISP-01.43.00
[ 1040.472348] FWMSG:   H4ISPAPPLE : 11536
[ 1040.472380] FWMSG:   H4ISPCD : 4081
[ 1040.472406] FWMSG:   ffw : 4143
[ 1040.472480] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[ 1040.472494] FWMSG: 
[ 1040.472606] FWMSG: -------------------------------------------------------------------------------
[ 1040.472617] FWMSG: 
[ 1040.541896] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[ 1040.542235] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[ 1040.542335] CAMINFO 00000000: 80 f0 fa 02 02 00 00 00 02 00 00 00 00 00 00 00  ................
[ 1040.542337] CAMINFO 00000010: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1040.542566] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[ 1040.542822] CHINFO 00000000: 00 00 00 00 01 00 df 07 2b 00 05 00 07 00 03 00  ........+.......
[ 1040.542825] CHINFO 00000010: 01 00 00 00 01 00 00 00 70 97 00 00 03 00 00 00  ........p.......
[ 1040.542826] CHINFO 00000020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1040.542827] CHINFO 00000030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[ 1040.542828] CHINFO 00000040: 01 00 00 00 ff ff 01 00 00 00 ff ff 00 00 00 00  ................
[ 1040.542829] CHINFO 00000050: 04 00 00 00 0a 00 00 00 02 00 00 00 01 00 00 00  ................
[ 1040.542830] CHINFO 00000060: 80 37 00 00 40 00 00 00 01 00 00 00 03 00 00 00  .7..@...........
[ 1040.542831] CHINFO 00000070: 00 10 00 00 40 00 00 00 01 00 00 00 00 00 00 00  ....@...........
[ 1040.542832] CHINFO 00000080: 00 00 00 00 34 08 01 00 40 0d 03 00 43 43 32 35  ....4...@...CC25
[ 1040.542833] CHINFO 00000090: 31 39 35 31 51 48 41 47 31 48 4e 41 32 00        1951QHAG1HNA2.
[ 1040.542848] facetimehd 0000:05:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
[ 1040.543535] FWMSG: [ISP] CMD = 0x0001 [CISP_CMD_STOP]
[ 1040.553638] facetimehd 0000:05:00.0: fthd_hw_deinit

salboa, 2016-03-02:
Actually I'm experiencing the same issue on debian (testing) with kernel 4.3.0.
Installation procedure worked very well apart for module loading:
on dmesg I could see error -2

[ 5846.049478] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[ 5846.049651] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[ 5846.054903] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[ 5846.054936] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[ 5846.064881] facetimehd 0000:02:00.0: PLL reset finished
[ 5846.064885] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[ 5846.318938] facetimehd 0000:02:00.0: Failed to lock S2 PLL: 0xc902c902
[ 5846.338854] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[ 5846.338868] facetimehd 0000:02:00.0: STRAP valid
[ 5846.338869] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[ 5846.338876] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[ 5846.338885] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[ 5846.338895] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[ 5846.338896] facetimehd 0000:02:00.0: Using step size 150
[ 5846.338901] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[ 5846.338905] facetimehd 0000:02:00.0: Virtual VTT enabled
[ 5846.359289] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[ 5846.359304] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[ 5846.359499] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[ 5846.359697] facetimehd 0000:02:00.0: firmware: failed to load facetimehd/firmware.bin (-2)
[ 5846.359700] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/firmware.bin failed with error -2
[ 5846.359829] facetimehd: probe of 0000:02:00.0 failed with error -2

Any suggestions?

salboa, 2016-03-02:
I also tried with a newer kernel 4.4.2-3

`uname -a`
`Linux macproRet 4.4.0-1-amd64 #1 SMP Debian 4.4.2-3 (2016-02-21) x86_64 GNU/Linux`

and nothing changed.

`[  243.854920] Linux video capture interface: v2.00
[  243.857553] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[  243.857660] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[  243.862838] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[  243.862867] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[  243.872813] facetimehd 0000:02:00.0: PLL reset finished
[  243.872816] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[  243.872828] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[  243.882782] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[  243.902698] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[  243.902711] facetimehd 0000:02:00.0: STRAP valid
[  243.902712] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[  243.902719] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[  243.902728] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[  243.902738] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[  243.902739] facetimehd 0000:02:00.0: Using step size 150
[  243.902744] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[  243.902748] facetimehd 0000:02:00.0: Virtual VTT enabled
[  243.923144] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[  243.923158] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[  243.923352] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[  243.923514] facetimehd 0000:02:00.0: firmware: failed to load facetimehd/firmware.bin (-2)
[  243.923516] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/firmware.bin failed with error -2
[  243.923638] facetimehd: probe of 0000:02:00.0 failed with error -2`

t1msammut, 2016-03-08:
Just in case it is relevant, I get the following warnings when compiling the module:

$ make
make -C /lib/modules/4.4.4-hardened/build M=/home/tsammut/data/temp/bcwc_pcie modules
make[1]: Entering directory '/usr/src/linux-4.4.4-hardened'
  CC [M]  /home/tsammut/data/temp/bcwc_pcie/fthd_ddr.o
  CC [M]  /home/tsammut/data/temp/bcwc_pcie/fthd_hw.o
  CC [M]  /home/tsammut/data/temp/bcwc_pcie/fthd_drv.o
  CC [M]  /home/tsammut/data/temp/bcwc_pcie/fthd_ringbuf.o
  CC [M]  /home/tsammut/data/temp/bcwc_pcie/fthd_isp.o
Function isp_fill_channel_info is missing from the size_overflow hash table +isp_fill_channel_info+fndecl+3+28355+
Function _FTHD_ISP_REG_READ is missing from the size_overflow hash table +_FTHD_ISP_REG_READ+fndecl+0+64777+
Function _FTHD_ISP_REG_READ is missing from the size_overflow hash table +_FTHD_ISP_REG_READ+fndecl+0+64777+
  CC [M]  /home/tsammut/data/temp/bcwc_pcie/fthd_v4l2.o
/home/tsammut/data/temp/bcwc_pcie/fthd_v4l2.c:289:2: warning: initialization from incompatible pointer type
  .queue_setup            = fthd_buffer_queue_setup,
  ^
/home/tsammut/data/temp/bcwc_pcie/fthd_v4l2.c:289:2: warning: (near initialization for ‘vb2_queue_ops.queue_setup’)
  CC [M]  /home/tsammut/data/temp/bcwc_pcie/fthd_buffer.o
  CC [M]  /home/tsammut/data/temp/bcwc_pcie/fthd_debugfs.o
  LD [M]  /home/tsammut/data/temp/bcwc_pcie/facetimehd.o
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: modpost: Found 23 writable function pointer(s).
To see full details build your kernel with:
'make CONFIG_DEBUG_SECTION_MISMATCH=y'
  CC      /home/tsammut/data/temp/bcwc_pcie/facetimehd.mod.o
  LD [M]  /home/tsammut/data/temp/bcwc_pcie/facetimehd.ko
make[1]: Leaving directory '/usr/src/linux-4.4.4-hardened'
$

semeano, 2016-04-04:
I'm having the same issue as @rvangsgaard.
I did unload modprobe -r bdc_pci before modprobe facetimehd.

I'm using a Macbook Air 6,2 with Elementary OS (based on Ubuntu 14.04).

mikenerone, 2016-06-04:
Same issue here: Macbook Pro 11,5, Gentoo with kernel gentoo-sources-4.4.6 (perusing other issues suggests that something broke this driver with 4.4, but don't hold me to it).

thalesac, 2016-06-20:
try to cd firmware (inside driver's cloned directory), and then:

patjak, 2016-07-04:
There are no fatal errors in the logs above so things should work fine. You can try a newer kernel, otherwise not sure how to help.

cyrrill, 2016-09-08:
I can confirm this happens as well.
```
$ sudo dmidecode -s system-product-name
MacBookPro12,1

$ uname -r
4.2.0-42-generic

$ dmesg
(...)
[ 4284.830987] brcmf_add_if: ignore IF event
[ 4286.330589] brcmf_p2p_create_p2pdev: timeout occurred
[ 7495.294860] media: Linux media interface: v0.10
[ 7495.300993] Linux video capture interface: v2.00

$ lsmod | grep facetimehd
facetimehd             98304  0
videobuf2_dma_sg       20480  1 facetimehd
videobuf2_core         49152  1 facetimehd
videodev              172032  3 v4l2_common,facetimehd,videobuf2_core

$ mplayer tv://
MPlayer2 2.0-728-g2c378c7-4build1 (C) 2000-2012 MPlayer Team

Playing tv://.
Detected file format: TV
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: unable to open '/dev/video0': No such file or directory
v4l2: ioctl set mute failed: Bad file descriptor
v4l2: 0 frames successfully processed, 0 frames dropped.
Opening as detected format "TV" failed.
Failed to recognize file format.

```

Any ideas on how to get more debug information? I don't see any errors or clues into what is wrong. The module seems to be loaded fine, but no ```/dev/video0``` is created even with ```modprobe -r bdc_pci``` beforehand.

Thanks

8leggedunicorn, 2017-02-06:
I'm having the exact same issue:

```bash
yigal@monkeys-gone-wild:~/git/bcwc_pcie$ sudo dmidecode -s system-product-name
MacBookPro11,4
yigal@monkeys-gone-wild:~/git/bcwc_pcie$ sudo modprobe -r bdc_pci
yigal@monkeys-gone-wild:~/git/bcwc_pcie$ sudo depmod -a
yigal@monkeys-gone-wild:~/git/bcwc_pcie$ sudo modprobe facetimehd
```
doesn't produce a `/dev/video{,0}` device:
yigal@monkeys-gone-wild:~/git/bcwc_pcie$ sudo ls -1 /dev/vid*
ls: cannot access '/dev/vid*': No such file or directory
```

8leggedunicorn, 2017-02-06:
Following the suggestion of @patjak in https://github.com/patjak/bcwc_pcie/issues/90 I removed the module using a usual `modprobe -r facetimehd` and then attempted to insert it with debugging info:

```
yigal@monkeys-gone-wild:/var/log$ sudo modprobe -r facetimehd
yigal@monkeys-gone-wild:/var/log$ sudo updatedb
yigal@monkeys-gone-wild:/var/log$ locate facetimehd.ko
/lib/modules/4.8.0-32-generic/extra/facetimehd.ko
yigal@monkeys-gone-wild:/var/log$ sudo insmod /lib/modules/4.8.0-32-generic/extra/facetimehd.ko dyndbg=pmf
insmod: ERROR: could not insert module /lib/modules/4.8.0-32-generic/extra/facetimehd.ko: Unknown symbol in module
yigal@monkeys-gone-wild:/var/log$ sudo insmod /lib/modules/4.8.0-32-generic/extra/facetimehd.ko
insmod: ERROR: could not insert module /lib/modules/4.8.0-32-generic/extra/facetimehd.ko: Unknown symbol in module
yigal@monkeys-gone-wild:/var/log$ lsmod |grep facetime
yigal@monkeys-gone-wild:/var/log$ sudo modprobe facetimehd
yigal@monkeys-gone-wild:/var/log$ ls /dev/video0 -l
crw-rw----+ 1 root video 81, 0 Feb  5 19:15 /dev/video0
```
While `insmod`  appears to have discovered a few errors it also may have done something useful, because either it did succeed or caused something that allowed `modprobe` to insert the kernel module successfully. Lighting is poor or the picture would look better.

So more testing is to be done but it seems an `sudo insmod ${facetimehd}.ko dyndbg=pmf` might do more than simply provide verbose debugging info. I've attached the dmesg output as it's rather verbose.
[image]
[dmesg.txt](https://github.com/patjak/bcwc_pcie/files/753582/dmesg.txt)

patjak, 2017-02-06:
Hi, try loading the V4L2 modules before loading facetimehd
# modprobe -a v4l2-common videobuf2-dma-sg videobuf2-core videobuf2-v4l2

t1msammut, 2017-02-07:
I am still seeing the issue. When attempting to load the module with  dyndbg==pmf from the command line, insmod refuses:

$ sudo insmod ./facetimehd.ko dyndbg==pmf
insmod: ERROR: could not insert module ./facetimehd.ko: Unknown symbol in module
$

I also see these errors in the kernel log:

[  543.952968] facetimehd: Unknown symbol vb2_queue_init (err 0)
[  543.952979] facetimehd: Unknown symbol vb2_dma_sg_memops (err 0)
[  543.952998] facetimehd: Unknown symbol vb2_ioctl_streamoff (err 0)
[  543.953006] facetimehd: Unknown symbol vb2_ioctl_expbuf (err 0)
[  543.953021] facetimehd: Unknown symbol vb2_ops_wait_finish (err 0)
[  543.953039] facetimehd: Unknown symbol vb2_ioctl_querybuf (err 0)
[  543.953061] facetimehd: Unknown symbol vb2_fop_release (err 0)
[  543.953069] facetimehd: Unknown symbol vb2_plane_cookie (err 0)
[  543.953073] facetimehd: Unknown symbol vb2_ioctl_dqbuf (err 0)
[  543.953078] facetimehd: Unknown symbol vb2_ioctl_create_bufs (err 0)
[  543.953089] facetimehd: Unknown symbol vb2_buffer_done (err 0)
[  543.953100] facetimehd: Unknown symbol vb2_ioctl_qbuf (err 0)
[  543.953105] facetimehd: Unknown symbol vb2_fop_mmap (err 0)
[  543.953115] facetimehd: Unknown symbol vb2_fop_read (err 0)
[  543.953126] facetimehd: Unknown symbol vb2_ops_wait_prepare (err 0)
[  543.953135] facetimehd: Unknown symbol vb2_ioctl_streamon (err 0)
[  543.953139] facetimehd: Unknown symbol vb2_fop_poll (err 0)
[  543.953155] facetimehd: Unknown symbol vb2_wait_for_all_buffers (err 0)
[  543.953167] facetimehd: Unknown symbol vb2_ioctl_reqbufs (err 0)

I think I have all of the modules inserted correctly...

$ sudo lsmod | egrep '(v4l2|videobuf2)'
videobuf2_dma_sg        7197  1 facetimehd
videobuf2_memops        2545  1 videobuf2_dma_sg
videobuf2_v4l2         10587  1 facetimehd
videobuf2_core         27180  2 facetimehd,videobuf2_v4l2
$

8leggedunicorn, 2017-02-11:
@t1msammut  Yes, I observed the errors similar to yours but more importantly even though the errors were generated /dev/video0 was created.

plaspada, 2019-02-09:
[image]
After all the instruction i'm in the situation in pic. Skype "see" the camera, but doesn't show the video. BTW with mplayer tv the camera works, as on facebook videochat. Any tips?

krpovmu, 2020-04-25:
I have the same issue with Microsoft apps (Skype, Teams) desktop, the camera work on Zoom app but in the other it recognize the camera Facetime HD but doesn't work properly.