patjak/facetimehd #51: [SOLVED] Module 'facetimehd' not loaded at startup

arnal88, 2016-01-24:
Hi,

I use Linux Mint 17.3 64 bits on a MacBook Air 11" (early 2014 version) with the 3.19.8 kernel and I'm very happy to be able to use the webcam, thanks a lot to the developers!

I followed the instructions on the Wiki ('Get Started' section) to extract the firmware (I had to install `curl` to get it work). Then I used the following commands to install the module:
```
make
sudo make install
sudo depmod -a
sudo modprobe -r bdc_pci
sudo modprobe facetimehd
```

and I tested it using 
```
mplayer tv://
```

It worked like a charm!

**Problem:**
It no longer works after rebooting. Indeed, the module `facetimehd` is not loaded at startup.

After rebooting, `lsmod | grep facetimehd` returns nothing and I couldn't find why in the logs... I tried to blacklist `bdc_pci` but the webcam module still wasn't loaded at startup.
Did I miss a step during the install procedure ?

**Solution:**
I finally got it work by adding the following lines in `/etc/rc.local`
```
modprobe -r bdc_pci
modprobe facetimehd
```

I put "SOLVED" in the title because I eventually found a solution to my problem. 
I hope it will help people that might face the same issue.

patjak, 2016-02-11:
Will close this since how you blacklist modules can be distro specific. In your case it might be that you have bdc_pci in initrd and gets loaded before rootfs is mounted (where your blacklist config sits).

closed 2016-02-11 by patjak

arnal88, 2016-02-11:
I've just realized that the solution to this issue had been added to the Wiki in the meantime (`Get started` section)!
I simply needed to add `facetimehd` to `/etc/modules`...

I'm not familiar with Linux boot and kernel modules loading, so whatever the way (`/etc/modules` or `/etc/rc.local`) what matters to me is that it works!

Thanks again for this amazing module ;-)

bjchau, 2016-05-19:
I am using basically the same setup, and I followed the steps in the Getting Started section, but when I run mplayer tv:// or I start up anything that uses the webcam, it freezes my conouter, forcing me to restart to use it.

Do you know how to fix it?