patjak/facetimehd #52: [SOLVED] Low resolution with Skype

arnal88, 2016-01-24:
Hi,

I use Linux Mint 17.3 64 bits on a MacBook Air 11" (early 2014 version) with the 3.19.8 kernel and I'm very happy to be able to use the webcam, thanks a lot to the developers!

I followed the instructions on the Wiki ('Get Started' section) to extract the firmware and to install the module, everything works like a charm when I test it with `mplayer tv://`

**Problem:**
When I use Skype (version 4.3.0.37), the webcam is automatically detected as `Apple Facetime HD (/dev/video0)` but Skype uses only the lowest available resolution (320x240).

I added the following code in `~/.Skype/shared.xml` inside the `<Video>` tag in order to force Skype to use a higher resolution (640x480 is fine for me):
```
<CaptureWidth>640</CaptureWidth>
<CaptureHeight>480</CaptureHeight>
```
but the webcam didn't work anymore!
The webcam is still detected by Skype but there is no video stream (the green led next to the webcam is off)...

**Diagnosis:**
The webcam only works with Skype when the lowest available resolution (320x240) is defined in `~/.Skype/shared.xml`.

**Solution:**
I changed the minimum resolution in the `facetimehd` module!

In the `fthd_v4l2.c` file, I replaced
```
#define FTHD_MIN_WIDTH 320
#define FTHD_MIN_HEIGHT 240
```
by
```
#define FTHD_MIN_WIDTH 640
#define FTHD_MIN_HEIGHT 480
```

I compiled and installed the `facetimehd` module again. 
Even though Skype still uses the minimum resolution, 640x480 is now a decent resolution (much better than 320x240!)

**Afterword:**
People using Skype and facing the same issue (low 320x240 resolution) should consider changing the minimum resolution and compiling the module again. I didn't notice any side effect in doing so.

I put "SOLVED" in the title because I eventually found a solution to my problem. I hope it might help!

ivanperez-keera, 2019-06-29:
@arnal88 Does Skype still work for you? I am not able to get an image at all.

> The webcam only works with Skype when the lowest available resolution (320x240) is defined in ~/.Skype/shared.xml.

@arnal88 I could not make this work. Can you show me where you put these? I created a video tag like this:
```xml
<config version="1.0" serial="<some_maybe_private_num?>" timestamp="<a_time_stamp>">                  
  <Lib>                                                                         
    <Access>                                                                    
      <Cookies>4100</Cookies>                                                   
      <Info></Info>                                                             
      <LogoutURL></LogoutURL>                                                   
      <SSID>SomeLongKickAssStringThatIDon'tKnowIfIt'sPrivate</SSID>
    </Access>                                                                   
    <Account></Account>                                                         
    <Video>                                                                     
      <CaptureWidth>1280</CaptureWidth>                                         
      <CaptureHeight>720</CaptureHeight>                                        
    </Video> 
    ... File continues ...
```
Is this where you put it? Does it work?

arnal88, 2019-06-29:
Hi,

Editing `~/.Skype/shared.xml` has never worked for me, as mentioned in my previous post:

> but the webcam didn't work anymore!

so you are experiencing the same issue as I did.

The only solution I found is increasing the minimum resolution in `facetimehd` module (see description above).

Arnaud

ivanperez-keera, 2019-06-29:
Hmm, I was led to believe it did by this comment:

> Diagnosis:
> The webcam only works with Skype when the lowest available resolution (320x240) is defined in ~/.Skype/shared.xml.

Which appeared **after**:

> but the webcam didn't work anymore!

So obviously, I thought you meant making that change would work (even if at very low res).

ivanperez-keera, 2019-06-29:
(Just to be clear: making that change, even with very low res, does not work for me either. The 1280x720 is irrelevant.)