patjak/facetimehd #53: Not working on kernel 4.5

hpe009, 2016-01-25:
Not working black screen after installing driver and mplayer gives this:

```
MPlayer2 2.0-728-g2c378c7-4build1 (C) 2000-2012 MPlayer Team
Cannot open file '/home/havardpettersen/.mplayer/input.conf': No such file or directory
Failed to open /home/havardpettersen/.mplayer/input.conf.
Cannot open file '/etc/mplayer/input.conf': No such file or directory
Failed to open /etc/mplayer/input.conf.

Playing tv://.
Detected file format: TV
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: your device driver does not support VIDIOC_G_STD ioctl, VIDIOC_G_PARM was used instead.
Selected device: Apple Facetime HD
 Capabilities:  video capture  read/write  streaming
 supported norms:
 inputs: 0 = Camera;
 Current input: 0
 Current format: YUYV
tv.c: norm_from_string(pal): Bogus norm parameter, setting default.
v4l2: ioctl enum norm failed: Inappropriate ioctl for device
Error: Cannot set norm!
Selected input hasn't got a tuner!
v4l2: ioctl request buffers failed: Invalid argument
v4l2: ioctl set mute failed: Invalid argument
v4l2: 0 frames successfully processed, 0 frames dropped.
Opening as detected format "TV" failed.
Failed to recognize file format.

Exiting... (End of file)
```

xgnata, 2016-01-29:
First note that 4.5 is only in rc1 stage as of today. It could be a bug in the kernel. Most probably it is a change of the API in V4L2.

Any warning when you compile the driver?

Try:
modprobe v4l2-common
modprobe videobuf2-dma-sg
modprobe videobuf2-core
rmmod facetimehd
modprobe -r bdc_pci
insmod ./bcwc_pcie/facetimehd.ko dyndbg==pmf
and post the relevant part of the dmesg after a try with mplayer.

hpe009, 2016-01-29:
Commands did not work, how do I give you the appropriate parts?

xgnata, 2016-01-30:
Why are you using a kernel in rc1 is your not familiar with the command line? 
Anyway, it is good to know that it is not working with 4.5-rc1.

"did not work" is not very helpful. If you don't copy/paste the errors messages then there is not much we can do.

hpe009, 2016-01-30:
Closed do to beta kernel release

hpe009, 2016-02-01:
@patjak Linux kernel 4.5 rc2 alsonot working. Cheese just says that an error occured while trying to play from webcam

patjak, 2016-02-10:
Will look into >= 4.5 kernels at a later time. Will only target 4.2 to 4.4 at the moment.

closed 2016-02-10 by patjak