patjak/facetimehd #54: Ubuntu Studio 15.10 on MacBook Air (Mid 2013) fails to build Facetime HD driver

CoolGames, 2016-01-25:
## Tried to build using make command and won't begin on Ubuntu Studio 15.10

> lcpci -nnk
```
02:00.0 Multimedia controller [0480]: Broadcom Corporation 720p FaceTime HD Camera [14e4:1570]
	Subsystem: Broadcom Corporation 720p FaceTime HD Camera [14e4:1570]
	Kernel driver in use: bdc-pci
```

### So I tried to find why and discovered this effort. Hope this helps.

```
make[1]: Entering directory '/usr/src/linux-headers-4.2.0-25-lowlatency'
arch/x86/Makefile:138: CONFIG_X86_X32 enabled but no binutils support
Makefile:669: Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
make[1]: *** No rule to make target 'MacBook'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.2.0-25-lowlatency'
Makefile:9: recipe for target 'all' failed
make: *** [all] Error 2
```

CoolGames, 2016-01-25:
Solved [Spaces in Directory name](http://askubuntu.com/questions/367838/compiling-error-while-installing-realtek-rtl8111e-in-64-bit-13-10-config-x86-x].
Now extraction error
```
 $ ./extract-firmware.sh -x ../../AppleCameraInterface 

Error: Mismatching driver hash for ../../AppleCameraInterface
Error: The unknown hash is dfac86799c6cf0aceb59bb4e732be8f030e7943eb1146830c7136f62621c9853
Error: No firmware extracted!
```

CoolGames, 2016-01-25:
Still not working even after trying [Additional Notes](https://github.com/patjak/bcwc_pcie/wiki/Get-Started#additional-notes).
I already have a /dev/video0 camera that responds to all the tests listed [here](https://github.com/patjak/bcwc_pcie/wiki/Get-Started#get-started-on-debian).

Previously working /dev/video0 no longer working with guvcview

dcunited001, 2016-01-28:
@CoolGames for me, after i installed the `mtrack` touchpad drivers, then `facetimehd` does not autoload and instead bdc_pci does.

however, after i configured facetimehd in `/etc/modules`, i guess this blocks `bdc_pci` from loading. so now the webcam works on startup.

patjak, 2016-02-10:
Added the "unload bdc_pci" info to the wiki.

closed 2016-02-10 by patjak