patjak/facetimehd #57: modprobe facetimehd fails with unknown symbols

t1msammut, 2016-01-30:
Hi, thanks for working this. Very much appreciated.

Trying to insert the facetimehd module fails on my 11,5 system, which runs 4.3.3 with pax/grsec.

    $ uname -a
    Linux t 4.3.3-hardened-r7 #3 SMP Mon Jan 25 18:25:13 GMT 2016 x86_64 Intel(R) Core(TM) i7-4870HQ CPU @ 2.50GHz GenuineIntel GNU/Linux

The required modules appear loaded:

    $ sudo lsmod | egrep 'vid|v4l'
    videobuf2_dma_sg        7864  0
    videobuf2_memops        2609  1 videobuf2_dma_sg
    videobuf2_core         36044  0
    v4l2_common             3309  1 videobuf2_core
    videodev              126967  2 v4l2_common,videobuf2_core

But the modprobe fails:

    $ sudo modprobe facetimehd
    modprobe: ERROR: could not insert 'facetimehd': Invalid argument

The following appears in dmesg:

    [  737.720095] no symbol version for vb2_dma_sg_memops
    [  737.720098] facetimehd: Unknown symbol vb2_dma_sg_memops (err -22)
    [  737.720123] no symbol version for vb2_dma_sg_init_ctx
    [  737.720125] facetimehd: Unknown symbol vb2_dma_sg_init_ctx (err -22)
    [  737.720192] no symbol version for vb2_dma_sg_cleanup_ctx
    [  737.720194] facetimehd: Unknown symbol vb2_dma_sg_cleanup_ctx (err -22)

Help?! Thanks!

t1msammut, 2016-02-01:
Rebuilding the module appears to have resolved this issue. I do not see /dev/video being created -- I don't have bdc_pci compiled or loaded -- but I'll watch that in issue #49.

closed 2016-02-01 by t1msammut

derjohn, 2016-02-01:
@t1msammut You probably forget to do a 'ldconfig' before loading the module.

schemacs, 2016-02-01:
@derjohn sudo ldconfig do help me.