patjak/facetimehd #58: FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x8a!

yaguphire, 2016-02-01:
Jan 31 22:54:39 arch kernel: FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x8a!

While using the cam and working logs are getting filled with this FWMSG: ERR.

StefanKrauth, 2016-04-21:
+1 on MacBook Air 6.1

derjohn, 2016-05-16:
On Kernel 4.6 I get with the Ubuntu PPA Kernel:

```FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x8a!```

daper, 2017-01-22:
Hello!

How about this error? I get it work but after a few seconds image freezes and camera stop working. In addition to this error, system can't be powered off; 'modprobe -rf facetimehd' also never ends. I'm using kernel 4.8.15. I could provide more info if needed, and help as I can.

Thanks!

Kethen, 2017-02-05:
Sadly same here
With ubuntu, tried kernels 4.4.0-38, 4.4.0-59 and 4.4.0-62 when this started to happen today

`[  220.214027] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x804!
[  220.222889] FWMSG: [ISP] CH = 0 CMD = 0x0101 [CISP_CMD_CH_STOP]
[  220.247395] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x804!
[  222.221068] facetimehd 0000:02:00.0: IO: timeout
[  227.017259] facetimehd 0000:02:00.0: IO: timeout
[  227.017274] ------------[ cut here ]------------
[  227.017290] WARNING: CPU: 2 PID: 3602 at /build/linux-R0TiM8/linux-4.4.0/drivers/media/v4l2-core/videobuf2-core.c:1315 vb2_start_streaming+0xf8/0x180 [videobuf2_core]()
[  227.017293] Modules linked in: wl(POE) cfg80211 rfcomm nvram msr bnep btusb btrtl btbcm btintel bluetooth facetimehd(OE) videobuf2_dma_sg videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common joydev videodev bcm5974 media input_leds binfmt_misc nls_iso8859_1 applesmc input_polldev intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul aesni_intel snd_hda_codec_hdmi aes_x86_64 lrw gf128mul glue_helper snd_hda_codec_cirrus snd_hda_codec_generic ablk_helper cryptd snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi lpc_ich snd_seq thunderbolt snd_seq_device snd_timer mei_me mei shpchp snd soundcore sbs acpi_als sbshc kfifo_buf industrialio apple_bl mac_hid parport_pc ppdev lp parport autofs4
[  227.017386]  hid_generic hid_apple usbhid hid i915 i2c_algo_bit drm_kms_helper uas syscopyarea sysfillrect ahci sysimgblt fb_sys_fops libahci drm usb_storage fjes video [last unloaded: cfg80211]
[  227.017413] CPU: 2 PID: 3602 Comm: v4l2src0:src Tainted: P        W  OE   4.4.0-38-generic #57-Ubuntu
[  227.017417] Hardware name: Apple Inc. MacBookPro11,1/Mac-189A3D4F975D5FFC, BIOS MBP111.88Z.0138.B17.1602221718 02/22/2016
[  227.017420]  0000000000000286 000000003dcd2e46 ffff880225d1fbf0 ffffffff813f1b73
[  227.017427]  0000000000000000 ffffffffc0aff4f0 ffff880225d1fc28 ffffffff810811c2
[  227.017432]  0000000000000000 ffff880062c663f8 ffff880062c66580 00000000ffffff92
[  227.017438] Call Trace:
[  227.017452]  [<ffffffff813f1b73>] dump_stack+0x63/0x90
[  227.017463]  [<ffffffff810811c2>] warn_slowpath_common+0x82/0xc0
[  227.017470]  [<ffffffff8108130a>] warn_slowpath_null+0x1a/0x20
[  227.017478]  [<ffffffffc0afbc88>] vb2_start_streaming+0xf8/0x180 [videobuf2_core]
[  227.017486]  [<ffffffffc0afc268>] vb2_core_streamon+0xf8/0x130 [videobuf2_core]
[  227.017495]  [<ffffffffc0b05de9>] vb2_streamon+0x29/0x50 [videobuf2_v4l2]
[  227.017503]  [<ffffffffc0b05e52>] vb2_ioctl_streamon+0x42/0x50 [videobuf2_v4l2]
[  227.017522]  [<ffffffffc0bc198a>] v4l_streamon+0x1a/0x20 [videodev]
[  227.017536]  [<ffffffffc0bc52c1>] __video_do_ioctl+0x291/0x310 [videodev]
[  227.017551]  [<ffffffffc0bc4d96>] video_usercopy+0x336/0x5b0 [videodev]
[  227.017565]  [<ffffffffc0bc5030>] ? video_ioctl2+0x20/0x20 [videodev]
[  227.017578]  [<ffffffffc0bc5025>] video_ioctl2+0x15/0x20 [videodev]
[  227.017590]  [<ffffffffc0bc0685>] v4l2_ioctl+0xb5/0xe0 [videodev]
[  227.017597]  [<ffffffff812211ff>] do_vfs_ioctl+0x29f/0x490
[  227.017606]  [<ffffffff81103541>] ? SyS_futex+0x81/0x180
[  227.017610]  [<ffffffff81221469>] SyS_ioctl+0x79/0x90
[  227.017618]  [<ffffffff818306f2>] entry_SYSCALL_64_fastpath+0x16/0x71
[  227.017623] ---[ end trace 83934f62a5e6813d ]---
[  229.173273] facetimehd 0000:02:00.0: IO: timeout
[  229.173309] ------------[ cut here ]------------
[  229.173327] WARNING: CPU: 0 PID: 3609 at /build/linux-R0TiM8/linux-4.4.0/drivers/media/v4l2-core/videobuf2-core.c:1315 vb2_start_streaming+0xf8/0x180 [videobuf2_core]()
[  229.173331] Modules linked in: wl(POE) cfg80211 rfcomm nvram msr bnep btusb btrtl btbcm btintel bluetooth facetimehd(OE) videobuf2_dma_sg videobuf2_memops videobuf2_v4l2 videobuf2_core v4l2_common joydev videodev bcm5974 media input_leds binfmt_misc nls_iso8859_1 applesmc input_polldev intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul aesni_intel snd_hda_codec_hdmi aes_x86_64 lrw gf128mul glue_helper snd_hda_codec_cirrus snd_hda_codec_generic ablk_helper cryptd snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi lpc_ich snd_seq thunderbolt snd_seq_device snd_timer mei_me mei shpchp snd soundcore sbs acpi_als sbshc kfifo_buf industrialio apple_bl mac_hid parport_pc ppdev lp parport autofs4
[  229.173460]  hid_generic hid_apple usbhid hid i915 i2c_algo_bit drm_kms_helper uas syscopyarea sysfillrect ahci sysimgblt fb_sys_fops libahci drm usb_storage fjes video [last unloaded: cfg80211]
[  229.173488] CPU: 0 PID: 3609 Comm: v4l2src1:src Tainted: P        W  OE   4.4.0-38-generic #57-Ubuntu
[  229.173491] Hardware name: Apple Inc. MacBookPro11,1/Mac-189A3D4F975D5FFC, BIOS MBP111.88Z.0138.B17.1602221718 02/22/2016
[  229.173495]  0000000000000286 00000000fd6877e9 ffff88023d7cfbf0 ffffffff813f1b73
[  229.173501]  0000000000000000 ffffffffc0aff4f0 ffff88023d7cfc28 ffffffff810811c2
[  229.173507]  0000000000000000 ffff880062c663f8 ffff880062c66580 00000000ffffff92
[  229.173513] Call Trace:
[  229.173526]  [<ffffffff813f1b73>] dump_stack+0x63/0x90
[  229.173536]  [<ffffffff810811c2>] warn_slowpath_common+0x82/0xc0
[  229.173542]  [<ffffffff8108130a>] warn_slowpath_null+0x1a/0x20
[... cut ...]
[  229.173627]  [<ffffffffc0bc5030>] ? video_ioctl2+0x20/0x20 [videodev]
[  229.173641]  [<ffffffffc0bc5025>] video_ioctl2+0x15/0x20 [videodev]
[  229.173652]  [<ffffffffc0bc0685>] v4l2_ioctl+0xb5/0xe0 [videodev]
[  229.173659]  [<ffffffff812211ff>] do_vfs_ioctl+0x29f/0x490
[  229.173668]  [<ffffffff81103541>] ? SyS_futex+0x81/0x180
[  229.173672]  [<ffffffff81221469>] SyS_ioctl+0x79/0x90
[  229.173679]  [<ffffffff818306f2>] entry_SYSCALL_64_fastpath+0x16/0x71
[  229.173684] ---[ end trace 83934f62a5e6813e ]---`

Kethen, 2017-02-06:
Putting PCI-E on performance mode instead of powersave seems to be a working work around of the problem, the camera has been on for 15 minutes, without dmesg messages nor image freezing in cheese.

For TLP, /etc/default/tlp:

> 

PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
to
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=performance

patjak, 2017-02-06:
@Kethen great find. I don't remember how the OSX driver handles this but it might be easiest to just always power off on "stream stop" and back on when doing "stream start". Now we know where to start looking at least.

patjak, 2019-03-18:
Fixed in the driver by disabling ASPM completely on the device at driver load time.

closed 2019-03-18 by patjak