patjak/facetimehd #60: Ubuntu 15.10: make install failed

mbelang, 2016-02-03:
Hi,

I have a Macbook Pro with Ubuntu 15.10 and I follow the procedure and it worked unit I got this :

`sudo make install
make -C /lib/modules/4.2.0-27-generic/build M=/home/mbelanger/dev/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.2.0-27-generic'
  INSTALL /home/mbelanger/dev/bcwc_pcie/facetimehd.ko
Can't read private key
  DEPMOD  4.2.0-27-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.2.0-27-generic'`

I also have this error when I do `dmesg` command :
`facetimehd: module verification failed: signature and/or required key missing - tainting kernel`

I tried several solution like disabling the CONFIG_MODULE_SIG and it didn't work.
Also tried [https://github.com/patjak/bcwc_pcie/issues/29](url)

Thanks

xgnata, 2016-02-05:
You can ignore these warnings.
It only means that the module is not signed with the key used by ubuntu to sign their kernels.

mbelang, 2016-02-05:
Ok cool but still, the camera doesn't work at all! I tried with hangout and it just tells me that I do not have any camera available.

xgnata, 2016-02-05:
Go to the folder containing facetimehd.ko adn try:

modprobe v4l2-common
modprobe videobuf2-dma-sg
modprobe videobuf2-core
rmmod facetimehd
modprobe -r bdc_pci
insmod ./facetimehd.ko dyndbg==pmf
dmesg

and post the relevant lines of the dmesg.

It won't solve the install issue (if any) but it should work. it is a "manual" way of loading the driver.
modprobe -r bdc_pci is needed due to a bug on this module (and not in facetimehd).

mbelang, 2016-02-05:
It worked like you said. Here is the dmesg log and there is a lot of messages but the only one I see relevant is : 

`facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2`

patjak, 2016-02-10:
That error is expected as you don't have a calibration settings file.

closed 2016-02-10 by patjak