patjak/facetimehd #61: "vb2: drop v4l2_format argument from queue_setup" commit breaks facetimehd

xgnata, 2016-02-05:
This commit applied in linux>=4.4
http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/include/media/videobuf2-core.h?id=df9ecb0cad14b952a2865f8b3af86b2bbadfab45

produces this warning

/usr/local/src/bcwc_pcie/fthd_v4l2.c:289:28: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .queue_setup            = fthd_buffer_queue_setup,
                            ^
/usr/local/src/bcwc_pcie/fthd_v4l2.c:289:28: note: (near initialization for ‘vb2_queue_ops.queue_setup’)

and then dmesg is full of:
[  340.238342] facetimehd: Unknown symbol vb2_queue_init (err 0)
[  340.238396] facetimehd: Unknown symbol vb2_ioctl_streamoff (err 0)
[  340.238417] facetimehd: Unknown symbol vb2_ioctl_expbuf (err 0)
[  340.238499] facetimehd: Unknown symbol vb2_ops_wait_finish (err 0)
[  340.238553] facetimehd: Unknown symbol vb2_ioctl_querybuf (err 0)
[  340.238634] facetimehd: Unknown symbol vb2_fop_release (err 0)
[  340.238681] facetimehd: Unknown symbol vb2_ioctl_dqbuf (err 0)
[  340.238714] facetimehd: Unknown symbol vb2_ioctl_create_bufs (err 0)
[  340.238784] facetimehd: Unknown symbol vb2_ioctl_qbuf (err 0)
[  340.238815] facetimehd: Unknown symbol vb2_fop_mmap (err 0)

Happy refactoring :)

Xavier

patjak, 2016-02-05:
Been ignoring this for a while since I'm sure yet if I want to track upstream or keep it working for people who don't use the latest kernel. I don't want to litter the code if ifdefs. I think I'll finish the driver up and then prepare it for upstream as the last thing I do. Then development will stop here are we'll focus on the mainline kernel instead.

xgnata, 2016-02-05:
Ok that's probably the best thing to do. Tons of ifdefs are a nogo.
We have until April 2016 before getting tons of complains for many 
ubuntu users (16.04 will be shipped with linux 4.4)
Thanks for your work on the driver.

Xavier

hobarrera, 2016-02-06:
> keep it working for people who don't use the latest kernel

Given that this targets desktop (as in "not servers"), I don't thinkg people will tend to use LTS stuff, or very slow-moving distros (eg: Red Hat?). 4.4 has been out for a while now (running 4.4.1 right now).

psanford, 2016-02-06:
Having branches/tags that correspond with kernel releases would make it a lot easier for someone running an older kernel to be able to use the driver.  So for example in a year from now when the kernel 4.9 is released and it changes all the v4l APIs, a user of Ubuntu 16.04 could at least easily find a version of the driver that would compile for their kernel.

 

> Given that this targets desktop (as in "not servers"), I don't thinkg people will tend to use LTS stuff

I'm just a dataset of one, but I only use LTS releases on my desktops.

hobarrera, 2016-02-06:
> keep it working for people who don't use the latest kernel

Right, but I'm sure you don't stagnate as much as those server sometimes seen running 4-6 year old kernels, that really what I meant.

patjak, 2016-02-10:
Going to try to finish the driver up soon so we can mainline it. When that is done distros will have to use that version and these problems will go away. In the meantime I'll try to keep this stable on 4.2 to 4.4 kernels.

closed 2016-02-10 by patjak

chatziko, 2016-02-15:
For whoever's having the same issue, I managed to make it work under linux 4.5 using a small fix: https://github.com/chatziko/bcwc_pcie/commit/0e679c094057d37031b12d862db41ac5dc24ed60.

Warning: I don't really know what I'm doing, I just read the message in the kernel commit and figured out what to change. Use at your own risk.

PS. thanks @patjak for the driver!

xgnata, 2016-02-16:
I had only a quick look at 
https://patchwork.linuxtv.org/patch/31986/
They do
alloc_ctxs[0] = dev_priv->alloc_ctx;
before
+	if (*nplanes)
+		return sizes[0] < skel->format.sizeimage ? -EINVAL : 0;
I haven't check if it matters but it is done like this in many other driver.
However, there is also this /* FIXME: We assume single plane format here but not below */ which should be fixed first in order to have a clean code (looking at exemples in https://patchwork.linuxtv.org/patch/31986/ tells me that is should be an easy fix).

patjak, 2016-04-22:
I'll soon create a mainline branch and start working on getting this upstream. Then I'll fix any breakage we currently have.

Agyapal, 2017-04-13:
Hello Guys,

I am also unable  to update ixgbevf driver on latest ubuntu 16.04 community AMI - kernel 4.4.0-1012-aws 

make install
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-1012-aws'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-1012-aws'
Copying manpages...
Installing modules...
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-1012-aws'
  INSTALL /root/ixgbevf-3.4.3/src/ixgbevf.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-1012-aws
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-1012-aws'
Running depmod...

uname -r
4.4.0-1012-aws

sudo modinfo ixgbevf 
filename:       /lib/modules/4.4.0-1012-aws/kernel/drivers/net/ethernet/intel/ixgbevf/ixgbevf.ko
version:        2.12.1-k
license:        GPL

Please suggest

xgnata, 2017-04-13:
ixgbevf ??  Why do you post here?

Agyapal, 2017-04-13:
As issue is same, so I posted here.. Unable to compile

Any hints

xgnata, 2017-04-13:
It's just a warning telling you the module is not singed. It's not a
problem. Your module is compiled.Try to load your module with modprobe. It
should work.

derjohn, 2017-04-13:
Don't forget to 'sudo depmod -a' before loading (or reboot ...). Otherwise modprobe will not find the shiny new fresh installed module.