patjak/facetimehd #62: OSX 10.11.3 has different firmware and extract/make does not work

chris-milsted, 2016-02-11:
I was just getting around to trying this on my MBP (Late 2014 15" model) and after updating OSX I have found I am running 10.11.3 and the hashes are wrong again.

I coped the firmware into the ./firmware/ folder:

$ ls
AppleCameraInterface extract-firmware.sh Makefile

I then ran make:

$ make

Error: Mismatching driver hash for AppleCameraInterface
Error: The unknown hash is dfac86799c6cf0aceb59bb4e732be8f030e7943eb1146830c7136f62621c9853
Error: No firmware extracted!
Makefile:21: recipe for target 'all' failed
make: *** [all] Error 1

Hence I just tried to dd out the firmware using the old offsets and found that the firmware may have changed.

I copied the System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface file onto a memory key and when I sha256sum this on Fedora 23 I get:

dfac86799c6cf0aceb59bb4e732be8f030e7943eb1146830c7136f62621c9853 ./AppleCameraInterface

So another new Hash.

I ran the following to try and extract the firmware, just using the old offset:

dd bs=1 skip=81920 if=./AppleCameraInterface of=firmware.bin

But when I run the check.sh script it fails:

$ ./check.sh firmware.bin 
The extracted firmware does not seem good (wrong header)

So I need to figure out how to work out the right offset, then I can add the new hash and the new offset as well to a pull request.

patjak, 2016-02-11:
You don't need to copy AppleCameraInterface from OSX anymore. Just type ```make``` and the correct file will be downloaded and the firmware extracted. Then do ```make install``` and you're good to go.

patjak, 2016-02-11:
I'll have a look at the new firmware version to see if we can use it or not.

chris-milsted, 2016-02-11:
Want me to upload the firmware here?

patjak, 2016-02-11:
No need, I probably have it already if I do the latest updates in OSX.

chris-milsted, 2016-02-11:
OK - thanks!

chris-milsted, 2016-02-12:
So I had a quick play - looking at your extract file the signature of the firmware is:

feffffeafeffffeafeffffeafeffffeafeffffeafeffffeafeffffea

I had a look for the feff string to see if I could find this - I guess it should be followed by the ffea block:

$ hexdump ./AppleCameraInterface  |grep feff
0052ae0 816c b63e 808c e99d b6c4 **feff** 6135 fa67
00686d0 ef32 e274 064f **feff** 0d6a ad33 6c7d 25a6
0098550 1565 f4a2 **feff** d7ce 8f03 12a3 7256 1fdf

I can only see the following patterns:

 feff 6135 
 feff 0d6a
feff d7ce 

I guess this means the firmware blob has changed?

Thanks

Chris

lkocman, 2016-02-16:
Hi Guys, is there any progress or some sort of guide on how to help?

Lubos

jhgorse, 2016-03-08:
The curl script to download the 10.11.2 update fails, though the url still exists and works. Manually downloading it works.
https://support.apple.com/downloads/DL1849/en_US/osxupd10.11.2.dmg

patjak, 2016-03-09:
make still works fine for me. Could it be something else that is wrong? Here's my setup:

```
curl 7.47.1 (x86_64-pc-linux-gnu) libcurl/7.47.1 OpenSSL/1.0.2g zlib/1.2.8 libidn/1.32 libssh2/1.7.0
Protocols: dict file ftp ftps gopher http https imap imaps pop3 pop3s rtsp scp sftp smb smbs smtp smtps telnet tftp 
Features: AsynchDNS IDN IPv6 Largefile GSS-API Kerberos SPNEGO NTLM NTLM_WB SSL libz TLS-SRP UnixSockets
```
```
xz (XZ Utils) 5.2.2
liblzma 5.2.2
```
```
cpio (GNU cpio) 2.12
```

jhgorse, 2016-03-09:
```
curl -V
curl 7.40.0 (x86_64-redhat-linux-gnu) libcurl/7.40.0 NSS/3.21 Basic ECC zlib/1.2.8 libidn/1.32 libssh2/1.5.0
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtsp scp sftp smb smbs smtp smtps telnet tftp 
Features: AsynchDNS IDN IPv6 Largefile GSS-API Kerberos SPNEGO NTLM NTLM_WB SSL libz UnixSockets Metalink 

xz (XZ Utils) 5.2.0
liblzma 5.2.0

cpio (GNU cpio) 2.11
```
I am re-learning linux sys admin and it is not entirely clear how to get those next versions on Fedora 22. [Fedora 24](http://mirrors.rit.edu/fedora/fedora/linux/development/24/Everything/x86_64/os/Packages/c/) appears to have them. And now the update minigame... or something else entirely. =)

patjak, 2016-03-09:
Just to be clear, I'm not sure upgrading those utils will help but it might be worth a try.

yantarou, 2016-03-11:
I just created above PR to make the firmware scripts work with 10.11.3.

The contained firmware version didn't change  though.

patjak, 2016-03-11:
Hmm, would be nice if we could support more than one version of the extraction in the Makefile. Not sure how long Apple usually keeps their updates available as downloads.

chris-milsted, 2016-03-11:
I am going to close this as the PR from @yantarou solves this for me.

closed 2016-03-11 by chris-milsted

patjak, 2016-03-11:
@chris-milsted Did I understand it right that you never tried downloading the dmg by running ```make``` but instead copied the driver from your OSX installation and ran the extraction script?

patjak, 2016-03-11:
...I'm trying to figure out if we have a problem with the downloading part or not.

chris-milsted, 2016-03-11:
@patjak yes - I booted into OSX, copied the AppleCameraInterface file onto a USB memory key and then rebooted into grub and Fedora 23 and placed that into the bcwc_pcie/firmware/ folder before I ran the make.

Once I updated with the changes for 10.11.3 in the PR it now all works perfectly.

I suspect therefore that the issue could be with the downloading not the extraction as I have just tested this all and it works fine here.

Thanks

Chris