patjak/facetimehd #64: Brightness and color different between OS X and Linux

aanm, 2016-02-21:
I've realized that in Linux the brightness is way lower. Is there a way we can fix this?
Linux: https://www.dropbox.com/s/0ouvg0zgtxhyq2v/linux.png
OS X: https://www.dropbox.com/s/wltgf5g850by55p/macox.png
Thanks.

patjak, 2016-02-21:
You can adjust the brightness with the V4L2 brightness control. You can also turn on/off auto white balance. There are even more things that can be tuned on the ISP but they are not implemented yet. You might also get different image quality with the settings file for your sensor. Extracting the sensor files from the firmware is not yet supported but we'll add that some day.

aanm, 2016-02-21:
Thanks, it's away better! Do you want me to close this issue?

closed 2016-02-25 by patjak

matinlotfali, 2025-01-17:
I came up with this:

```bash
v4l2-ctl --set-ctrl white_balance_automatic=0,brightness=255,contrast=128,saturation=180
```

Do you think it needs to be set every time the computer boots up? If so, can we add it to the installation script?

matinlotfali, 2025-01-17:
Maybe this is useful?
https://superuser.com/questions/471597/linux-v4l-webcam-make-settings-stick