patjak/facetimehd #69: kernel 4.5 cheese no video

yaguphire, 2016-03-21:
Mar 21 00:13:03 arch kernel: media: Linux media interface: v0.10
Mar 21 00:13:03 arch kernel: Linux video capture interface: v2.00
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Setting 64bit DMA mask
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: S2 PCIe link init succeeded
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: PLL reset finished
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: S2 PLL is in bypass mode
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: STRAP valid
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Using step size 151
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Virtual VTT enabled
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: ISP woke up after 0ms
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: ISP second int after 0ms
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Channel description table at 00800000
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Enabling interrupts
Mar 21 00:13:03 arch kernel: FWMSG: 
Mar 21 00:13:03 arch kernel: FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
Mar 21 00:13:03 arch kernel: FWMSG: 
Mar 21 00:13:03 arch kernel: FWMSG:   Restart count : 0
Mar 21 00:13:03 arch kernel: FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
Mar 21 00:13:03 arch kernel: FWMSG:   System Clock : 200000000 Hz
Mar 21 00:13:03 arch kernel: FWMSG:   Processor mode : SUPERVISOR
Mar 21 00:13:03 arch kernel: FWMSG:   Cache architecture type : SEPARATE
Mar 21 00:13:03 arch kernel: FWMSG:   Cache type : WRITEBACK
Mar 21 00:13:03 arch kernel: FWMSG:   Data Cache Line : 32 [0x20]
Mar 21 00:13:03 arch kernel: FWMSG:   Boot arguments entries : 2
Mar 21 00:13:03 arch kernel: FWMSG:     0000: 0x00000000 0x00000000
Mar 21 00:13:03 arch kernel: FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
Mar 21 00:13:03 arch kernel: FWMSG:   Main memory :   base : 0x00000000
Mar 21 00:13:03 arch kernel: FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
Mar 21 00:13:03 arch kernel: FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
Mar 21 00:13:03 arch kernel: FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
Mar 21 00:13:03 arch kernel: FWMSG:   Shared window : base : 0x00800000 [static wiring]
Mar 21 00:13:03 arch kernel: FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
Mar 21 00:13:03 arch kernel: FWMSG:   Shared memory : base : 0x00800000
Mar 21 00:13:03 arch kernel: FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
Mar 21 00:13:03 arch kernel: FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
Mar 21 00:13:03 arch kernel: FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
Mar 21 00:13:03 arch kernel: FWMSG:   Heap free space : 10002626 [0x0098a0c2]
Mar 21 00:13:03 arch kernel: FWMSG:   Heap allocated space : 94656 [0x000171c0]
Mar 21 00:13:03 arch kernel: FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
Mar 21 00:13:03 arch kernel: FWMSG:   Application : adc [release]
Mar 21 00:13:03 arch kernel: FWMSG:   Linked on : Jul 25 2015 - 08:48:55
Mar 21 00:13:03 arch kernel: FWMSG:   Release : S2ISP-01.43.00
Mar 21 00:13:03 arch kernel: FWMSG:   H4ISPAPPLE : 11536
Mar 21 00:13:03 arch kernel: FWMSG:   H4ISPCD : 4081
Mar 21 00:13:03 arch kernel: FWMSG:   ffw : 4143
Mar 21 00:13:03 arch kernel: FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
Mar 21 00:13:03 arch kernel: FWMSG: 
Mar 21 00:13:03 arch kernel: FWMSG: -------------------------------------------------------------------------------
Mar 21 00:13:03 arch kernel: FWMSG: 
Mar 21 00:13:03 arch kernel: FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
Mar 21 00:13:03 arch kernel: FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
Mar 21 00:13:03 arch kernel: FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
Mar 21 00:13:03 arch kernel: facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2

Linux arch 4.5.0-ARCH #1 SMP PREEMPT Sun Mar 20 23:04:35 CDT 2016 x86_64 GNU/Linux

it creates /dev/video0 although it isn't usable in vlc mplayer or cheese.

hobarrera, 2016-03-22:
I've the same issue on ArchLinux since 4.5.0 hit the stable repos. `mpv tv://` shows no video.

xgnata, 2016-03-23:
Which patches do you apply to compile it without warnings on 4.5??

hpe009, 2016-03-24:
Building driver from this source will work: https://github.com/chatziko/bcwc_pcie

xgnata, 2016-03-24:
No it does not work on ubuntu "4.4.0-15-generic".

I still get this warning:
/usr/local/src/bcwc_pcie/fthd_v4l2.c:287:28: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .queue_setup            = fthd_buffer_queue_setup,
                            ^
/usr/local/src/bcwc_pcie/fthd_v4l2.c:287:28: note: (near initialization for ‘vb2_queue_ops.queue_setup’)

and this error if I try to load the module anyway:
insmod: ERROR: could not insert module ./bcwc_pcie/facetimehd.ko: Unknown symbol in module

Strangely enough this patch looks ok...I must be missing something quite obvious.

xgnata, 2016-03-31:
stupid me...insmod does not resolve module dependencies.
My script was loading the dependencies with modprobe and then insmod facetimehd did work with kernel4.3. It looks like the v4l2 modules have been reshuffled.
Anyway:
modprobe -r bdc_pci
modprobe facetimehd
does work :)

yaguphire, 2016-04-03:
doesn't work with https://github.com/patjak/bcwc_pcie 
it creates /dev/video0 although doesn't seem to work

pulled the source from here https://github.com/chatziko/bcwc_pcie seems to work confirmed for 4.5 kernel.

dietervanhoutte, 2016-04-05:
Using Debian and 4.4.0-1-amd64. Cheese not providing video neither.

Using the camera from another application does work as it should.

stefansedich, 2016-04-16:
Just installed Arch, running 4.5 kernel, tried to install but I get no video in Cheese or Skype or even Chrome, I also tried the https://github.com/chatziko/bcwc_pcie repo from source but still get no video (have followed all other issues too), is there anything else I can try?

dmesg

[ 1599.457745] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[ 1599.458083] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[ 1599.463434] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[ 1599.463488] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[ 1599.473534] facetimehd 0000:02:00.0: PLL reset finished
[ 1599.473551] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[ 1599.729300] facetimehd 0000:02:00.0: Failed to lock S2 PLL: 0xc902c902
[ 1599.749393] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[ 1599.749407] facetimehd 0000:02:00.0: STRAP valid
[ 1599.749409] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[ 1599.749415] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[ 1599.749424] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[ 1599.749434] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[ 1599.749435] facetimehd 0000:02:00.0: Using step size 150
[ 1599.749440] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[ 1599.749444] facetimehd 0000:02:00.0: Virtual VTT enabled
[ 1599.770005] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[ 1599.770019] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[ 1599.770219] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[ 1600.226839] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[ 1600.376995] facetimehd 0000:02:00.0: ISP woke up after 0ms
[ 1600.377001] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[ 1600.377004] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[ 1600.387034] facetimehd 0000:02:00.0: ISP second int after 0ms
[ 1600.387037] facetimehd 0000:02:00.0: Channel description table at 00800000
[ 1600.397220] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[ 1600.397222] facetimehd 0000:02:00.0: Enabling interrupts
[ 1600.472359] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2

patjak, 2016-04-16:
Have you checked that you have a /dev/video0 and correct permissions to access it?

stefansedich, 2016-04-16:
I do indeed have a /dev/video0 and I thought I had the permissions to
access it (I added myself to the video group and the owner of it is
root:video, although I did try change owner to my user to no avail).

patjak, 2016-04-16:
```mplayer tv://``` might tell you something
You can also try qv4l2 which is quite nice for testing video

stefansedich, 2016-04-16:
Thanks tried that, are those ioctl errors something I should expect?

Creating config file: /root/.mplayer/config
MPlayer SVN-r37379 (C) 2000-2015 MPlayer Team
210 audio & 441 video codecs
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote
control.

Playing tv://.
TV file format detected.
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: your device driver does not support VIDIOC_G_STD ioctl, VIDIOC_G_PARM
was used instead.
Selected device: Apple Facetime HD
 Capabilities:  video capture  read/write  streaming
 supported norms:
 inputs: 0 = Camera;
 Current input: 0
 Current format: YUYV
tv.c: norm_from_string(pal): Bogus norm parameter, setting default.
v4l2: ioctl enum norm failed: Inappropriate ioctl for device
Error: Cannot set norm!
Selected input hasn't got a tuner!
v4l2: ioctl request buffers failed: Invalid argument
v4l2: ioctl set mute failed: Invalid argument
v4l2: 0 frames successfully processed, 0 frames dropped.

Exiting... (End of file)

patjak, 2016-04-16:
Oh, stupid me, you already posted that. Haven't had my cup of coffee yet :)
```v4l2: ioctl request buffers failed: Invalid argument``` is the problem but don't know why it fails. Did you compile with the correct headers for your kernel? What kernel version are you running?

stefansedich, 2016-04-16:
I am running 4.5.0-1-ARCH, I will try look at the proper headers (I am new
to this so may have done this wrong).

I appreciate your help!

stefansedich, 2016-04-16:
Looks like I have the right headers too. Any other ideas?

Thanks

Jack12816, 2016-04-24:
I have the exact same problem with linux 4.5.0 and 4.5.1 on Archlinux.
I've used these AUR packages:
* https://aur.archlinux.org/packages/bcwc-pcie-firmware/
* https://aur.archlinux.org/packages/bcwc-pcie-git/

With linux 4.4.5 everything is working fine. I have the correct linux-headers in place.
Is there something I can try? Or any patches? I would like to help here.

Jack12816, 2016-04-25:
Build of patjak/bcwc_pcie (not working, no video):

```bash
make[1]: Entering directory '/usr/lib/modules/4.5.1-1-ARCH/build'
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_ddr.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_hw.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_drv.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_ringbuf.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_isp.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_v4l2.o
/tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_v4l2.c:289:28: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .queue_setup            = fthd_buffer_queue_setup,
                            ^
/tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_v4l2.c:289:28: note: (near initialization for 'vb2_queue_ops.queue_setup')
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_buffer.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_debugfs.o
  LD [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/facetimehd.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/facetimehd.mod.o
  LD [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/facetimehd.ko
make[1]: Leaving directory '/usr/lib/modules/4.5.1-1-ARCH/build'
```

---

Build of chatziko/bcwc_pcie (working video):

```bash
make[1]: Entering directory '/usr/lib/modules/4.5.1-1-ARCH/build'
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_ddr.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_hw.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_drv.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_ringbuf.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_isp.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_v4l2.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_buffer.o
  CC [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/fthd_debugfs.o
  LD [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/facetimehd.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/facetimehd.mod.o
  LD [M]  /tmp/yaourt-tmp-hmayer/aur-bcwc-pcie-git/src/bcwc-pcie-git/facetimehd.ko
make[1]: Leaving directory '/usr/lib/modules/4.5.1-1-ARCH/build'
```

Jack12816, 2016-04-25:
#74 is related to this

tomodachi, 2016-04-25:
Same here
I'm on  4.6.0-0.rc4.git0.1.fc25.x86_64
can successfully load the module I get a /dev/video0 
but no picture when using cheese or mpv  tv:// 
  
camera activity  light on the web camera itself turns green so something is happening though..

engstk, 2016-04-26:
I mnage to compile for Linux 4.5.2 with some aroud patches https://github.com/engstk/bcwc_pcie/commits/master
But as soon as it loads Cheese on Ubuntu 16.04, cam LED goes on and image shows for a second then it crashes with a "There was an error playing video from the webcam"
I used then guvcview and all works good

slashrsm, 2016-04-28:
I experienced the same problem on 4.5.1-1-ARCH. Recompiled from https://github.com/engstk/bcwc_pcie/commits/master and camera started to work again (tested with `mplayer tv://` and with Hangouts in Chromium).

yaguphire, 2016-04-29:
i can confirm on with slashrsm for macbook pro 13" early 2015.

akosiaris, 2016-05-08:
Just reporting that it works fine for me on 4.5.0. I did obviously need a recompilation to solve my PEBKAC problem of trying to use the 4.4 kernel driver. 

Debian Jessie with jessie-backports, uname reports `4.5.0-0.bpo.1-amd64 #1 SMP Debian 4.5.1-1~bpo8+1 (2016-04-20) x86_64 GNU/Linux`

abustany, 2016-06-07:
The engstk clone works for me too on macbook pro 13" (same as for slashrsm and yaguphire)

love2d-lua, 2024-12-20:
https://github.com/patjak/facetimehd/wiki/Installation#get-started-on-debian works on mxlinux in 2024!