patjak/facetimehd #70: Does not compile in Ubuntu 16.04 with 4.4 kernel

nicolaasuni, 2016-03-22:
I am getting errors when trying to compile it using Ubuntu 16.04 with kernel 4.4.0-14-generic:

```
  CC [M]  /home/nick/WWW/OTHER/bcwc_pcie/fthd_v4l2.o
/home/nick/WWW/OTHER/bcwc_pcie/fthd_v4l2.c:289:28: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .queue_setup            = fthd_buffer_queue_setup,
                            ^
/home/nick/WWW/OTHER/bcwc_pcie/fthd_v4l2.c:289:28: note: (near initialization for ‘vb2_queue_ops.queue_setup’)
```

xgnata, 2016-03-22:
See https://github.com/patjak/bcwc_pcie/issues/61

nicolaasuni, 2016-03-22:
In the #61 ticket @chatziko linked a patch that is supposed to fix the warnings but Ubuntu 16.04 this patch seems not working.

xgnata, 2016-03-22:
I must be missing something simple....I don't understand why gcc says:
/usr/local/src/bcwc_pcie/fthd_v4l2.c:291:35: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
         .queue_setup            = fthd_buffer_queue_setup,
                                   ^
/usr/local/src/bcwc_pcie/fthd_v4l2.c:291:35: note: (near initialization for ‘vb2_queue_ops.queue_setup’)

"static int fthd_buffer_queue_setup(struct vb2_queue *vq,
  				   unsigned int *nbuffers, unsigned int *nplanes,
  				   unsigned int sizes[], void *alloc_ctxs[])"
The signature of this function looks ok...What am I missing???

xgnata, 2016-03-29:
ok the API is changing quite often...
In the 4.4 kernel of ubuntu 16.04, the signature is 
	int (*queue_setup)(struct vb2_queue *q, const void *parg,
			   unsigned int *num_buffers, unsigned int *num_planes,
			   unsigned int sizes[], void *alloc_ctxs[]);

but in the current git it's

int (*queue_setup)(struct vb2_queue *q,
			   unsigned int *num_buffers, unsigned int *num_planes,
			   unsigned int sizes[], void *alloc_ctxs[]);

http://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/include/media/videobuf2-core.h?id=df9ecb0cad14b952a2865f8b3af86b2bbadfab45

It means that we would need 2 #ifdef to support <4.4. 4.4 and >4.4.
That's the fate of an out of tree driver.

Anyway, I'm going to try to install a 4.5 on my machine.

psanford, 2016-04-24:
Head (8cc44d67f3c924f30a895dd44b9065ed04b4ed0b) compiles just fine for me on ubuntu 16.04 with kernel 4.4.0-21-generic. Are you still having problems compiling on 16.04?

blauch, 2016-04-24:
I just tried after upgrading a few days ago. 

first attempt:
```
christoph@christoph-MacBookPro:~/gitrepos/bcwc_pcie$ make 
make -C /lib/modules/4.4.0-21-generic/build M=/home/christoph/gitrepos/bcwc_pcie modules
make[1]: Verzeichnis „/usr/src/linux-headers-4.4.0-21-generic“ wird betreten
  CC [M]  /home/christoph/gitrepos/bcwc_pcie/fthd_ddr.o
  CC [M]  /home/christoph/gitrepos/bcwc_pcie/fthd_hw.o
  CC [M]  /home/christoph/gitrepos/bcwc_pcie/fthd_drv.o
  CC [M]  /home/christoph/gitrepos/bcwc_pcie/fthd_ringbuf.o
  CC [M]  /home/christoph/gitrepos/bcwc_pcie/fthd_isp.o
  CC [M]  /home/christoph/gitrepos/bcwc_pcie/fthd_v4l2.o
/home/christoph/gitrepos/bcwc_pcie/fthd_v4l2.c:289:28: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .queue_setup            = fthd_buffer_queue_setup,
                            ^
/home/christoph/gitrepos/bcwc_pcie/fthd_v4l2.c:289:28: note: (near initialization for ‘vb2_queue_ops.queue_setup’)
  CC [M]  /home/christoph/gitrepos/bcwc_pcie/fthd_buffer.o
  CC [M]  /home/christoph/gitrepos/bcwc_pcie/fthd_debugfs.o
  LD [M]  /home/christoph/gitrepos/bcwc_pcie/facetimehd.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/christoph/gitrepos/bcwc_pcie/facetimehd.mod.o
  LD [M]  /home/christoph/gitrepos/bcwc_pcie/facetimehd.ko
make[1]: Verzeichnis „/usr/src/linux-headers-4.4.0-21-generic“ wird verlassen
```

second time executing make:
```
christoph@christoph-MacBookPro:~/gitrepos/bcwc_pcie$ make
make -C /lib/modules/4.4.0-21-generic/build M=/home/christoph/gitrepos/bcwc_pcie modules
make[1]: Verzeichnis „/usr/src/linux-headers-4.4.0-21-generic“ wird betreten
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Verzeichnis „/usr/src/linux-headers-4.4.0-21-generic“ wird verlassen
```

psanford, 2016-04-24:
It looks like the module complied just fine for you.

There is a compiler warning (which is different than a compiler error) about a function pointer type mismatch. While slightly annoying, on the 4.4 kernel the types are compatible so it doesn't actually cause any problems. 

If you just want the warning to go away you can apply this patch:
```
diff --git a/fthd_v4l2.c b/fthd_v4l2.c
index 1d62a9e..c134ad4 100644
--- a/fthd_v4l2.c
+++ b/fthd_v4l2.c
@@ -41,10 +41,11 @@
 #define FTHD_NUM_FORMATS 2 /* NV16 is disabled for now */
 
 static int fthd_buffer_queue_setup(struct vb2_queue *vq,
-                                  const struct v4l2_format *fmt,
+                                  const void *parg,
                                   unsigned int *nbuffers, unsigned int *nplanes,
                                   unsigned int sizes[], void *alloc_ctxs[])
 {
+       const struct v4l2_format *fmt = parg;
        struct fthd_private *dev_priv = vb2_get_drv_priv(vq);
        struct v4l2_pix_format *cur_fmt = &dev_priv->fmt.fmt;
        int i, total_size = 0;
```

egobrain, 2016-04-25:
I have some issues on Ubuntu 16.04

```
$ sudo make install
make -C /lib/modules/4.4.0-21-generic/build M=/tmp/bcwc_pcie modules_install
make[1]: вход в каталог «/usr/src/linux-headers-4.4.0-21-generic»
  INSTALL /tmp/bcwc_pcie/facetimehd.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-21-generic
make[1]: выход из каталога «/usr/src/linux-headers-4.4.0-21-generic»
```

```
$ sudo insmod /lib/modules/4.4.0-21-generic/extra/facetimehd.ko 
insmod: ERROR: could not insert module /lib/modules/4.4.0-21-generic/extra/facetimehd.ko: Unknown symbol in module
```

```
$ dmesg
[60206.924593] facetimehd: Unknown symbol vb2_queue_init (err 0)
[60206.924628] facetimehd: Unknown symbol vb2_dma_sg_memops (err 0)
[60206.924668] facetimehd: Unknown symbol video_ioctl2 (err 0)
[60206.924706] facetimehd: Unknown symbol vb2_ioctl_streamoff (err 0)
[60206.924733] facetimehd: Unknown symbol vb2_ioctl_expbuf (err 0)
[60206.924769] facetimehd: Unknown symbol vb2_dma_sg_init_ctx (err 0)
[60206.924793] facetimehd: Unknown symbol vb2_ops_wait_finish (err 0)
[60206.924819] facetimehd: Unknown symbol v4l2_ctrl_handler_init_class (err 0)
[60206.924857] facetimehd: Unknown symbol vb2_ioctl_querybuf (err 0)
[60206.924889] facetimehd: Unknown symbol v4l2_fh_open (err 0)
[60206.924942] facetimehd: Unknown symbol video_devdata (err 0)
[60206.924967] facetimehd: Unknown symbol vb2_fop_release (err 0)
[60206.924999] facetimehd: Unknown symbol vb2_plane_cookie (err 0)
[60206.925021] facetimehd: Unknown symbol vb2_ioctl_dqbuf (err 0)
[60206.925047] facetimehd: Unknown symbol vb2_ioctl_create_bufs (err 0)
[60206.925084] facetimehd: Unknown symbol vb2_buffer_done (err 0)
[60206.925106] facetimehd: Unknown symbol v4l2_ctrl_subscribe_event (err 0)
[60206.925131] facetimehd: Unknown symbol video_unregister_device (err 0)
[60206.925159] facetimehd: Unknown symbol vb2_ioctl_qbuf (err 0)
[60206.925180] facetimehd: Unknown symbol vb2_fop_mmap (err 0)
[60206.925206] facetimehd: Unknown symbol video_device_alloc (err 0)
[60206.925235] facetimehd: Unknown symbol vb2_fop_read (err 0)
[60206.925256] facetimehd: Unknown symbol v4l2_device_register (err 0)
[60206.925330] facetimehd: Unknown symbol __video_register_device (err 0)
[60206.925375] facetimehd: Unknown symbol vb2_ops_wait_prepare (err 0)
[60206.925411] facetimehd: Unknown symbol vb2_dma_sg_cleanup_ctx (err 0)
[60206.925457] facetimehd: Unknown symbol vb2_ioctl_streamon (err 0)
[60206.925495] facetimehd: Unknown symbol vb2_fop_poll (err 0)
[60206.925534] facetimehd: Unknown symbol v4l2_ctrl_new_std (err 0)
[60206.925561] facetimehd: Unknown symbol v4l2_ctrl_handler_free (err 0)
[60206.925583] facetimehd: Unknown symbol v4l2_device_unregister (err 0)
[60206.925614] facetimehd: Unknown symbol vb2_wait_for_all_buffers (err 0)
[60206.925652] facetimehd: Unknown symbol video_device_release (err 0)
[60206.925709] facetimehd: Unknown symbol v4l2_event_unsubscribe (err 0)
[60206.925766] facetimehd: Unknown symbol vb2_ioctl_reqbufs (err 0)
```

```
$ uname -a
Linux macbook 4.4.0-21-generic #37-Ubuntu SMP Mon Apr 18 18:33:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
```

xgnata, 2016-04-25:
Insmod does not resolve the dependencies. Try modprobe.

egobrain, 2016-04-25:
Ok. I've tried modprobe

```
$ echo extra/facetimehd.ko >> /lib/modules/4.4.0-21-generic/modules.dep
$ depmod
$ modprobe facetimehd
$ lsmod | grep facetimehd
facetimehd             98304  0
videobuf2_dma_sg       20480  1 facetimehd
videobuf2_v4l2         28672  1 facetimehd
videobuf2_core         36864  2 facetimehd,videobuf2_v4l2
videodev              176128  4 v4l2_common,facetimehd,videobuf2_core,videobuf2_v4l2
```

Everything is ok, but there is no /dev/video* device :(

psanford, 2016-04-25:
Did you unload the `bdc_pci` module before loading `facetimehd` as the wiki page says to do: https://github.com/patjak/bcwc_pcie/wiki/Get-Started#devvideo-not-created

egobrain, 2016-04-25:
I've just lost firmware file :(
Now it's ok thanks.

xgnata, 2016-04-25:
Still...insmod should work too but I have no idea how modules dependencies
are listed / resolved...

blauch, 2016-04-26:
It works really smooth... I am impressed. If you read the Wiki properly - like I did not - everything is explained nicely.

nicolaasuni, 2016-04-26:
To be able to compile without errors in Ubuntu 16.04 with kernel 4.4.0 I am using the following signature:

{code}
static int fthd_buffer_queue_setup(
    struct vb2_queue *vq,
#if LINUX_VERSION_CODE < KERNEL_VERSION(4,4,0)
    const struct v4l2_format *fmt,
#endif
#if !(LINUX_VERSION_CODE >= KERNEL_VERSION(4,5,0))
    const void *parg,
#endif
    unsigned int *nbuffers,
    unsigned int *nplanes,
    unsigned int sizes[],
    void *alloc_ctxs[]
) {
{code}

See: 
https://github.com/tecnickcom/bcwc_pcie/blob/master/fthd_v4l2.c#L44

nicolaasuni, 2016-05-03:
Fixed by merging the PR:
https://github.com/patjak/bcwc_pcie/pull/77

closed 2016-05-03 by nicolaasuni

ivanperez-keera, 2016-05-10:
Sorry for commenting on a closed issue, but I keep seeing the following error message during installation:

```
========================= Installation results ===========================
make -C /lib/modules/4.4.0-21-generic/build M=/home/dash/.root/opt/facetimehd/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-21-generic'
  INSTALL /home/dash/.root/opt/facetimehd/bcwc_pcie/facetimehd.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-21-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-21-generic'

======================== Installation successful ==========================
```

Other info:
```
$ uname -a
Linux dash-MacBookAir 4.4.0-21-generic #37-Ubuntu SMP Mon Apr 18 18:33:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04 LTS
Release:	16.04
Codename:	xenial
```

I just rebooted my machine, pulled the repo, cleaned, extracted and installed the firmware successfully and then run make and sudo checkinstall.

doublebyte1, 2016-06-30:
Hi, I also get the same compiling (warnings?):

```
make -C /lib/modules/4.4.0-28-generic/build M=/home/joana/git/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-28-generic'
  INSTALL /home/joana/git/bcwc_pcie/facetimehd.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-28-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-28-generic'

```

Nevertheless, by following the instructions on the wiki, I am able to successfully load the kernel module and launch mplayer with the correct device created. As pointed out, it was crucial to load the bdc_pci module, before loading facetimehd.

I am using Ubuntu Xenial, with kernel 4.40.0-28

> Linux invismac 4.4.0-28-generic #47-Ubuntu SMP Fri Jun 24 10:09:13 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
> 
> No LSB modules are available.
> Distributor ID: Ubuntu
> Description:    Ubuntu 16.04 LTS
> Release:        16.04
> Codename:       xenial
> 

Thanks for maintaining this project: I find it really useful +1

adityaputatunda, 2017-03-13:
I myself am getting the same error while trying to install the drivers for a pen tablet. I am using ubuntu 16.04 too.

`
make -C /lib/modules/4.4.0-66-generic/build SUBDIRS=/home/adityaputatunda/apps/digimend-kernel-drivers-master modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-66-generic'
  INSTALL /home/adityaputatunda/apps/digimend-kernel-drivers-master/hid-kye.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/adityaputatunda/apps/digimend-kernel-drivers-master/hid-polostar.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  INSTALL /home/adityaputatunda/apps/digimend-kernel-drivers-master/hid-uclogic.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-66-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-66-generic'
install -D -m 0644 digimend.conf /etc/depmod.d/digimend.conf
depmod -a
install hid-rebind /sbin/hid-rebind
install -m 0644 90-hid-rebind.rules /lib/udev/rules.d/90-hid-rebind.rules
udevadm control --reload
`

Can someone help? I am stuck because as other commentors mention that it is a harmless error. But for my case it doesn't seem so as I cannot go ahead to install the drivers.

@patjak Hello there! You sound like an expert. Can you please guide me out here as You already seem to have closed off few threads on the issue previously.

xgnata, 2017-03-19:
A pen tablet? which one? which tablet has the same camera as these macbook air?
See above. You can ignore this "error". It is only a warning telling you that the driver is not signed.
It is not a security issue as you know were the driver comes from.
At the end of the compilation process, you get a facetime.ko , don't you?

Try
sudo depmod -a
sudo modprobe -r bdc_pci
sudo modprobe facetimehd

@patjak : If a tabel is using the same hardware then it is yet another good reason to try to merge this exellent work in mainline (and to ask them ask to fix bdc_pci)

seumasmac, 2017-03-20:
@xgnata I don't think the tablet has any camera (it's a drawing tablet, not like an android one or anything). I think @adityaputatunda just found the same error (possibly from google) and is looking for help with installing the input drivers for it.

@adityaputatunda It would probably be better to go to: https://github.com/DIGImend/digimend-kernel-drivers and there is a Troubleshooting Guide there. And, if that still fails, open an issue on that repo.

mamantoha, 2017-11-09:
I installed on Ubuntu 16.04 using dkms without problems.

```
sudo dkms add .
sudo dkms build facetimehd/0.1
sudo dkms install facetimehd/0.1
sudo update-initramfs -u
```