patjak/facetimehd #72: Device not detected - 05ac:8510 Apple, Inc. FaceTime HD Camera (Built-in)

granis, 2016-04-16:
Hi,

Is this camera supported?
Bus 001 Device 004: ID 05ac:8510 Apple, Inc. FaceTime HD Camera (Built-in)

I have loaded the module but it does not detect when the camera is plugged in, this is all I get in the console:

> [  988.150417] usb 1-1: New USB device found, idVendor=05ac, idProduct=8510
[  988.150422] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  988.150424] usb 1-1: Product: FaceTime HD Camera (Built-in)
[  988.150426] usb 1-1: Manufacturer: Apple Inc.
[  988.150428] usb 1-1: SerialNumber: DJHC6TZ7PSDNYGE0

Running Debian, Linux webcam 4.4.0-1-amd64 #1 SMP Debian 4.4.6-1 (2016-03-17) x86_64 GNU/Linux

patjak, 2016-04-17:
That's a USB camera and is not supported. Never seen that one before. What system is it?

granis, 2016-04-17:
It's just the camera, from a A1398 model. I works fine in OS X (via USB), that's why I thought it could work with the Linux-driver as well. :)

patjak, 2016-04-17:
Ok, so is this the built-in webcam on a MacBook Pro retina? It'll not work with this driver but it would be interesting to know more about it.

granis, 2016-04-17:
Yes it is! Anyway I can help?

hobarrera, 2016-04-17:
> It's just the camera, from a A1398 model

From what I can tell, this is slightly older than the first model to feature the PCI-E driver (the mid 2013 MBA). Shouldn't the older webcam driver support this?

patjak, 2016-04-17:
Yes, sounds like it's an older version. @granis can you give us more info on the model?

granis, 2016-04-17:
Sure,

I bought the camera here: http://www.ebay.com/itm/181296475979
And soldered the USB-cable with pinout from here: http://mikesmods.com/mm-wp/?p=340

patjak, 2016-04-18:
Ah ok it's a homebrewed solution. Then I guess your best bet is to look at one of the USB UVC drivers. Perhaps uvc_isight.

closed 2016-04-18 by patjak