patjak/facetimehd #75: Release stable version

p-kozlov, 2016-04-25:
This project is quite stable for me - it's compiling and working. Although, I see some known issues for different people, perhaps with different devices and different linux distributions.
Why not release some stable version (1.0.0 for instance, or 2016.04.25, up to the authors), to ensure that in case of active development we will have some stable version which at least is compiling well. It is usual thing in modern software development to break something.

hobarrera, 2016-04-25:
The only way this will actually be considered stable if it's mainlined.
The Linux kernel is a moving target, and the driver has required changes
with most recent releases. A single  “1.0” version could only be
considered stable for a single kernel release.

aanm, 2016-04-28:
@hobarrera What about kernel-bcwc?. i.e. `4.4.5-1.0.0` or `4.5.0-0.0.1` Meaning bcwc for kernel 4.4.5 is stable and bcwc for kernel 4.5 is still in alpha?

xgnata, 2016-06-04:
patjak said he's going to work to mainline this driver.
A driver outside of the main tree is often a dead driver.

I don't know when this process it going to start but look a the frequency at which new kernel versions are released : it gonna take quite some time.

p-kozlov, 2016-06-06:
But, for instance, virtualbox-modules are not inside linux kernel mainline and they are quite stable.

xgnata, 2016-06-06:
For bcwc_pcie we have already #ifdef for  4.4 and 4.5 kernel. V4L2
interface has not been very stable recently.  Moreover, the number of users
will grow a lot if it works out of the box (Ok the firmware is needed...but
it could be in a non-free package). Once a driver is in the mainline,  most
of the V4L2 fairly trivial interfaces changes are reflected in the driver
by the V4L2 authors...or at least they try to reflect these changes. That
is why it is so much better to be in the mainline

fidalgo, 2016-08-09:
I'm using the driver with 4.6.4 kernel and it works pretty well. I really would like to see it in the upstream, first because it would be included in every new kernel and I don't need to compile it but more important it would allow other video driver devs to take a look at the code and improve it when possible.
Also if we want to make a push to get some documentation from broadcom, a mainline driver will catch more attention for sure.

xgnata, 2016-12-19:
Just to say that it would be great to see this driver in the mainline.
I hope that the inclusion to mainline process will start.

xgnata, 2017-04-13:
Any news on the mainline inclusion ? That would be cool to have this driver in mainline and hence working out of the box.