patjak/facetimehd #76: elementaryOS Snap does not open but mplayer does.

ziazon, 2016-05-01:
Single booting elementaryOS 0.32 on a Macbook Air 6,1. The driver is succesfully installed, mplayer displays a video feed however when I attempt to run the native elementaryOS app Snap, the FacetimeHD initializes (The green LED comes on for a second) but the app fails to launch and no video feed is shown. I open up lxtask and check the processes, Snap is using 25% CPU. Is this an issue with the driver or an issue with Snap?

patjak, 2016-05-01:
Probably an issue with the driver. We're still experimental and exposes things that are only partially supported or simply incorrect.

ziazon, 2016-05-01:
No problems. By the way for anyone else looking at a later stage. This driver works with Skype.

ziazon, 2016-05-05:
Okay, back to square one now. It's completely stopped work to the point where if I try and initialise the device, it locks up elementaryOS, can't even get a log.

When I run;
`lsmod | grep facetimhd`

I receive;
`facetimehd             98304  0 
videobuf2_dma_sg       20480  1 facetimehd
videobuf2_core         53248  1 facetimehd
videodev              159744  3 v4l2_common,facetimehd,videobuf2_core`

I run;
`ls /dev/video*`

I recieve;
`/dev/video0`

I'm not sure what is going on here.

ziazon, 2016-05-05:
So after updating the kernel to 3.19.0-58-generic, I'm able to get an error output that I'll paste now;

`MPlayer 1.1-4.8 (C) 2000-2012 MPlayer Team
mplayer: could not connect to socket
mplayer: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing tv://.
TV file format detected.
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: your device driver does not support VIDIOC_G_STD ioctl, VIDIOC_G_PARM was used instead.
Selected device: Apple Facetime HD
 Capabilities:  video capture  read/write  streaming
 supported norms:
 inputs: 0 = Camera;
 Current input: 0
 Current format: YUYV
tv.c: norm_from_string(pal): Bogus norm parameter, setting default.
v4l2: ioctl enum norm failed: Inappropriate ioctl for device
Error: Cannot set norm!
Selected input hasn't got a tuner!`

patjak, 2016-09-28:
I recommend you try a newer kernel. 4.2 also have known issues so preferably something even newer.

patjak, 2016-09-28:
Please reopen if the above solution doesn't work.

closed 2016-09-28 by patjak