patjak/facetimehd #78: driver not working on Ubuntu 16.04 with all updates

costastf, 2016-05-20:
After all the updates the driver fails to load with messages like:
modprobe: ERROR: ../libkmod/libkmod-module.c:977 command_do() Error running install command for facetimehd
modprobe: ERROR: could not insert 'facetimehd': Operation not permitted
 uname -r = 4.4.0-22-generic
sudo dmidecode -s system-product-name = MacBookPro12,1

Please let me know if there is any other information that you would need and i will provide it asap.

cblaupanda, 2016-05-23:
try running modprode as root

```
modprobe: ERROR: could not insert 'facetimehd': Operation not permitted
```

costastf, 2016-05-23:
Thanks, but I am running the command as root. This is the output I get 
by running it as root.

costastf, 2016-05-23:
This was caused by an illegal modprobe script i had. Sorry.

closed 2016-05-23 by costastf