patjak/facetimehd #79: Left-right reversed video with latest version/commit

jbfavre, 2016-05-23:
Hello,
I successfully installed the lastest driver version on a Macbook Pro 12.1 on Debian sid.
The only problem is that left & right are reversed. I don't have the mirror effect I was waiting.
As far as I can remember, it used to work with previous versions.

- Driver commit:  5a7083b
- Debian: sid
- kernel: 4.5.4-1 (from Debian)

patjak, 2016-05-23:
I'm unaware of an ISP command to horizontally flip the image (though it should be possible). Will look into this some more. Perhaps there's a setting in whatever program you're using to flip in software.

jbfavre, 2016-05-23:
Have tried several video players, including some I never used like mpv.
Same result with each ones. So, I'm not sure it's related to softwares settings.

I don't remember having the same problem with previous driver versions

patjak, 2016-05-23:
Don't think we have touched anything that could change this so you probably had the same issue before. If not, please try to figure out which previous version that was ok.

Regarding software flip, my idea was that some players might have support for flipping the image "manually" in software. Adding software flip in the kernel driver is something I'd rather not do.

jbfavre, 2016-05-24:
Problem found: I did not upgrade the firmware to the latest version.
Just tested with PR #81 and image is now well oriented

Sorry for the noise

patjak, 2016-05-24:
Which version did not work? I would need to blacklist it if it's broken.

jbfavre, 2016-05-24:
I'm sorry, I closed the bug report too fast.
I thought I had found the root cause, but it was actually just an effect in cheese which showed the image in the right position :(

I still have the issue. Problem is, I don't have an kernel <4.4 anymore so, testing will take time since the last supposed to work version was with kernel < 4.4

patjak, 2016-10-12:
Hi any update on this? Flipping in player software might be the best solution for now.

jbfavre, 2016-10-13:
Nope. Still have the issue.
Need to test with latest code version though

tom-dalton-fanduel, 2016-11-11:
For what it's worth, I have the same issue using a 3.19 kernel. I'm not sure if it's always been flipped - I only noticed it recently when trying to share some writing on a whiteboard!