patjak/facetimehd #83: SSL Fails on 4.4.0-28-generic #47-Ubuntu

LSenf, 2016-06-30:
Hi,
I am trying to compile on ubuntu 16.04 with current kernel. Earlier kernel versions worked well, but now I get (after a successful make)

`$ sudo make install
make -C /lib/modules/4.4.0-28-generic/build M=/home/linus/Downloads/bcwc_pcie modules_install
make[1]: Verzeichnis „/usr/src/linux-headers-4.4.0-28-generic“ wird betreten
  INSTALL /home/linus/Downloads/bcwc_pcie/facetimehd.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-28-generic
make[1]: Verzeichnis „/usr/src/linux-headers-4.4.0-28-generic“ wird verlassen`

Kernel
`$ uname -a
Linux mba 4.4.0-28-generic #47-Ubuntu SMP Fri Jun 24 10:09:13 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

patjak, 2016-07-03:
Hi, seems you need the libssl-dev package installed.

LSenf, 2016-07-04:
Ouch. That worked, thank you very much! :)

closed 2016-07-04 by LSenf