patjak/facetimehd #87: Module init failure - Macbook Pro 11,2 / Qubes 3.2 -rc1/ Xen 4.6.1 / Linux 4.5.7

sprin, 2016-07-03:
I am seeing a module failure on Qubes OS 3.2-rc1 on a Macbook Pro 11,2. Qubes runs the Xen hypervisor, which is able to [pass PCI devices through to guests](https://www.qubes-os.org/doc/assigning-devices/) ([xen docs](http://wiki.xen.org/wiki/Xen_PCI_Passthrough)). I am passing through the camera device to a Fedora guest with kernel 4.5.7. After trying to load the module, /dev/video0 is not created, and I see the following in dmesg:

```
$ dmesg
[   21.450766] media: Linux media interface: v0.10
[   21.459469] Linux video capture interface: v2.00
[   21.473046] facetimehd 0000:00:00.0: Found FaceTime HD camera with device id: 1570
[   21.473300] facetimehd 0000:00:00.0: Xen PCI mapped GSI19 to IRQ28
[   21.489406] facetimehd 0000:00:00.0: Setting 64bit DMA mask
[   21.705058] facetimehd 0000:00:00.0: S2 PCIe link init succeeded
[   21.706204] facetimehd 0000:00:00.0: Refclk: 25MHz (0xa)
[   21.716672] facetimehd 0000:00:00.0: PLL reset finished
[   21.716675] facetimehd 0000:00:00.0: Waiting for S2 PLL to lock at 450 MHz
[   22.013648] facetimehd 0000:00:00.0: Failed to lock S2 PLL: 0xc902c902
[   22.034394] facetimehd 0000:00:00.0: DDR40 PHY PLL locked on safe settings
[   22.034936] facetimehd 0000:00:00.0: STRAP valid
[   22.034940] facetimehd 0000:00:00.0: Configuring DDR PLLs for 450 MHz
[   22.035168] facetimehd 0000:00:00.0: DDR40 PLL is locked after 0 us
[   22.035316] facetimehd 0000:00:00.0: First DDR40 VDL calibration completed after 0 us
[   22.035517] facetimehd 0000:00:00.0: Second DDR40 VDL calibration completed after 0 us
[   22.035521] facetimehd 0000:00:00.0: Using step size 143
[   22.035664] facetimehd 0000:00:00.0: VDL set to: coarse=0x10007, fine=0x10118
[   22.035854] facetimehd 0000:00:00.0: Virtual VTT enabled
[   22.056842] facetimehd 0000:00:00.0: S2 DRAM memory address: 0x22159559
[   22.057241] facetimehd 0000:00:00.0: Rewrite DDR mode registers succeeded
[   22.057588] facetimehd 0000:00:00.0: Full memory verification succeeded! (0)
[   22.518548] facetimehd 0000:00:00.0: Loaded firmware, size: 1392kb
[   22.518831] facetimehd 0000:00:00.0: Failed to get S2 CMPE ACPI handle
[   22.563291] facetimehd 0000:00:00.0: ISP woke up after 0ms
[   22.563459] facetimehd 0000:00:00.0: Number of IPC channels: 7, queue size: 44865
[   22.563467] facetimehd 0000:00:00.0: Firmware requested heap size: 3072kb
[   22.573980] facetimehd 0000:00:00.0: ISP second int after 0ms
[   22.574011] facetimehd 0000:00:00.0: Channel description table at 00800000
[   22.584338] facetimehd 0000:00:00.0: magic value: 00000000 after 0 ms
[   22.584344] facetimehd 0000:00:00.0: Enabling interrupts
[   24.584071] facetimehd 0000:00:00.0: IO: timeout
[   24.584083] facetimehd 0000:00:00.0: fthd_hw_deinit
[   24.585387] FWMSG: 
[   24.585503] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[   24.585621] FWMSG: 
[   24.585744] FWMSG:   Restart count : 0
[   24.585864] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[   24.586042] FWMSG:   System Clock : 200000000 Hz
[   24.586116] FWMSG:   Processor mode : SUPERVISOR
[   24.586229] FWMSG:   Cache architecture type : SEPARATE
[   24.586341] FWMSG:   Cache type : WRITEBACK
[   24.586476] FWMSG:   Data Cache Line : 32 [0x20]
[   24.586593] FWMSG:   Boot arguments entries : 2
[   24.586708] FWMSG:     0000: 0x00000000 0x00000000
[   24.586831] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[   24.586939] FWMSG:   Main memory :   base : 0x00000000
[   24.587076] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[   24.587179] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[   24.587286] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[   24.587398] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[   24.587509] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[   24.587615] FWMSG:   Shared memory : base : 0x00800000
[   24.587733] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[   24.587845] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[   24.587964] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[   24.588104] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[   24.588206] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[   24.588324] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[   24.588427] FWMSG:   Application : adc [release]
[   24.588544] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[   24.588656] FWMSG:   Release : S2ISP-01.43.00
[   24.588770] FWMSG:   H4ISPAPPLE : 11536
[   24.588885] FWMSG:   H4ISPCD : 4081
[   24.588997] FWMSG:   ffw : 4143
[   24.589121] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[... 153 lines cut ...]
	Capabilities: [100 v1] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr-
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- NonFatalErr+
		AERCap:	First Error Pointer: 00, GenCap+ CGenEn- ChkCap+ ChkEn-
	Capabilities: [13c v1] Device Serial Number 00-00-00-ff-ff-00-00-00
	Capabilities: [150 v1] Power Budgeting <?>
	Capabilities: [160 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [1b0 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [220 v1] #15
	Kernel driver in use: pciback
	Kernel modules: bdc_pci
```

patjak, 2016-07-03:
Hi,

From your dmesg I can see several problems but the deal breaker is probably that you don't have access to the camera related ACPI methods in the virtual machine. See ```Failed to get S2 CMPE ACPI handle```. This likely causes the ```IO: timeout``` when trying to send the start command to the ISP. One way to get around this could be to reimplement whatever the ACPI CMPE method do in the driver but it's not at the top of the todo list atm.

patjak, 2016-07-03:
Actually my camera works fine without calling the CMPE method so perhaps it's something else.

sprin, 2016-07-03:
Thanks for your reply and all the work on this driver!

To rule out other problems, I installed the driver on a Fedora 23 Live USB, and it loaded without fatal error and created /dev/video0. So it would appear that the problem has something to do with Xen's PCI passthrough.

```
$ dmesg
[  350.324846] media: Linux media interface: v0.10
[  350.333068] Linux video capture interface: v2.00
[  350.339912] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[  350.340227] facetimehd 0000:03:00.0: Found FaceTime HD camera with device id: 1570
[  350.340363] facetimehd 0000:03:00.0: Setting 64bit DMA mask
[  350.346319] facetimehd 0000:03:00.0: S2 PCIe link init succeeded
[  350.346351] facetimehd 0000:03:00.0: Refclk: 25MHz (0xa)
[  350.356372] facetimehd 0000:03:00.0: PLL reset finished
[  350.356374] facetimehd 0000:03:00.0: Waiting for S2 PLL to lock at 450 MHz
[  350.356386] facetimehd 0000:03:00.0: S2 PLL is locked after 10 us
[  350.366416] facetimehd 0000:03:00.0: S2 PLL is in bypass mode
[  350.386488] facetimehd 0000:03:00.0: DDR40 PHY PLL locked on safe settings
[  350.386503] facetimehd 0000:03:00.0: STRAP valid
[  350.386504] facetimehd 0000:03:00.0: Configuring DDR PLLs for 450 MHz
[  350.386513] facetimehd 0000:03:00.0: DDR40 PLL is locked after 0 us
[  350.386522] facetimehd 0000:03:00.0: First DDR40 VDL calibration completed after 2 us
[  350.386531] facetimehd 0000:03:00.0: Second DDR40 VDL calibration completed after 1 us
[  350.386532] facetimehd 0000:03:00.0: Using step size 144
[  350.386538] facetimehd 0000:03:00.0: VDL set to: coarse=0x10008, fine=0x10118
[  350.386543] facetimehd 0000:03:00.0: Virtual VTT enabled
[  350.407102] facetimehd 0000:03:00.0: S2 DRAM memory address: 0x22159559
[  350.407119] facetimehd 0000:03:00.0: Rewrite DDR mode registers succeeded
[  350.407358] facetimehd 0000:03:00.0: Full memory verification succeeded! (0)
[  350.864639] facetimehd 0000:03:00.0: Loaded firmware, size: 1392kb
[  350.904988] facetimehd 0000:03:00.0: ISP woke up after 0ms
[  350.904995] facetimehd 0000:03:00.0: Number of IPC channels: 7, queue size: 44865
[  350.904998] facetimehd 0000:03:00.0: Firmware requested heap size: 3072kb
[  350.915025] facetimehd 0000:03:00.0: ISP second int after 0ms
[  350.915027] facetimehd 0000:03:00.0: Channel description table at 00800000
[  350.925229] facetimehd 0000:03:00.0: magic value: 00000000 after 0 ms
[  350.925230] facetimehd 0000:03:00.0: Enabling interrupts
[  350.925349] FWMSG: 
[  350.925375] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[  350.925385] FWMSG: 
[  350.925400] FWMSG:   Restart count : 0
[  350.925524] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[  350.925543] FWMSG:   System Clock : 200000000 Hz
[  350.925558] FWMSG:   Processor mode : SUPERVISOR
[  350.925577] FWMSG:   Cache architecture type : SEPARATE
[  350.925591] FWMSG:   Cache type : WRITEBACK
[  350.925606] FWMSG:   Data Cache Line : 32 [0x20]
[  350.925624] FWMSG:   Boot arguments entries : 2
[  350.925638] FWMSG:     0000: 0x00000000 0x00000000
[  350.925719] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[  350.925736] FWMSG:   Main memory :   base : 0x00000000
[  350.925754] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[  350.925773] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[  350.925818] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[  350.925839] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[  350.925915] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[  350.925935] FWMSG:   Shared memory : base : 0x00800000
[  350.925956] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[  350.926064] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[  350.926088] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[  350.926136] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[  350.926155] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[  350.926202] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[  350.926218] FWMSG:   Application : adc [release]
[  350.926235] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[  350.926251] FWMSG:   Release : S2ISP-01.43.00
[  350.926290] FWMSG:   H4ISPAPPLE : 11536
[  350.926306] FWMSG:   H4ISPCD : 4081
[  350.926318] FWMSG:   ffw : 4143
[  350.926337] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[  350.926349] FWMSG: 
[  350.926396] FWMSG: -------------------------------------------------------------------------------
[  350.926409] FWMSG: 
[  350.999506] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[  350.999646] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[  350.999888] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[  351.000245] facetimehd 0000:03:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
```

sprin, 2016-07-04:
Solved!

I'm very happy that facetimehd is working in Qubes/Xen; there's a big security benefit in running the driver inside a standalone VM.

# The fix

The camera must be put in [permissive mode](http://wiki.xenproject.org/wiki/Xen_PCI_Passthrough#PV_guests_and_PCI_quirks) after the VM is started, but before attempting to load the facetimehd driver.

Start the VM that has the camera attached, then enable permissive mode for the device on dom0:
```
$ echo 0000:03:00.0 > /sys/bus/pci/drivers/pciback/permissive
```

Load the driver in the VM,
```
$ sudo modprobe facetimehd
```

Success:
```
$ dmesg
[   37.719157] media: Linux media interface: v0.10
[   37.747933] Linux video capture interface: v2.00
[   37.760576] facetimehd 0000:00:00.0: Found FaceTime HD camera with device id: 1570
[   37.760669] facetimehd 0000:00:00.0: Xen PCI mapped GSI19 to IRQ28
[   37.776361] facetimehd 0000:00:00.0: Setting 64bit DMA mask
[   37.825867] facetimehd 0000:00:00.0: S2 PCIe link init succeeded
[   37.826114] facetimehd 0000:00:00.0: Refclk: 25MHz (0xa)
[   37.836227] facetimehd 0000:00:00.0: PLL reset finished
[   37.836230] facetimehd 0000:00:00.0: Waiting for S2 PLL to lock at 450 MHz
[   38.151970] facetimehd 0000:00:00.0: Failed to lock S2 PLL: 0xc902c902
[   38.172227] facetimehd 0000:00:00.0: DDR40 PHY PLL locked on safe settings
[   38.172359] facetimehd 0000:00:00.0: STRAP valid
[   38.172361] facetimehd 0000:00:00.0: Configuring DDR PLLs for 450 MHz
[   38.172395] facetimehd 0000:00:00.0: DDR40 PLL is locked after 0 us
[   38.172427] facetimehd 0000:00:00.0: First DDR40 VDL calibration completed after 0 us
[   38.172472] facetimehd 0000:00:00.0: Second DDR40 VDL calibration completed after 0 us
[   38.172475] facetimehd 0000:00:00.0: Using step size 143
[   38.172504] facetimehd 0000:00:00.0: VDL set to: coarse=0x10007, fine=0x10119
[   38.172547] facetimehd 0000:00:00.0: Virtual VTT enabled
[   38.193246] facetimehd 0000:00:00.0: S2 DRAM memory address: 0x22159559
[   38.193391] facetimehd 0000:00:00.0: Rewrite DDR mode registers succeeded
[   38.193647] facetimehd 0000:00:00.0: Full memory verification succeeded! (0)
[   38.657157] facetimehd 0000:00:00.0: Loaded firmware, size: 1392kb
[   38.657416] facetimehd 0000:00:00.0: Failed to get S2 CMPE ACPI handle
[   38.698378] facetimehd 0000:00:00.0: ISP woke up after 0ms
[   38.698398] facetimehd 0000:00:00.0: Number of IPC channels: 7, queue size: 44865
[   38.698450] facetimehd 0000:00:00.0: Firmware requested heap size: 3072kb
[   38.708576] facetimehd 0000:00:00.0: ISP second int after 0ms
[   38.708582] facetimehd 0000:00:00.0: Channel description table at 00800000
[   38.718924] facetimehd 0000:00:00.0: magic value: 00000000 after 0 ms
[   38.718929] facetimehd 0000:00:00.0: Enabling interrupts
[   38.719330] FWMSG: 
[   38.719369] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[   38.719398] FWMSG: 
[   38.719425] FWMSG:   Restart count : 0
[   38.719461] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[   38.719494] FWMSG:   System Clock : 200000000 Hz
[   38.719523] FWMSG:   Processor mode : SUPERVISOR
[   38.719551] FWMSG:   Cache architecture type : SEPARATE
[   38.719578] FWMSG:   Cache type : WRITEBACK
[   38.719606] FWMSG:   Data Cache Line : 32 [0x20]
[   38.719632] FWMSG:   Boot arguments entries : 2
[   38.719658] FWMSG:     0000: 0x00000000 0x00000000
[   38.719690] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[   38.719718] FWMSG:   Main memory :   base : 0x00000000
[   38.719748] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[   38.719778] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[   38.719808] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[   38.719838] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[   38.719870] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[   38.719900] FWMSG:   Shared memory : base : 0x00800000
[   38.719931] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[   38.719965] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[   38.720000] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[   38.720039] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[   38.720066] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[   38.720096] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[   38.720123] FWMSG:   Application : adc [release]
[   38.720150] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[   38.720175] FWMSG:   Release : S2ISP-01.43.00
[   38.720199] FWMSG:   H4ISPAPPLE : 11536
[   38.720224] FWMSG:   H4ISPCD : 4081
[   38.720248] FWMSG:   ffw : 4143
[   38.720276] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[   38.720298] FWMSG: 
[   38.720332] FWMSG: -------------------------------------------------------------------------------
[   38.720356] FWMSG: 
[   38.793991] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[   38.794165] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[   38.794286] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[   38.795085] facetimehd 0000:00:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
```

I had tried this before I created this issue, but I was previously attempting to put the device into permissive mode before it was attached to the pciback driver, in other words, before the VM was started. The result is:

```
$ echo 0000:03:00.0 > /sys/bus/pci/drivers/pciback/permissive
-bash: echo: write error: No such device
```

Thanks again for the great work on the driver!

closed 2016-07-04 by sprin

sprin, 2016-07-04:
One more note: I started seeing buffer allocation failures with swiotlb:
```
facetimehd 0000:00:00.0: swiotlb buffer is full (sz: 1048576 bytes)
```
I needed to [increase swiotlb size as suggested by the Qubes docs](https://www.qubes-os.org/doc/assigning-devices/#dma-buffer-size). I found a value of 2048 works well. Larger values seem to run into issues, with either the VM failing to restart, or the same "swiotlb buffer is full" error. These are purportedly [symptoms of memory fragmentation](https://github.com/QubesOS/qubes-issues/issues/1038). With 2048, so far, the camera VM can be restarted any number of times, and the camera works without error.

Since the VM kernel is booted using [PV GRUB](https://www.qubes-os.org/doc/managing-vm-kernel/#installing-pv-grub2), we can adjust swiotlb by creating /etc/default/grub inside the VM:
```
GRUB_CMDLINE_LINUX='iommu=soft swiotlb=2048'
```

And regenerating grub.cfg:
```
sudo grub2-mkconfig -o /boot/grub2/grub.cfg
```

patjak, 2016-07-12:
Nice to see that it's working! If you have the time, please add this info to the wiki.

sprin, 2016-07-15:
I created a wiki page for your review here:
https://github.com/sprin/bcwc_pcie-wiki/blob/master/QubesOS.md

If it looks good to you, you can merge it manually, or you can let me know if you'd like me to add it through the web UI.

patjak, 2016-07-16:
Please add it to the wiki directly. Thanks!

sprin, 2016-07-17:
Done!
https://github.com/patjak/bcwc_pcie/wiki/Qubes-OS