patjak/facetimehd #88: no /dev/video0 on kernel 4.6 (fedora 24)

miekg, 2016-07-16:
Hi,

Just upgraded from Ubuntu 16.04 to Fedora 24 and my working setup on Ubuntu isn't working anymore (obviously).

The module compiles and loads OK, but the last lines of `dmesg` has:
~~~
[ 1721.954499] Linux video capture interface: v2.00
[ 1721.966974] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[ 1721.967144] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[ 1721.973168] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[ 1721.973202] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[ 1721.983245] facetimehd 0000:02:00.0: PLL reset finished
[ 1721.983248] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[ 1721.983261] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[ 1721.993311] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[ 1722.013421] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[ 1722.013437] facetimehd 0000:02:00.0: STRAP valid
[ 1722.013438] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[ 1722.013446] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[ 1722.013455] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[ 1722.013466] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[ 1722.013467] facetimehd 0000:02:00.0: Using step size 161
[ 1722.013473] facetimehd 0000:02:00.0: VDL set to: coarse=0x1000a, fine=0x1011d
[ 1722.013478] facetimehd 0000:02:00.0: Virtual VTT enabled
[ 1722.034072] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[ 1722.034087] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[ 1722.034365] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[ 1722.277659] facetimehd 0000:02:00.0: Loaded firmware, size: 737kb
[ 1732.343438] facetimehd 0000:02:00.0: Init failed! No wake signal
~~~

Loading also takes it's sweet time (like 10+ seconds)

miekg, 2016-07-16:
Ah, and the very last line:
~~~
[ 1732.343632] facetimehd: probe of 0000:02:00.0 failed with error -5
~~~

patjak, 2016-07-16:
Looks like you're loading the wrong firmware file. 737 KiB is way too small. Should be around 1300-1400 KiB. ```Init failed! No wake signal``` means the ISP didn't boot.

miekg, 2016-07-16:
Yep, my fault, didn't extract the firmware, copied the origin Apple<something> file. Extracted firmware now it works.

Thanks for your time!

closed 2016-07-16 by miekg