patjak/facetimehd #89: make install error on ubuntu 16.04

awjrichards, 2016-08-18:
On my MacBook Air on Ubuntu 16.04 with kernel 4.4.0-31-generic I get the following error upon make install:
`$ sudo make install
make -C /lib/modules/4.4.0-31-generic/build M=/home/awjrichards/src/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-31-generic'
  INSTALL /home/awjrichards/src/bcwc_pcie/facetimehd.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-31-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-31-generic'
`

awjrichards, 2016-08-18:
Tried again using `checkinstall` rather than `make install`, and it worked just fine.

patjak, 2016-09-28:
Glad you got to work. Closing issue.

patjak, 2016-09-28:
BTW, please update the wiki installation instructions if you feel that something is missing. Thanks

closed 2016-09-28 by patjak