patjak/facetimehd #90: Gentoo on MacBook Pro 12,1: facetimehd: probe of 0000:02:00.0 failed with error -38

aayla-secura, 2016-08-19:
I'm running Gentoo, kernel 4.6.4 on a MacBook Pro 12,1. I built the module using the ebuild in the toaster overlay, but it fails to load:

`[   51.188678] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[   51.189027] facetimehd 0000:02:00.0: Failed to enable MSI
[   51.189183] facetimehd: probe of 0000:02:00.0 failed with error -38`

I have enabled CONFIG_VIDEOBUF2_DMA_SG and CONFIG_PCI_MSI. Not sure what other relevant information I can give.

patjak, 2016-10-12:
Hi, can you load the driver with debugging enabled e.g: ```insmod ./facetimehd.ko dyndbg==pmf``` to see if you get any additional info?

aayla-secura, 2016-10-16:
Hi, I tried that, no additional information was output. I am now using kernel 4.8.1 and it works fine, I have not tested it with any other kernel (just 4.6.4 and 4.8.1).

closed 2019-12-09 by patjak