patjak/facetimehd #91: modprobe doesn't locate driver on Fedora 24

lkocman, 2016-08-23:
Hi Guys,

( just to note that I did follow getting started, and I'm doing it on clean install of Fedora 24)

as root"
make
make install # simply puts firmware.bin into /usr/lib/firmware/facetimehd/firmware.bin
depmod # tried depmod -a as well

[root@localhost pci]# modprobe facetimehd
modprobe: FATAL: Module facetimehd not found in directory /lib/modules/4.6.6-300.fc24.x86_64

[lkocman@localhost firmware]$ ls -la /usr/lib/firmware/facetimehd/
total 1412
drwxr-xr-x.  2 root root    4096 Aug 23 09:45 .
drwxr-xr-x. 73 root root   12288 Aug 23 09:36 ..
-rw-r--r--.  1 root root 1425412 Aug 23 09:45 firmware.bin

I also did tried to copy facetime/firmware.bin into /lib/modules/`uname -r`/extra/drivers/pci/ however modprobe skips it unless there is a .ko suffix (instead of .bin) ... and as expected it fails with .ko due invalid format. Is there any missing extra step that does something with .bin file and puts it under expected location (/lib/modules /usr/lib/modules)? I remember that modprobe worked on fedora23 few months ago

Thanks
Thanks

nhed, 2016-09-03:
I think @lkocman just extracted+installed the firmware (common section of instructions) but did not proceed in the [fedora-specific section](https://github.com/patjak/bcwc_pcie/wiki/Get-Started#get-started-on-fedora).

Just installed today and  the only thing that was broken for me was fixed by first running `dnf update` and reboot (upstream kernel upgraded and `kernel-devel` that was installed was not for me then-installed kernel).

lkocman, 2016-09-04:
Hello folks

@xgnata @nhed you're correct I originally did just make in the firmware dir ... and forgot to do make in the upper section. I do believe that the confusion came from Getting started guide, which leads you to firmware dir rather to main Makefile. The actual Makefile that I skipped was covered in the part in "Installation on Fedora" etc .

https://github.com/patjak/bcwc_pcie/wiki/Get-Started

All worked just fine in the end. 

Thanks

patjak, 2016-09-28:
Feel free to improve the install instructions if needed. The wiki is open for anyone to edit. Thanks.

closed 2016-09-28 by patjak