patjak/facetimehd #95: BUG: unable to handle kernel paging request

cyrrill, 2016-09-09:
On Ubuntu 15.10 with a MacBookPro 12.1 running Kernel 4.2.0-42

module loaded, and /dev/video0 created, however Mplayer does not work, Cheese crashes on start, and Skype only shows a black box, though it does detect the device.

```
$ mplayer -vo gl3 tv://
MPlayer2 2.0-728-g2c378c7-4build1 (C) 2000-2012 MPlayer Team

Playing tv://.
Detected file format: TV
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: your device driver does not support VIDIOC_G_STD ioctl, VIDIOC_G_PARM was used instead.
Selected device: Apple Facetime HD
 Capabilities:  video capture  read/write  streaming
 supported norms:
 inputs: 0 = Camera;
 Current input: 0
 Current format: YUYV
tv.c: norm_from_string(pal): Bogus norm parameter, setting default.
v4l2: ioctl enum norm failed: Inappropriate ioctl for device
Error: Cannot set norm!
Selected input hasn't got a tuner!

```

```
[62443.737950] media: Linux media interface: v0.10
[62443.743443] Linux video capture interface: v2.00
[62443.751957] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[62443.752107] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[62443.757602] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[62443.757635] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[62443.767566] facetimehd 0000:02:00.0: PLL reset finished
[62443.767572] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[62443.767584] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[62443.777503] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[62443.797364] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[62443.797378] facetimehd 0000:02:00.0: STRAP valid
[62443.797380] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[62443.797387] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[62443.797396] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[62443.797406] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[62443.797407] facetimehd 0000:02:00.0: Using step size 153
[62443.797412] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[62443.797416] facetimehd 0000:02:00.0: Virtual VTT enabled
[62443.817756] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[62443.817770] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[62443.817971] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[62444.278290] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[62444.429392] facetimehd 0000:02:00.0: ISP woke up after 0ms
[62444.429398] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[62444.429475] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[62444.439405] facetimehd 0000:02:00.0: ISP second int after 0ms
[62444.439408] facetimehd 0000:02:00.0: Channel description table at 00800000
[62444.449528] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[62444.449529] facetimehd 0000:02:00.0: Enabling interrupts
[62444.449660] FWMSG: 
[62444.449682] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[62444.449693] FWMSG: 
[62444.449736] FWMSG:   Restart count : 0
[62444.449781] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[62444.449794] FWMSG:   System Clock : 200000000 Hz
[62444.449835] FWMSG:   Processor mode : SUPERVISOR
[62444.449849] FWMSG:   Cache architecture type : SEPARATE
[62444.449863] FWMSG:   Cache type : WRITEBACK
[62444.449900] FWMSG:   Data Cache Line : 32 [0x20]
[62444.449914] FWMSG:   Boot arguments entries : 2
[... 50 lines cut ...]
[62449.885679]  0000000000000020 00000000ab6d7c7f ffff88025fe93800 ffff8801d5bcca88
[62449.885697]  ffff88023db205c0 ffff8801d5bcc85c 0000000000000000 ffff88023db20604
[62449.885716] Call Trace:
[62449.885725]  [<ffffffffc06cba44>] __vb2_queue_alloc+0x164/0x530 [videobuf2_core]
[62449.885744]  [<ffffffffc06cc008>] __reqbufs.isra.20+0x1f8/0x360 [videobuf2_core]
[62449.885763]  [<ffffffffc06cd6ac>] vb2_ioctl_reqbufs+0x6c/0xa0 [videobuf2_core]
[62449.885793]  [<ffffffffc0724b63>] v4l_reqbufs+0x43/0x50 [videodev]
[62449.885810]  [<ffffffffc07237fd>] __video_do_ioctl+0x29d/0x310 [videodev]
[62449.885828]  [<ffffffff811e1fbb>] ? __kmalloc+0x1cb/0x270
[62449.885843]  [<ffffffffc07232c6>] video_usercopy+0x336/0x5b0 [videodev]
[62449.885861]  [<ffffffffc0723560>] ? video_ioctl2+0x20/0x20 [videodev]
[62449.885877]  [<ffffffff810b0ab9>] ? update_curr+0x79/0x160
[62449.885891]  [<ffffffff81242476>] ? fsnotify+0x316/0x4a0
[62449.885904]  [<ffffffff810bea88>] ? __wake_up+0x48/0x60
[62449.885919]  [<ffffffffc0723555>] video_ioctl2+0x15/0x20 [videodev]
[62449.885935]  [<ffffffffc071e681>] v4l2_ioctl+0xb1/0xe0 [videodev]
[62449.885951]  [<ffffffff81214205>] do_vfs_ioctl+0x295/0x480
[62449.885965]  [<ffffffff81214469>] SyS_ioctl+0x79/0x90
[62449.885978]  [<ffffffff817fa072>] entry_SYSCALL_64_fastpath+0x16/0x75
[62449.885993] Code: ff ff ff 48 8b 73 10 8b 53 40 48 8b 7b 30 31 c9 4c 8b 45 b0 41 b9 d0 00 00 00 e8 c8 c7 ce c0 85 c0 0f 85 47 01 00 00 48 8b 45 a8 <48> 8b 38 e8 24 20 e3 c0 49 89 c5 48 89 03 8b 43 1c 4d 85 ed 44 
[62449.886067] RIP  [<ffffffffc06f00f4>] vb2_dma_sg_alloc+0x194/0x10a0 [videobuf2_dma_sg]
[62449.886087]  RSP <ffff88011fadfaf8>
[62449.886095] CR2: 0000000000096000
[62449.898065] ---[ end trace caddc6404bfa3b80 ]---
```

patjak, 2016-09-28:
Probably something we can fix but I would recommend that you upgrade the kernel instead. 4.2 is quite old (though distros might still ship it).

cyrrill, 2016-09-28:
I tried upgrading to 4.3, but it disabled my trackpad, so had to revert. Will look for a cabled mouse and try again. :)

closed 2016-10-12 by patjak

cyrrill, 2016-10-22:
@patjak  Just to confirm this issue got solved by upgrading to Kernel 4.4