patjak/facetimehd #96: All images orange/red

cachedout, 2016-09-21:
I compiled the driver from the HEAD of the master branch and it works! Yay!

However, all video is coming through with a very strong orange/red tint.

Kernel: 4.6.4-1-ARCH

Below are system logs from the point where the driver was loaded. After loading the driver, an application was launched to view the video:

```
[  129.800857] media: Linux media interface: v0.10
[  129.804449] Linux video capture interface: v2.00
[  129.807130] facetimehd 0000:04:00.0: Found FaceTime HD camera with device id: 1570
[  129.807223] facetimehd 0000:04:00.0: Setting 64bit DMA mask
[  129.812458] facetimehd 0000:04:00.0: S2 PCIe link init succeeded
[  129.812486] facetimehd 0000:04:00.0: Refclk: 25MHz (0xa)
[  129.822517] facetimehd 0000:04:00.0: PLL reset finished
[  129.822519] facetimehd 0000:04:00.0: Waiting for S2 PLL to lock at 450 MHz
[  129.822532] facetimehd 0000:04:00.0: S2 PLL is locked after 10 us
[  129.832579] facetimehd 0000:04:00.0: S2 PLL is in bypass mode
[  129.852671] facetimehd 0000:04:00.0: DDR40 PHY PLL locked on safe settings
[  129.852684] facetimehd 0000:04:00.0: STRAP valid
[  129.852685] facetimehd 0000:04:00.0: Configuring DDR PLLs for 450 MHz
[  129.852692] facetimehd 0000:04:00.0: DDR40 PLL is locked after 0 us
[  129.852702] facetimehd 0000:04:00.0: First DDR40 VDL calibration completed after 2 us
[  129.852714] facetimehd 0000:04:00.0: Second DDR40 VDL calibration completed after 2 us
[  129.852714] facetimehd 0000:04:00.0: Using step size 154
[  129.852719] facetimehd 0000:04:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[  129.852724] facetimehd 0000:04:00.0: Virtual VTT enabled
[  129.873305] facetimehd 0000:04:00.0: S2 DRAM memory address: 0x22159559
[  129.873319] facetimehd 0000:04:00.0: Rewrite DDR mode registers succeeded
[  129.873518] facetimehd 0000:04:00.0: Full memory verification succeeded! (0)
[  130.331501] facetimehd 0000:04:00.0: Loaded firmware, size: 1392kb
[  130.371801] facetimehd 0000:04:00.0: ISP woke up after 0ms
[  130.371806] facetimehd 0000:04:00.0: Number of IPC channels: 7, queue size: 44865
[  130.371809] facetimehd 0000:04:00.0: Firmware requested heap size: 3072kb
[  130.381847] facetimehd 0000:04:00.0: ISP second int after 0ms
[  130.381849] facetimehd 0000:04:00.0: Channel description table at 00800000
[  130.392042] facetimehd 0000:04:00.0: magic value: 00000000 after 0 ms
[  130.392044] facetimehd 0000:04:00.0: Enabling interrupts
[  130.392233] FWMSG: 
[  130.392272] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[  130.392287] FWMSG: 
[  130.392309] FWMSG:   Restart count : 0
[  130.392348] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[  130.392372] FWMSG:   System Clock : 200000000 Hz
[  130.392399] FWMSG:   Processor mode : SUPERVISOR
[  130.392427] FWMSG:   Cache architecture type : SEPARATE
[  130.392452] FWMSG:   Cache type : WRITEBACK
[  130.392476] FWMSG:   Data Cache Line : 32 [0x20]
[  130.392490] FWMSG:   Boot arguments entries : 2
[  130.392512] FWMSG:     0000: 0x00000000 0x00000000
[  130.392547] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[  130.392575] FWMSG:   Main memory :   base : 0x00000000
[  130.392594] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[  130.392612] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[  130.392630] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[  130.392648] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[  130.392825] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[  130.392841] FWMSG:   Shared memory : base : 0x00800000
[  130.392861] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[  130.392881] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[  130.392903] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[  130.393015] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[  130.393032] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[  130.393051] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[  130.393066] FWMSG:   Application : adc [release]
[  130.393081] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[  130.393096] FWMSG:   Release : S2ISP-01.43.00
[  130.393108] FWMSG:   H4ISPAPPLE : 11536
[  130.393122] FWMSG:   H4ISPCD : 4081
[... 346 lines cut ...]
[  349.203480] FWMSG: [ISP] CH = 0 CMD = 0x0c00 [CISP_CMD_CH_DRC_START]
[  349.203621] FWMSG: [ISP] CH = 0 CMD = 0xc300 [CISP_CMD_APPLE_CH_TONE_CURVE_ADAPTATION_START]
[  349.203763] FWMSG: [ISP] CH = 0 CMD = 0x0218 [CISP_CMD_CH_AE_SPEED_SET]
[  349.203780] FWMSG: [DSC] CH = 0   AE Speed Set [60] 
[  349.203911] FWMSG: [ISP] CH = 0 CMD = 0x021a [CISP_CMD_CH_AE_STABILITY_SET]
[  349.203927] FWMSG: [DSC] CH = 0   AE Stability Set [75] 
[  349.204061] FWMSG: [ISP] CH = 0 CMD = 0x0229 [CISP_CMD_CH_AE_STABILITY_TO_STABLE_SET]
[  349.204076] FWMSG: [DSC] CH = 0   AE Stability Set [8] 
[  349.204205] FWMSG: [ISP] CH = 0 CMD = 0x0115 [CISP_CMD_CH_SIF_PIXEL_FORMAT_SET]
[  349.204337] FWMSG: [ISP] CH = 0 CMD = 0x0124 [CISP_CMD_CH_CAMERA_ERR_HANDLE_CONFIG]
[  349.204471] FWMSG: [ISP] CH = 0 CMD = 0x0d05 [CISP_CMD_CH_FACE_DETECTION_ENABLE]
[  349.204700] FWMSG: [ISP] CH = 0 CMD = 0x0d00 [CISP_CMD_CH_FACE_DETECTION_START]
[  349.204928] FWMSG: [ISP] CH = 0 CMD = 0x0208 [CISP_CMD_CH_AE_FRAME_RATE_MAX_SET]
[  349.205086] FWMSG: [DSC] CH = 0   AE Frame Rate Max Set [8448] 
[  349.205214] FWMSG: [ISP] CH = 0 CMD = 0x020a [CISP_CMD_CH_AE_FRAME_RATE_MIN_SET]
[  349.205321] FWMSG: [DSC] CH = 0   AE Frame Rate Min Set [8448] 
[  349.205469] FWMSG: [ISP] CH = 0 CMD = 0xc100 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_START]
[  349.205734] FWMSG: [ISP] CH = 0 CMD = 0xc102 [CISP_CMD_APPLE_CH_MOTION_HISTORY_START]
[  349.205986] FWMSG: [ISP] CH = 0 CMD = 0xc113 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_ENABLE]
[  349.206144] FWMSG: [ISP] CH = 0 CMD = 0x8106 [CISP_CMD_APPLE_CH_STREAMING_MODE_SET]
[  349.206160] FWMSG: [DSC] CH = 0   Streaming Mode Set [0]
[  349.206296] FWMSG: [ISP] CH = 0 CMD = 0x0b02 [CISP_CMD_CH_SCALER_BRIGHTNESS_SET]
[  349.206431] FWMSG: [ISP] CH = 0 CMD = 0x0b03 [CISP_CMD_CH_SCALER_CONTRAST_SET]
[  349.206550] FWMSG: [ISP] CH = 0 CMD = 0x0100 [CISP_CMD_CH_START]
```

patjak, 2016-09-28:
Which program are you using to stream from the webcam? Have you tried changing any of the controle (hue, etc.) to see if you're just stuck at some bad default value. Could also be a confusion of formats between driver and application. Can you try qv4l2 or mplayer to see if you get the same results?

cachedout, 2016-09-29:
@patjak I have tried both `qv4l2` and `mplayer` and both produce the same results. Even playing around with the `hue` setting on `qv4l2` doesn't seem to provide any relief, I'm afraid.

AmaanC, 2016-09-29:
I had the same problem, but it seems to have vanished. The only things I've
done recently as far as I can remember, since last seeing the problem are:
- pacman -Syu
- Reboot

Give that a shot and see if it works?

AmaanC, 2016-10-14:
Okay, turns out the problem isn't gone. Let me know if I can provide any logs (and how to access them, if you could).

I'm running Arch Linux (Kernel 4.7.2-2) on a MacBook Pro Retina (11,5)

tilarids, 2016-11-19:
I've got the same issue. 4.8.8-gentoo MacBook Pro Retina (13'' early 2015). Tried using mplayer & cheese.
Any way I can help debugging the issue?

cachedout, 2016-12-06:
Just a note to say that I've upgraded my Arch system fully and pulled the latest from this repo and recompiled and the problem still exists. :-/

patjak, 2016-12-06:
Hi, thanks for reporting. If this isn't a regression I don't think there is much we can do from a driver point of view. Instead you should use an application that can mirror the image in software. With that said, one would think that such a fancy camera have hardware support for mirroring but I can't find any commands for it.

patjak, 2016-12-06:
Sounds like an issue with the firmware or camera model you're using. I'm trying to figure out what the difference is between a working / non working config. Found nothing obvious so far :(

cachedout, 2016-12-06:
Thank you for investigating, @patjak. Happy to provide any additional detail you might find helpful.

dustinmoorenet, 2017-01-13:
It seems to happen because of some auto adjusting. It looks fine when I position my body forward and turns orange as I lean back. This is very consistent. I don't know if it has to do with my orange shirt or not :smile: My laptop is 2013 Macbook Air. If you need any logs, try out code or other samples just ask.

[image]

cachedout, 2017-01-13:
I don't have the module loaded at the moment, but if memory serves I saw similar behavor to what @dustinmoorenet is reporting.

patjak, 2017-01-16:
Thanks, that makes it a bit easier to understand. I was thinking it looked more like a format conversion error. Maybe I've already asked but did any of you try to turn off auto white balance?

dustinmoorenet, 2017-01-22:
Hmm, I can't get it to reproduce after I had to manually recompile when my kernel updated to `4.8.0-34-generic`.

hobarrera, 2017-02-08:
I'm having the same issue on a MacBook Pro 2015.

As a reference, I *just* upgraded from a MacBook Air 2013. I literally `dd`'d the entire disk over `ssh`, so the setup is identical. I think this makes it very clear that the issue is specifically with this hardware, and not all webcams (but it's clearly hardware related, and not setup-related).

```
$ sudo lspci -vnn -d 14e4:1570  # MBP 2015
04:00.0 Multimedia controller [0480]: Broadcom Limited 720p FaceTime HD Camera [14e4:1570]
        Subsystem: Broadcom Limited 720p FaceTime HD Camera [14e4:1570]
        Flags: bus master, fast devsel, latency 0, IRQ 38
        Memory at a0a00000 (64-bit, non-prefetchable) [size=64K]
        Memory at 80000000 (64-bit, prefetchable) [size=256M]
        Memory at a0900000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [48] Power Management version 3
        Capabilities: [58] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [68] Vendor Specific Information: Len=44 <?>
        Capabilities: [ac] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [13c] Device Serial Number 00-00-00-ff-ff-00-00-00
        Capabilities: [150] Power Budgeting <?>
        Capabilities: [160] Virtual Channel
        Capabilities: [1b0] Latency Tolerance Reporting
        Capabilities: [220] #15
        Kernel driver in use: facetimehd
```

```
$ sudo lspci -vnn -d 14e4:1570  # MBA 2013
02:00.0 Multimedia controller [0480]: Broadcom Limited 720p FaceTime HD Camera [14e4:1570]
        Subsystem: Broadcom Limited 720p FaceTime HD Camera [14e4:1570]
        Flags: bus master, fast devsel, latency 0, IRQ 44
        Memory at b0900000 (64-bit, non-prefetchable) [size=64K]
        Memory at 90000000 (64-bit, prefetchable) [size=256M]
        Memory at b0800000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [48] Power Management version 3
        Capabilities: [58] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [68] Vendor Specific Information: Len=44 <?>
        Capabilities: [ac] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [13c] Device Serial Number 00-00-00-ff-ff-00-00-00
        Capabilities: [150] Power Budgeting <?>
        Capabilities: [160] Virtual Channel
        Capabilities: [1b0] Latency Tolerance Reporting
        Capabilities: [220] #15
        Kernel driver in use: facetimehd
```

Both:
```
$ uname -a
Linux athena.barrera.io 4.9.8-1-ARCH #1 SMP PREEMPT Mon Feb 6 12:59:40 CET 2017 x86_64 GNU/Linux
```

patjak, 2017-07-06:
Having the sensor settings file loaded might improve the situation. Might also help if we add more of the color settings. The ISP can do lots of things to improve image quality it seems.

nitaigao, 2018-01-08:
@patjak Is it possible to point me in the direction on how to find the sensor settings file and load it?

jeancochrane, 2018-01-23:
I was able to fix the orange color cast (running Ubuntu 16.04 on a 2015 Macbook Pro) using [`v4l2-ctl`](https://www.mankier.com/1/v4l2-ctl):

```
sudo apt-get install v4l-utils
v4l2-ctl -l

> User Controls
>
>                     brightness (int)    : min=0 max=255 step=1 default=128 value=128 flags=slider
>                      contrast (int)    : min=0 max=255 step=1 default=128 value=128 flags=slider
>                     saturation (int)    : min=0 max=255 step=1 default=128 value=128 flags=slider
>                            hue (int)    : min=0 max=255 step=1 default=128 value=128 flags=slider
>        white_balance_automatic (bool)   : default=1 value=1

v4l2-ctl -c white_balance_automatic=0
```

Maybe this was obvious, but I'm new to video drivers on Linux, so I figure it might be helpful to someone in a similar position.

Thanks for the fantastic work on this driver, it's a lifesaver!

nitaigao, 2018-03-30:
Switching off AWB doesnt fix it for me unfortunately. I'd be happy to try out any fixes or a settings file if someone can point me in the right direction.

SlightlyCyborg, 2018-10-14:
Same. ArchLinux using this package https://aur.archlinux.org/packages/bcwc-pcie-git/

raelik, 2018-11-20:
So I'm having this same issue, but one thing I've noticed is in guvcview, if I keep toggling the auto white balance, during the instant that it's changing the value, the white balance is correct.

NoMod-Programming, 2018-12-09:
Just wondering, I managed to fix this issue (at least for now) on my system by going to sleep and then waking it up. Does that work for you guys?

abelsromero, 2019-09-30:
Same issue and nothing seems to fix it (5.2.11-1-MANJARO). Another fact is that cheese does not work, but I can use the came for Hangouts in Firefox...in orange mode though.

patjak, 2019-12-09:
Note: I've only tested this on 1871_01XX.dat. Use at your own risk and report errors here.

I can confirm that when using the set files (sensor calibration settings) the colors look much better. The problem is that I only know how to extract them from the windows bootcamp driver. These are the steps you can follow:

1) Download driver package from https://support.apple.com/kb/DL1837
2) Extract the \BootCamp\Drivers\Apple\AppleCamera64.exe file
3) Run: unrar x AppleCamera64.exe

You should now have the AppleCamera.sys file which is the driver containing the set files.

4) Run the following commands:
```
dd bs=1 skip=1663920 count=33060 if=AppleCamera.sys of=9112_01XX.dat
dd bs=1 skip=1644880 count=19040 if=AppleCamera.sys of=1771_01XX.dat
dd bs=1 skip=1606800 count=19040 if=AppleCamera.sys of=1871_01XX.dat
dd bs=1 skip=1625840 count=19040 if=AppleCamera.sys of=1874_01XX.dat
```
You should now have the four .dat files

5) Copy the .dat files into your facetimehd firmware directory (eg. /lib/firmware/facetimehd/)
6) Reload the driver
7) Check that you don't get the "Direct firmware load for facetimehd..." error

You should now have correct colors.

patjak, 2019-12-09:
I added this to the wiki: https://github.com/patjak/bcwc_pcie/wiki/Extracting-the-sensor-calibration-files