patjak/facetimehd #98: Driver breaks suspend and hibernate (without being open)

justanotherjason, 2016-09-22:
Hello there, 

I am not sure if anyone else has the same issue but installing the driver completely breaks the hibernation and suspend routines. I am using a late 2015 MacBook Pro with Arch Linux (Gnome) and installed the driver using the AUR package.

The driver works perfectly fine, I tested it in Cheese and it does load up. The problem is that just by having the driver installed on my machine (not running) it prevents me from suspending or hibernating. Once I suspend or hibernate the computer can never resume properly. The screen lights up for a second but GDM never shows up.

Is this a known issue? Is there a way around it?

I am sure the problem arises because of the driver since once I uninstall it, I can successfully suspend and hibernate once again.

grahamc, 2016-09-22:
To avoid the problem all together, we unload it before hibernate / suspend
on NixOS and load it again on resume.

Check out the PowerManagement code blocks here:
https://github.com/NixOS/nixpkgs/blob/master/nixos/modules/hardware/video/webcam/facetimehd.nix

grahamc, 2016-09-22:
(Note that because of this forced unloading I have no idea if I have the
same problem.)

justanotherjason, 2016-09-22:
I tried running modprobe -r facetimehd before hibernating/suspending but it does not seem to affect anything. Is there a way to simply enable/disable it on demand? They only way of achieving this now seems to be uninstalling the driver and rebooting. I guess I may have to build it from source...

patjak, 2016-09-28:
I will have to find a proper fix for suspend/resume. The current solution is a hack and far from optimal. It's also interesting that ```modprobe -r``` doesn't work. That might indicate that we're not restoring the device to a proper state when unloading. Will investigate.

patjak, 2016-10-10:
btw, having the device open (even without streaming) when suspending will crash so all applications that are using the camera must be closed. Working on a fix...