patjak/facetimehd #100: Task frozen in non-interruptible sleep with misbehaving program

gcampax, 2016-10-10:
I'm not sure if this is an userspace or kernel problem, but I'm reporting it here because in theory userspace should not freeze the kernel in a manner that SIGKILL cannot recover.

I am testing 887d0f531ef7b91457be519474136c3355c5132b (mainline) on top of 4.8.0-0.rc7.git0.1.fc25.x86_64.
I have a possibly buggy program that uses V4L. I run it and the task froze in disk sleep state. Starting another, known correct program (cheese) also froze in disk sleep, possibly because the first program is still holding the camera open.

The first program (which I believe uses opencv to acquire the video) reported:
VIDEOIO ERROR: V4L: index 1 is not correct

The led of the camera briefly went on and then off again.

gcampax, 2016-10-10:
Stack traces of the hung tasks (captured while trying to suspend):

Oct 09 23:30:13 giovanni-macbook kernel: Freezing user space processes ... 
Oct 09 23:30:13 giovanni-macbook kernel: Freezing of tasks failed after 20.008 seconds (2 tasks refusing to freeze, wq_busy=0):
Oct 09 23:30:13 giovanni-macbook kernel: Magnifier       D ffff8c83ac323b98     0  1537   1391 0x00000004
Oct 09 23:30:13 giovanni-macbook kernel:  ffff8c83ac323b98 0000000000000000 ffff8c81590b5b80 ffff8c83eca09e80
Oct 09 23:30:13 giovanni-macbook kernel:  ffff8c80e6fdd5c0 ffff8c83ac324000 ffff8c80e6fdd5c0 ffffffffc12333a0
Oct 09 23:30:13 giovanni-macbook kernel:  ffff8c80e6fdd3f8 0000000000000000 ffff8c83ac323bb0 ffffffffa37fcb85
Oct 09 23:30:13 giovanni-macbook kernel: Call Trace:
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa37fcb85>] schedule+0x35/0x80
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc124fd78>] vb2_wait_for_all_buffers+0x98/0xf0 [videobuf2_core]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa30e4cd0>] ? prepare_to_wait_event+0xf0/0xf0
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc12698ce>] fthd_stop_streaming+0x2e/0xc0 [facetimehd]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc124ed9b>] __vb2_queue_cancel+0x2b/0x1a0 [videobuf2_core]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc124fc78>] vb2_core_streamoff+0x28/0x90 [videobuf2_core]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc125a1f9>] vb2_streamoff+0x29/0x50 [videobuf2_v4l2]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc125a262>] vb2_ioctl_streamoff+0x42/0x50 [videobuf2_v4l2]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc1217a7a>] v4l_streamoff+0x1a/0x20 [videodev]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc121b475>] __video_do_ioctl+0x295/0x310 [videodev]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa31238e5>] ? get_futex_key+0x155/0x480
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc121af0f>] video_usercopy+0x34f/0x600 [videodev]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc121b1e0>] ? video_ioctl2+0x20/0x20 [videodev]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa3125af4>] ? do_futex+0x2c4/0xaf0
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc121b1d5>] video_ioctl2+0x15/0x20 [videodev]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc121668b>] v4l2_ioctl+0xbb/0xf0 [videodev]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa32648c3>] do_vfs_ioctl+0xa3/0x5f0
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa3264e89>] SyS_ioctl+0x79/0x90
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa38012f2>] entry_SYSCALL_64_fastpath+0x1a/0xa4
Oct 09 23:30:13 giovanni-macbook kernel: cheese          D ffff8c7ff35bfd58     0  1723   1145 0x00000004
Oct 09 23:30:13 giovanni-macbook kernel:  ffff8c7ff35bfd58 00ffffffa31238e5 ffff8c7fc6f10000 ffffffffa3e0d500
Oct 09 23:30:13 giovanni-macbook kernel:  ffff8c7ff7c86100 ffff8c7ff35c0000 ffff8c7fc6f10000 ffff8c80e6fdd634
Oct 09 23:30:13 giovanni-macbook kernel:  00000000ffffffff ffff8c80e6fdd638 ffff8c7ff35bfd70 ffffffffa37fcb85
Oct 09 23:30:13 giovanni-macbook kernel: Call Trace:
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa37fcb85>] schedule+0x35/0x80
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa37fce4e>] schedule_preempt_disabled+0xe/0x10
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa37fe8ae>] __mutex_lock_slowpath+0xae/0x120
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa37fe93f>] mutex_lock+0x1f/0x30
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc12597c9>] _vb2_fop_release+0x29/0xa0 [videobuf2_v4l2]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc125986a>] vb2_fop_release+0x2a/0x40 [videobuf2_v4l2]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffc1216478>] v4l2_release+0x38/0x80 [videodev]
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa325246f>] __fput+0xdf/0x1f0
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa32525be>] ____fput+0xe/0x10
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa30bf010>] task_work_run+0x80/0xa0
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa30032d2>] exit_to_usermode_loop+0xc2/0xd0
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa3003be1>] syscall_return_slowpath+0xa1/0xb0
Oct 09 23:30:13 giovanni-macbook kernel:  [<ffffffffa380137a>] entry_SYSCALL_64_fastpath+0xa2/0xa4

It looks like the second task is waiting on a mutex locked by the first task.

patjak, 2016-10-12:
It sounds like a known driver issue. Suspending the device while it is still opened will cause problems. My plan is to fix this as soon as I get some time to spend on the driver again.

gcampax, 2016-10-12:
It don't think this is the same issue as suspending: this causes the task to freeze while the system is otherwise running.
It just happens that I got a stack trace when I suspended and the kernel realized what was going on.

patjak, 2016-10-12:
Ah, I misunderstood your report. I agree that it's probably a different issue so reopening. How can I reproduce this?